espressif/arduino-esp32 #2685: assertion "pbuf_free: p->ref > 0" failed: file

Chriserke, 2019-04-18:
[https://gyazo.com/ca1456ab9ee0bb049cce73b4db5d3ec5](url)

Why am i getting this error and how would i go about fixing it?

`#include "WiFi.h"
#include "AsyncUDP.h"

const char* ssid = "Chrisiaan wifi";
const char* password =  "Buijse1234";

AsyncUDP udp;

void setup()
{
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
    if(udp.listen(44445)) {
        Serial.print("UDP Listening on IP: ");
        Serial.println(WiFi.localIP());
        udp.onPacket([](AsyncUDPPacket packet) {
            Serial.print("UDP Packet Type: ");
            Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
            Serial.print(", From: ");
            Serial.print(packet.remoteIP());
            Serial.print(":");
            Serial.print(packet.remotePort());
            Serial.print(", To: ");
            Serial.print(packet.localIP());
            Serial.print(":");
            Serial.print(packet.localPort());
            Serial.print(", Length: ");
            Serial.print(packet.length());
            Serial.print(", Data: ");
            Serial.write(packet.data(), packet.length());
            Serial.println();
            //reply to the client
            packet.printf("Got %u bytes of data", packet.length());
        });
    }
}

void loop()
{
    delay(1000);
    //Send broadcast
    //udp.broadcast("Anyone here?");
}`

lbernstone, 2019-04-19:
https://github.com/espressif/arduino-esp32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md#L19

Chriserke, 2019-04-19:
Well the issue is shows up when i send Hello world from a small python server. What im trying to do is have  the esp32 be a UDP server while my python side sends a msg and have it recieve it. Every time it receives a message it will crash with this error code 
[image]

As for hardware im not sure where to find it all but 

board: ESP32 Dev Module v1 
IDE name: Arduino IDE 1.8.9   Flash Frequency: 80Mhz 
PSRAM enabled: no  
Upload Speed: 512000 Computer OS: Windows 10

Not sure about IDF component
And Core Installation version but i've installed most of the stuff yesterday through gitbash and i assume its up to date

Regards Chris.

m0dd, 2019-04-20:
> Well the issue is shows up when i send Hello world from a small python server. What im trying to do is have the esp32 be a UDP server while my python side sends a msg and have it recieve it. Every [...]

Hi! I have the same problem than Chriserke but with different hardware.

### Hardware:
Board:							WEMOS LOLIN32
Chip:                                                        ESP32D0WDQ6 (revision 1)
Core Installation version:			1.0.2 (https://github.com/espressif/arduino-esp32/releases/tag/1.0.2)
IDE name:						Arduino IDE 1.8.9 (Windows Store 1.8.21.0)
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:				       921600
Computer OS:                                        Windows 10

### Description:
As soon as the udp packet is received, the mcu resets with following serial message:
```
assertion "pbuf_free: p->ref > 0" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/pbuf.c", line 765, function: pbuf_free
abort() was called at PC 0x400d725b on core 1

Backtrace: 0x4008c7e0:0x3ffc8ce0 0x4008ca11:0x3ffc8d00 0x400d725b:0x3ffc8d20 0x400f0473:0x3ffc8d50 0x400d2bbf:0x3ffc8d70 0x400d2c0f:0x3ffc8e00 0x4008877d:0x3ffc8e30

```

I also tried to comment out everything inside the udp.onPacket(AsyncUDPPacket packet) {} to make sure that this has nothing to do with string problems.

### Sketch: 
```cpp

#include "WiFi.h"
#include "AsyncUDP.h"

const char * ssid           = "Schnitzelgrill5000";
const char * password       = "130657823630638916796606";
String       client_address = "C01";
int          udpPort        = 7001; // Make sure this port matches the 'Server' port!
int          Control_pin    = 22;   // The device pin on the client to control e.g. an LED or Relay On/Off
String       received_message;

AsyncUDP udp;

void setup()
{
  delay(5000);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("WiFi Failed");
    while (1) {
      delay(1000);
    }
  }
  Serial.println("Listening on Port: " + String(udpPort));
  if (udp.listen(udpPort)) {
    Serial.print("UDP Listening on IP: ");
    Serial.println(WiFi.localIP());
    udp.onPacket([](AsyncUDPPacket packet) {
      //received_message = packet.readStringUntil('\n');
      //Serial.println(received_message);
      //received_message.trim(); // Remove unwanted characters
      // Test for a valid message structure 
//      if (received_message.startsWith(client_address) || received_message.startsWith("HAC") || received_message.startsWith("LAC"))
//      { // e.g. C01H or C)1L or ACH or ACL
//        if (received_message.endsWith("H")) {
//          Serial.println("Turning ON Device for "  + client_address);
//          Device_Control("ON");
//        }
//        if (received_message.endsWith("L")) {
//          Serial.println("Turning OFF Device for " + client_address);
//          Device_Control("OFF");
//        }
//        if (received_message == "HAC")      {
//          Serial.println("Turning ON Device for "  + client_address + " - global ON");
//          Device_Control("ON");
//        }
//        if (received_message == "LAC")      {
//          Serial.println("Turning OFF Device for " + client_address + " - global OFF");
//          Device_Control("OFF");
//        }
//      }
    });
  }
  //udp.broadcast("Device operated correctly");
}

void loop()
{
  Serial.println("mainloop running");
  delay(1000);
  Device_Control("ON");
  delay(1000);
  Device_Control("OFF");
}

void Device_Control(String Device_mode) {
  if (Device_mode ==  "ON") {
    pinMode(Control_pin, OUTPUT);
    digitalWrite(Control_pin, HIGH);
  }
  else {
    pinMode(Control_pin, OUTPUT);
    digitalWrite(Control_pin, LOW);
  }
}
```

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
Listening on Port: 7001
UDP Listening on IP: 192.168.0.78
mainloop running
mainloop running
mainloop running
mainloop running
mainloop running
mainloop running
mainloop running
mainloop running
mainloop running
mainloop running
mainloop running
assertion "pbuf_free: p->ref > 0" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/pbuf.c", line 765, function: pbuf_free
abort() was called at PC 0x400d725b on core 1

Backtrace: 0x4008c7e0:0x3ffc8ce0 0x4008ca11:0x3ffc8d00 0x400d725b:0x3ffc8d20 0x400f0473:0x3ffc8d50 0x400d2bbf:0x3ffc8d70 0x400d2c0f:0x3ffc8e00 0x4008877d:0x3ffc8e30

```

me-no-dev, 2019-04-21:
revert this change and give it a go: https://github.com/espressif/arduino-esp32/commit/a87b2ec69054ce43a48bd042599b4185267ebc1d#diff-3b5c94ff125c7d1f27ea52a9139410ec

lbernstone, 2019-04-21:
Reverting works for me.  I don't see any memory leaking between receives, but it does not seem like I get all the memory back when I destroy the object (a little over 4k consumed).

Chriserke, 2019-04-21:
Now i can run the UDPAsync example, works fine thanks.

Edris89, 2019-05-09:
I had the same problem today....May 2019
Any fix for this?

Tony-Alvarez, 2019-05-13:
#MeToo

using AsyncUDP just after updating to ESP boards v1.0.2 and got this error upon receiving UDP packets.

```
assertion "pbuf_free: p->ref > 0" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/pbuf.c", line 765, function: pbuf_free
abort() was called at PC 0x400dfc0f on core 1

Backtrace: 0x4008c88c:0x3ffc9fc0 0x4008cabd:0x3ffc9fe0 0x400dfc0f:0x3ffca000 0x400fa29f:0x3ffca030 0x400d9e7f:0x3ffca050 0x400d9ecf:0x3ffca0e0 0x40088829:0x3ffca110

Rebooting...
```

Reverting to v1.0.1 fixed things

axa88, 2019-05-13:
#MeToo

using AsyncUDP just after updating to ESP boards v1.0.2 and got this error upon receiving UDP packets.

assertion "pbuf_free: p->ref > 0" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/pbuf.c", line 765, function: pbuf_free
abort() was called at PC 0x400dfc0f on core 1

Backtrace: 0x4008c88c:0x3ffc9fc0 0x4008cabd:0x3ffc9fe0 0x400dfc0f:0x3ffca000 0x400fa29f:0x3ffca030 0x400d9e7f:0x3ffca050 0x400d9ecf:0x3ffca0e0 0x40088829:0x3ffca110

Rebooting...

Reverting to v1.0.1 fixed things

me-no-dev, 2019-05-13:
change is reverted. will be released soon.

closed 2019-05-13 by me-no-dev

jweigelt, 2019-05-13:
So if I understand this correctly, this relies on the callback accepting the packet by value and not by reference (as specified in AsyncUDP.h). This will call the destructor twice but just copy the packet without calling the constructor and pbuf_ref().

Accepting the packet by reference will therefore leak memory (and eventually crash) as the destructor is only called once.

Passing the packet (by value) to another function from the callback function will also crash as the destructor will receive another call without a prior call to pbuf_ref().

If this intended behaviour it would be nice to note this somewhere. Without looking through the source files I find this to be rather counter-intuitive.

folosada, 2019-05-19:
@me-no-dev I downloaded the latest code to AsyncUDP but I still have the same issue.

``
assertion "pbuf_free: p->ref > 0" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/pbuf.c", line 765, function: pbuf_free
abort() was called at PC 0x400d8d83 on core 1

Backtrace: 0x4008cbac:0x3ffca530 0x4008cddd:0x3ffca550 0x400d8d83:0x3ffca570 0x400f372f:0x3ffca5a0 0x40120a20:0x3ffca5c0 0x40120fcd:0x3ffca5e0 0x40121023:0x3ffca660 0x40088b49:0x3ffca690
``
If I comment the line 329 in AsyncUDP.cpp the crash isn't happen, but the packets works 1 time and I need wait between five and ten minutes to send the packets again.

Line 329:
``
AsyncUDPPacket::~AsyncUDPPacket()
{
    pbuf_free(_pb);
}
``

Board: Espressif ESP32 Dev Module
IDE Name: Visual Studio Code with PlatformIO
Computer OS: Windows 10