espressif/arduino-esp32 #2687: exit Status 1 mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();

calimero1975, 2019-04-20:
Hello,

when i compile the example from ESP-MESH Programming Guide i have become a exit Status 1 failure at this line:

mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();

here is my Compiler Text:
Arduino: 1.8.9 (Windows Vista), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\Maik Ruthsatz\Documents\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Maik Ruthsatz\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10809 -build-path C:\Users\MAIKRU~1\AppData\Local\Temp\arduino_build_859648 -warnings=all -build-cache C:\Users\MAIKRU~1\AppData\Local\Temp\arduino_cache_868111 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Maik Ruthsatz\Documents\Arduino\Projekte\esp_mesh_3\esp_mesh_3.ino
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\Maik Ruthsatz\Documents\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Maik Ruthsatz\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10809 -build-path C:\Users\MAIKRU~1\AppData\Local\Temp\arduino_build_859648 -warnings=all -build-cache C:\Users\MAIKRU~1\AppData\Local\Temp\arduino_cache_868111 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Maik Ruthsatz\Documents\Arduino\Projekte\esp_mesh_3\esp_mesh_3.ino
Using board 'esp32' from platform in folder: C:\Users\Maik Ruthsatz\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\Maik Ruthsatz\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/config" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/app_trace" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/app_update" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/asio" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bootloader_support" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bt" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/coap" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/console" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/driver" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp-tls" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp32" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_event" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_http_client" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_http_server" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/ethernet" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/expat" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/fatfs" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/freemodbus" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/freertos" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/heap" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/idf_test" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/jsmn" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/json" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/libsodium" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/log" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/lwip" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/mbedtls" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/mdns" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/micro-ecc" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/mqtt" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/newlib" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/nghttp" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/nvs_flash" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/openssl" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/protobuf-c" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/protocomm" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/pthread" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/sdmmc" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/soc" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/spi_flash" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/spiffs" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/tcp_transport" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/tcpip_adapter" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/ulp" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/vfs" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/wear_levelling" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/wifi_provisioning" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/wpa_supplicant" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp32-camera" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp-face" "-IC:\\Users\\Maik Ruthsatz\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/fb_gfx" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volat [... cut ...]
  wifi_init_config_t config = WIFI_INIT_CONFIG_DEFAULT();
  ESP_ERROR_CHECK(esp_wifi_init(&config));
  ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_FLASH));
  ESP_ERROR_CHECK(esp_wifi_start());

  /*  mesh initialization */
  ESP_ERROR_CHECK(esp_mesh_init());

  /* Enable the Mesh IE encryption by default */
  mesh_cfg_t &cfg = MESH_INIT_CONFIG_DEFAULT();
  /* mesh ID */
  //memcpy((uint8_t *) &cfg.mesh_id, MESH_ID, 6);
  
}

void loop() {
  // put your main code here, to run repeatedly:

}

Thanks for your help.

lbernstone, 2019-04-20:
```
//mesh_cfg_t &cfg = MESH_INIT_CONFIG_DEFAULT();
mesh_cfg_t cfg;
cfg.crypto_funcs = &g_wifi_default_mesh_crypto_funcs;
```

calimero1975, 2019-04-20:
Hello Ibernstone,
thank you so much for this snipped.

bonastreyair, 2019-05-30:
I also have this problem on Arduino IDE, I tried this solution you mention https://github.com/espressif/arduino-esp32/issues/2687#issuecomment-485118917, but then following errors still appear...

- 'MESH_ID' was not declared in this scope
- 'mesh_event_handler' was not declared in this scope
- 'CONFIG_MESH_CHANNEL' was not declared in this scope
- 'CONFIG_MESH_ROUTER_SSID' was not declared in this scope
- 'CONFIG_MESH_ROUTER_PASSWD' was not declared in this sco
- 'CONFIG_MESH_AP_CONNECTIONS' was not declared in this scope
- 'CONFIG_MESH_AP_PASSWD' was not declared in this scope

Should I set them manually?

Is there out there any simple example to run ESP-MESH using Arduino IDE?

Thanks,

closed 2019-08-14 by stale[bot]