espressif/arduino-esp32 #2689: DHCP not exchanging gateway address for 192.168.1.1

ThinkalVB, 2019-04-20:
I was trying to set up a simple server WiFi softAP

Windows 10 (MSI GP63 Leopard 8RE)
The problem arises when the IP address that DHCP server leases is the same address as that of its gateway. ipconfig shows the gateway as 0.0.0.0 

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2-git
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:				       9600
Computer OS:                                        Windows 10

#include "WiFi.h"

char *ssid = "AirMesh";

IPAddress local_IP(192,168,1,0);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

void setup()
{
  Serial.begin(9600);
  Serial.println();

  Serial.print("Setting soft-AP ... ");
  Serial.println(WiFi.softAP(ssid) ? "Ready" : "Failed!");

  Serial.print("Setting soft-AP configuration ... ");
  Serial.println(WiFi.softAPConfig(local_IP, gateway, subnet) ? "Ready" : "Failed!");

  Serial.print("Soft-AP IP address = ");
  Serial.println(WiFi.softAPIP());

}

void loop() {}

### Debug Messages:
Wireless LAN adapter Wi-Fi:

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::8037:1f14:5989:a2f4%18
   IPv4 Address. . . . . . . . . . . : 192.168.1.1
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : 0.0.0.0

C:\Users\think>

lbernstone, 2019-04-20:
The class is not going to prevent you from creating completely broken networks.  Use 192.168.1.1 for both the local_ip and gateway.

stickbreaker, 2019-04-21:
@ThinkalVB your local IP address is invalid.

From:[TechStuff IPv4 classes, subnets](http://www.zytrax.com/tech/protocols/ip-classes.html)
>Every network created by a Net/Subnetmask is defined by the standards to have two special (unuseable by normal hosts or PCs) IP addresses as follows:
>
>    A host address part of all 0s is the network address for the network/subnetwork
>    A host address part of all 1s is the broadcast address for the network/subnetwork
>
>The following table illustrates this:
>
>Network Address | Netmask | No. of IPs | Network IP | Broadcast IP
>172.28.227.0 | 255.255.255.0 | 256 | 172.28.227.0 | 172.28.227.255
>192.168.2.64 | 255.255.255.224 | 32 | 192.168.2.64 | 192.168.2.95
>192.168.15.128 | 255.255.255.248 | 8 | 192.168.15.128 | 192.168.15.135
>
>Notes:
>
>    Technically, the Network address consists of a host address part (or host ID) of all 0s, which is >interpreted as meaning 'this' (host), whereas the Broadcast address consists of a host address part (or >host ID) of all 1s which is interpreted as 'all' (hosts).

Chuck.

ThinkalVB, 2019-04-21:
> The class is not going to prevent you from creating completely broken networks. Use 192.168.1.1 for both the local_ip and gateway.

I was thinking of an awkward scenario where 192.168.1.0 is assigning IP to a gateway node. I think you meant "The class is going to prevent ......."

lbernstone, 2019-04-21:
If you want a router at .1, put the server at .2 and manually assign the address on the gateway interface.  You should not be handing out critical addresses through DHCP, and .0 is a special address.

ThinkalVB, 2019-04-21:
> If you want a router at .1, put the softAP at .2 and manually assign the address on the gateway interface. You should not be handing out critical addresses through DHCP, and .0 is a special address.

Can you help me with the code in assigning; static IP and assigning an IP to a dynamic host?

lbernstone, 2019-04-21:
This forum is for issues with the codebase, not a training site.  There is an example for static IP with WiFi.

ThinkalVB, 2019-04-21:
> This forum is for issues with the codebase, not a training site. There is an example for static IP with WiFi.

Ok, thanks will look into it.

closed 2019-04-21 by ThinkalVB