espressif/arduino-esp32 #2693: certification file from spiffs

duqrlpig, 2019-04-21:
i get ca, cert, private from spiffs

  caCharArray = caStr.c_str();
  Serial.println("const ca file-----------------------------------------------------");
  Serial.println(caCharArray);

  certCharArray = certStr.c_str();
  Serial.println("const cert file-----------------------------------------------------");
  Serial.println(certCharArray);

  privateCharArray = privateStr.c_str();
  Serial.println("const private file-----------------------------------------------------");
  Serial.println(privateCharArray);

  net.setCACert(caCharArray);delay(1000);
  net.setCertificate(certCharArray);delay(1000);
  net.setPrivateKey(privateCharArray);delay(1000);

String getStringFromSpiffsFile(String path) {
  File file = SPIFFS.open(path, "r");
  if (!file) {
    Serial.print("Failed to open : ");
    Serial.println(path);
    return "";
  }
  else {
    Serial.print("success to open : ");
    Serial.println(path);
  }
  char a;
  String tmpStr = "";
  while (file.available()) {
    a = file.read();
    tmpStr += a;
  }
  return tmpStr;
}

but i cannot connect to mqtt host.

lbernstone, 2019-04-21:
Not exactly the same, but you can test with https://github.com/lbernstone/setCerts

duqrlpig, 2019-04-21:
Thank you. Bui it is not work. Too large certificate

closed 2019-08-15 by stale[bot]