espressif/arduino-esp32 #2696: 1.0.2 Version Interrupt makes Esp32 to reset!

ricaun, 2019-04-21:
I'm developing a library to work with Lora and esp32,
I've tested the library with the version 1.0.1 and works ok, but when I updated to 1.0.2 the board start reset when I try to make some things on the interrupt callback.

And I have the same problem when I try to do too many things on the callback.

I believe the esp32 doesn't like the idea to work on the interrupt, in my case I'm trying to send some Lora message on the interrupt and the board reset...
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008161a  PS      : 0x00050031  A0      : 0x800d1021  A1      : 0x3ffbe7c0  
A2      : 0x00000040  A3      : 0x00018040  A4      : 0x000637ff  A5      : 0x3ffbe7a0  
A6      : 0x00000000  A7      : 0x3ffbfe5c  A8      : 0x00000000  A9      : 0x04000000  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x80080e7c  A13     : 0x3ffbe780  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800d102d  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x4008161a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082835

Backtrace: 0x4008161a:0x3ffbe7c0 0x400d101e:0x3ffb1f90 0x400d2ae9:0x3ffb1fb0 0x40087911:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
LoRaNow Simple Gateway
```

The ARV and esp8266 work fine, but the esp32 doesn't, I probably need to move the callback to loop that should solve the problem.

This is the library https://github.com/ricaun/LoRaNow
I'm still working to make tight.

If someone has some tips and tricks about how I can solve this issue, especially if can improve the library I will be really glad.

Thanks.

lbernstone, 2019-04-22:
https://github.com/espressif/arduino-esp32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md#L19

ricaun, 2019-08-12:
Hello, I gonna close the issue.
Kinda fix the problem, removing some code from the interrupt.

closed 2019-08-12 by ricaun