espressif/arduino-esp32 #2699: millis() crashes if used in irq handler called too early

tht, 2019-04-23:
I'm trying to use `millis()` in an irq handler. But if the handler is called very early in the boot process the ESP32 crashes.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE 1.8.9
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        macOS 10.14.4

**There is a 10kHz signal generator on pin 26!**

### Description:

I'm trying to use `millis()` in an irq handler. But if the handler is called very early in the boot process the ESP32 crashes. If the signal generator on pin 26 is not switched on while the ESP32 is booting this code works perfectly fine.

The problem only occurs if WiFi is used.

### Sketch:
```cpp
#include <WiFi.h>

// add valid WiFi credentials here
const char* ssid = "....";
const char* password = "....";

void IRAM_ATTR _irqHandler() {
  uint8_t data = digitalRead(26); // dummy - not strictly needed
  uint32_t now = millis(); // comment out to prevent crash
}

void setup() {
  Serial.begin(115200);

  pinMode(26, INPUT);
  attachInterrupt(26, _irqHandler, FALLING);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
}

void loop() {
}
```

### Debug Messages with signal generator switched on during boot:
```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013b184: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr 
```

This is the full crash log on serial – there is no stack trace to decode.

### Debug Messages with signal generator switched OFF:
```
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.x.y, MASK: 255.255.255.0, GW: 192.168.x.1
```

tht, 2019-04-23:
Other, possible relevant issues, always recommend to make sure to flag the interrupt handler as `IRAM_ATTR` what I already did. I've also checked `millis()` - it also uses `IRAM_ATTR`.

TiSiAi, 2019-04-29:
Same Random problem with a timer interrupt

`
void IRAM_ATTR onTimerStd()
{
  portENTER_CRITICAL_ISR(&timerMuxStd);
  isrCounterStd++;
  //lastIsrAtStd = millis(); //> random crash
  portEXIT_CRITICAL_ISR(&timerMuxStd);
}
`

closed 2019-08-14 by stale[bot]