espressif/arduino-esp32 #2702: Fail to Build while using esp-idf v3.2

DarvosBR, 2019-04-23:
I'm having issues to build the MASTER arduin-esp32 with the matching esp-idf commit (I suppose)

### Hardware:
Board:	ESP32 Dev Module
Core Installation version: 1.0.1-git
IDE name: Esp-idf
Flash Frequency: 40Mhz
PSRAM enabled: yes
Upload Speed: 115200
Computer OS: Windows 10

### Description:
Trying to build a simple blink file using the arduino-esp32 as a component, the latest esp-idf compatible is the v3.2 as far as I know, there is a few instructions under the v3.2 build saying Remove BLE submodule ; Add BLE lib source. I belive thats my problem, only it does not make any sense to me.

Error MSG:
[image]

atanisoft, 2019-04-23:
You need to make sure you have the correct dependencies in place for using the WiFiClientSecure library, this includes the SSL (mbed tls) libraries. You can check the tools/sdk/sdkconfig file in the arduino-esp32 tree to see what differences there are (I'd actually suggest using this as the starting point and modify for your project)

me-no-dev, 2019-04-23:
arduino is currently NOT tracking the master and is on v3.2 instead. Will go back to tracking master soon. Need to make sure that if there are any issues in 3.2 that we fix them first. I am also preparing an easy-start repo for building Arduino with IDF

DarvosBR, 2019-04-23:
> arduino is currently NOT tracking the master and is on v3.2 instead. Will go back to tracking master soon. Need to make sure that if there are any issues in 3.2 that we fix them first. I am also preparing an easy-start repo for building Arduino with IDF

I'm actualy using v3.2

`commit 286202caa31b61c2182209f37f8069a0b60fb942 (HEAD, tag: v3.2-rc, tag: v3.2)
`

This one to be more specific!

> You need to make sure you have the correct dependencies in place for using the WiFiClientSecure library, this includes the SSL (mbed tls) libraries. You can check the tools/sdk/sdkconfig file in the [...]

I wish I could only switch WiFiClientSecure off, but this option is no valid for me on Menuconfig, it seems to me a few WiFi libs are mandatory.

DarvosBR, 2019-04-23:
> You need to make sure you have the correct dependencies in place for using the WiFiClientSecure library, this includes the SSL (mbed tls) libraries. You can check the tools/sdk/sdkconfig file in the [...]

Brilliant, ur tip solve almost all my problems, thank you, I was stuck on it for 5 hours.

Only one question to close this, 
`rtc_wdt_flashboot_mode_enable` in function `esp_restart_noos` of the file `system_api.o` is saying undefined reference, i'm about to get in there and see if I can just comment or take it out, but before i do this, any menuconfig can help me with it? taking it out is a good idea? idk, just want an opinion, besides that i'm supper happy, the only bug there is left!!

atanisoft, 2019-04-23:
esp_restart_noos was deprecated long back and removed IIRC from ESP-IDF.

DarvosBR, 2019-04-23:
> esp_restart_noos was deprecated long back and removed IIRC from ESP-IDF. I'd just remove it.

ok ty once again, gonna try to remove it

atanisoft, 2019-04-23:
another option is replace it with esp_restart()

DarvosBR, 2019-04-23:
I commented the whole function and replaced when it was used with esp_restart(), thank you!

closed 2019-07-01 by Nathan-ma