espressif/arduino-esp32 #2703: I2C issues with Sparkfun Triad AS7265x

kennycoder, 2019-04-23:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation version:			?1.0.2?
IDE name:						?Arduino IDE?
Flash Frequency:					?80Mhz?
PSRAM enabled:                                    ?no?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10?

### Description:
I'm porting my portable spectrometer from atmega2560 to esp32 board now along with other peripherals which work perfectly fine except for this AS7262x triad sensor. It's a rather simple i2c device that has a library provided by sparkfun ( https://github.com/sparkfun/SparkFun_AS7265x_Arduino_Library ). Under atmega chips it works perfectly fine but with esp32 I'm having big issues.

After reading plenty on the subject here I tried pretty much everything from pullup resistors ranging from 1k to 10k to slowing things down and setting bigger timeouts.

What is happening basically is that the library code that reads and writes to registers doesn't work as expected. In the readRegister() function, after endTransmission() and requestFrom() calls, the _i2cPort->available() is almost always false. Sometimes endTransmission also fails but rarely. I don't see anything crazy in these methods, just a small wrapper of what I would do myself.

The most bizarre thing is that after i reset esp32 couple of times and reset the triad module, *sometimes* it starts working, but usually very very slowly. I don't have a oscilloscope on me right now to dig deeper but maybe someone has an idea.

### Source from provided library that I mentioned above.
```cpp

//Reads from a give location from the AS726x
uint8_t AS7265X::readRegister(uint8_t addr)
{
  _i2cPort->beginTransmission(AS7265X_ADDR);
  _i2cPort->write(addr);
  if (_i2cPort->endTransmission() != 0)
  {
//    Serial.println("No ack! read 1");
    return (0); //Device failed to ack
  }

  _i2cPort->requestFrom((uint8_t)AS7265X_ADDR, (uint8_t)1);
  if (_i2cPort->available()) {
    return (_i2cPort->read());
  }

//  Serial.println("No ack! read 2");
  return (0); //Device failed to respond
}

//Write a value to a spot in the AS726x
boolean AS7265X::writeRegister(uint8_t addr, uint8_t val)
{
  _i2cPort->beginTransmission(AS7265X_ADDR);
  _i2cPort->write(addr);
  _i2cPort->write(val);
  if (_i2cPort->endTransmission() != 0)
  {
//    Serial.println("No ack! write");
    return (false); //Device failed to ack
  }

  return (true);
}
```

### Debug Messages totally randomly showing up.. not sure if 100% indication of what is wrong
```
[esp32-hal-i2c.c:1127] i2cProcQueue(): I2C exitCode=0x112
```

atanisoft, 2019-04-23:
@stickbreaker

kennycoder, 2019-04-23:
Interestingly enough if I add a some delays between beginTransmission, write, endTransmission, requestFrom, etc.. it tends to work more times.. but still super unreliable. I tried different wiring but my connection is rather simple:

```
#define SDA1 21
#define SCL1 22
...
  Wire.begin(SDA,SCL);
  Wire.setClock(100000);
```

stickbreaker, 2019-04-23:
@kennycoder Lets start by adding more error checking:
```c++
//Reads from a give location from the AS726x
uint8_t AS7265X::readRegister(uint8_t addr)
{
  _i2cPort->beginTransmission(AS7265X_ADDR);
  _i2cPort->write(addr);
  if (_i2cPort->endTransmission() != 0)
  {
   Serial.printf("set address for Read failed=%d(%s)\n",_i2cPort->lastError(),_i2cPort->getErrorText(_i2cPort->lastError()));
//    Serial.println("No ack! read 1");
    return (0); //Device failed to ack
  }

  _i2cPort->requestFrom((uint8_t)AS7265X_ADDR, (uint8_t)1);
  if (_i2cPort->available()) {
    return (_i2cPort->read());
  }
   Serial.printf("requestFrom() failed=%d(%s)\n",_i2cPort->lastError(),_i2cPort->getErrorText(_i2cPort->lastError()));
  return (0); //Device failed to respond
}

//Write a value to a spot in the AS726x
boolean AS7265X::writeRegister(uint8_t addr, uint8_t val)
{
  _i2cPort->beginTransmission(AS7265X_ADDR);
  _i2cPort->write(addr);
  _i2cPort->write(val);
  if (_i2cPort->endTransmission() != 0)
  {
   Serial.printf("Write Register(%d)=0x%02x failed=%d(%s)\n",addr,value,_i2cPort->lastError(),_i2cPort->getErrorText(_i2cPort->lastError()));//    
    return (false); //Device failed to ack
  }
  return (true);
}
```

A second thing you may be running into is the ESP32 Does not support SCL stretching longer than 13.1ms when the CPU clock is above 80mhz.  I have no knowledge on this AS7265x, does it do clock stretching?

If these error message don't help enough, you can increase the CORE DEBUG LEVEL and see any errors the I2C subsystem detects.  They will be displayed on the Serial monitor.

The next level is to enable indepth debugging inside the I2C driver go to `hardware\Espressif\esp32\cores\esp32\esp32-hal-i2c.c` line 45, uncomment:
```c++
//#define ENABLE_I2C_DEBUG_BUFFER
```
Then set CORE DEBUG LEVEL to VERBOSE and recompile and it will generate comprehensive debug info.

Chuck.

kennycoder, 2019-04-23:
Thank you very much Chuck. I'll give some updates tomorrow. Really appreciate it.

kennycoder, 2019-04-24:
So @stickbreaker , I'm getting roughly this (it loops over and over since the lib continues to try):

```
V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=1, scl(22)=1
[D][esp32-hal-i2c.c:1419] i2cCheckLineState(): Recovered after 5 Cycles
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
... 150 of these ...
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
....
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
... 150 of these ... and then it gets the busy, reinit again.

```

kennycoder, 2019-04-24:
One interesting observation.. if I reduce the cpu freq. down to 40Mhz it starts working almost flawlessly. (mode with BT/Wifi disabled as far as i understand)

Then the log is as follows:
```
i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=40Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=5
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=40Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=5
s[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=40Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=5
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
Sensor started
TFT Started
UI prepared
UI Drawn

```

stickbreaker, 2019-04-24:
@kennycoder The requestfrom() is acting weird, It is showing no error.  Change the error test to:
```c++
  uint8_t count=_i2cPort->requestFrom((uint8_t)AS7265X_ADDR, (uint8_t)1);
  if (count != 0) {
    return (_i2cPort->read());
  }
  else {
     Serial.printf("requestFrom() failed=%d(%s)\n",_i2cPort->lastError(),_i2cPort->getErrorText(_i2cPort->lastError()));
  }
  return (0); //Device failed to respond
}
```
Slowing down the CPU changes the `timeout` at 40mhz the max timeout is ~26ms.

Bus busy is indicative of a SCL stretch event that exceeded the timeout.

I don't see any timeout errors in your log dumps
>V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1

I would expect one Here:

>[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)

And Here:
>[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)

To handle this sensor, you might have to use `bool Wire.busy();` to poll the bus until the sensor is ready.
Chuck.

kennycoder, 2019-04-24:
Same story:
```requestFrom() failed=0(OK)```

Also some extra debug in loop
```requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
requestFrom() failed=0(OK)
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbebe8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb843c
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=5
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb84c0 bits=112
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb84f0
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb849c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:288] i2cDumpDqData(): [0] 7bit 49 W STOP buf@=0x3ffc062e, len=1, pos=1, ctrl=11111
[I][esp32-hal-i2c.c:306] i2cDumpDqData(): 0x0000: .                                00 
[I][esp32-hal-i2c.c:346] i2cDumpInts(): 0 row	count	INTR	TX	RX	Tick 
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [01]	0x0001	0x0002	0x0002	0x0000	0x00004ed7
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [02]	0x0001	0x0200	0x0000	0x0000	0x00004ed7
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [03]	0x0001	0x0020	0x0000	0x0000	0x00004ed7
```

stickbreaker, 2019-04-25:
@kennycoder If you changed the code you will not see this:
>same story: requestFrom() failed=0(OK)

`lastError()` is reset on the next `Wire()` call, `Wire.available()` will clear any error generated by `requestFrom()`, so the 0(OK) is saying Wire.available() encountered no problems.

The error =112 is showing an arbitration fault.  I read the datasheet for your sensor,  It show a module with 4 devices, 3 individual sensors and one controller. 

 Are you connected to the I2C bus between the controller and the individual sensors? 

 If the arbitration error is accurate, then the coding for the I2C transactions needs to understand and handle a multiMaster environment.  

Basically when an arbitration error occurs, the code needs to hang, waiting for `Wire.busy()` to clear(come back false) then reissue the failed communication sequence.  With timeouts and retry failure handling.

Chuck.

kennycoder, 2019-04-28:
Sorry for the delayed response. I've managed to extract some more meaningful errors but still have no clue.   
```
set address for Read failed=2(ACK)
Write Register(1)=0x2b failed=2(ACK)
...
Write Register(1)=0x00 failed=2(ACK)
```

I'll submit the issue to  sparkfun as it seems it's fully related to their sensor and i2c implementation there.
Thanks again,
Nikolai

kennycoder, 2019-04-28:
I feel so stupid. Sorry about this. It was somehow interference from external power supply. SPI was working fine but i2c was pretty sensitive. Now all works fine.
Thanks for all your help.

closed 2019-04-28 by kennycoder