espressif/arduino-esp32 #2704: Server.h not matching Arduino API

jandrassy, 2019-04-24:
The Arduino Server class has pure virtual begin() without parameters. The esp32 core Server.h has pure virtual begin() with parameter. Arduino networking libraries like Ethernet and UIPEthernet don't compile on esp32 core because they implement begin without parameter as required by Arduino API.

me-no-dev, 2019-04-25:
And your proposal is? We need to define the port for the server. How do those libraries you point to handle that?

me-no-dev, 2019-04-25:
You were presented with an issue template when you clicked `New issue`. Why is it so hard to fill it in?

jandrassy, 2019-04-25:
> And your proposal is? We need to define the port for the server. How do those libraries you point to handle that?

port is set with constructor for server in all Arduino networking libraries implementing Client and Server class. here is [an answer on how to change the port later](https://github.com/arduino-libraries/Ethernet/pull/39#issuecomment-239263298).

> You were presented with an issue template when you clicked `New issue`. Why is it so hard to fill it in?

I don't have this problem, I don't use esp32 in a project. I only try to make my libraries compatible with it. I created this issue because I know about this problem a long time and the new release still has it and added one more similar in Client.

closed 2019-08-15 by stale[bot]

k-zhai, 2020-06-12:
@me-no-dev Is there a fix for this?

me-no-dev, 2020-06-15:
there will be

mikatrash, 2020-06-21:
Would be soooooo coool. Curently building with the unapproved PR with unknow side affect, hands shaking @ every line :)

PlPt, 2020-07-12:
Move port definition to constructor and to it like EthernetServer from arduino-libs

tig, 2021-07-23:
This issue should not be closed as it has not been resolved.

I'm just a user trying to build some cross platform Arduino libraries. I just spent an hour trying to figure this issue out only to find a dozen or so closed issues scattered across a bunch of repos with suggestions of lame workarounds but no fix.'

```
error: cannot declare field 'Commands::_shellServer' to be of 
abstract type 'EthernetServer'```

The fix is to change esp32 to use 

@per1234 articulated the fix well here: https://github.com/arduino-libraries/Ethernet/issues/88

> The [ArduinoCore-api repository](https://github.com/arduino/ArduinoCore-API/blob/master/api/Server.h) contains the result of [Project Chainsaw](https://groups.google.com/a/arduino.cc/d/msg/developers/oyt_c5wePg8/_CWGyAG6CQAJ). This separates all the non-architecture specific code out of the core libraries so that the same code can be shared by all cores. This makes it much easier to maintain that code. The hope is that ArduinoCore-api will be used by the 3rd party hardware package authors in addition to its use by the official Arduino hardware packages. ArduinoCore-api is already in use in the [Arduino megaAVR Boards](https://github.com/arduino/ArduinoCore-megaavr) package.
> 
> Server.h is part of ArduinoCore-api. If Arduino core for the ESP32 eventually switches to using ArduinoCore-api, they will be using the official Server.h, rather than their non-standard Server.h.

I get the sense that the esp32 maintainers do not want to do this, but I can't understand why. 

In the meantime, if someone can suggest how I can get my cross-platform Arduino library build in platformio to automatically patch the esp32 `Server.h` as proposed here, I'd really appreciate it.

https://github.com/arduino-libraries/Ethernet/pull/107/files