espressif/arduino-esp32 #2705: sketch upload suddenly stops working

JohnGEyles, 2019-04-24:
### Hardware:
Board:							Adafruit HUZZAH32
Core Installation version:			not sure how to determine this
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      not sure how to determine this
Upload Speed:					921600
Computer OS:                                          Mac OSX

### Description:
I cannot upload a sketch.

After going through some contortions about a year ago trying to get the HUZZAH32 to work in my environment, and with things having worked perfectly since then, a similar problem has appeared out of nowhere.   It may be un-related, but at that time, using a different USB port (on the rear of the Apple monitor instead of on the tower itself) and using a different USB cable, fixed it.  And it's worked perfectly since then.   Maybe not the same problem, as it has no trouble finding the '/dev/cu.SLAB_USBtoUART' port.  

//Change the code below by your sketch

Doesn't matter.   Fails with the simplest sketches, for example, the built-in sketch "blink" (from Files->Examples->Basics in the Arduino IDE).

`
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
`
### Debug Messages:
esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

me-no-dev, 2019-04-25:
it sounds like the board can not reset itself into download mode. Try using a terminal like `CoolTerm` which allows manual switching the RTS/DTR lines so you can check if the board is able to go into Download Mode

JohnGEyles, 2019-04-25:
Thanks for the suggestion.   That looks like a cool little piece of software.   Maybe some learning curve, but it's worth a try.

JohnGEyles, 2019-04-25:
Installed CoolTerm, but not sure how to do what you suggest.   Can you point me to anything helpful ?

JohnGEyles, 2019-04-26:
One thing: when failure first occurred, I noticed I had another wire (a jumper clip) draped across the USB cord.  A little pressure against the micro-connector, not much though.  Possibly interrupted the upload ?

lbernstone, 2019-04-26:
Open a terminal session, and run `python <esp32_dir>/tools/esptool.py chip_id`.  If you get no response, you have a problem, either in the driver (make sure to check the driver security), or a physical issue.

JohnGEyles, 2019-04-26:
Yeah, I tried messing w/ esptool before.   This is what I got.   Don't understand it, seems like something broken in the python ...

`python esptool.py chip_id
esptool.py v2.1
Traceback (most recent call last):
  File "esptool.py", line 2524, in <module>
    _main()
  File "esptool.py", line 2517, in _main
    main()
  File "esptool.py", line 2240, in main
    esp = ESPLoader.detect_chip(args.port, initial_baud, args.before)
  File "esptool.py", line 203, in detect_chip
    detect_port = ESPLoader(port, baud)
  File "esptool.py", line 174, in __init__
    if isinstance(port, serial.Serial):
NameError: global name 'serial' is not defined
`

me-no-dev, 2019-04-26:
[image]

Those are the buttons that control RTS and DTR. Once you connect to the serial port, you can click them and see what will happen

JohnGEyles, 2019-04-26:
> Those are the buttons that control RTS and DTR. Once you connect to the serial port, you can click them and see what will happen

Yeah, I figured that out.   Tried wiggling 'em in various ways and no change.   Of course, if I didn't "disconnect" from the port before trying the upload again, the upload complained about not finding the port, that's to be expected I guess.

me-no-dev, 2019-04-26:
yes, but not getting any result from toggling those is not expected. There is something wrong with that board.

JohnGEyles, 2019-04-26:
> yes, but not getting any result from toggling those is not expected. There is something wrong with that board.

Yeah, I'm thinking that too (especially with that stuck-high SCK pin), and I'm happier that a $20 board failed than that my environment is somehow screwed up.  I need to make sure nothing in my system fried it, of course ...

JohnGEyles, 2019-04-30:
Received a new HUZZAH32 and it works fine.   Not sure what happened to the first one.   I believe I applied 5vdc to one of the input pins, I guess that fried it.

closed 2019-05-01 by JohnGEyles