espressif/arduino-esp32 #2712: Hold state of GIPIO LOW during deep sleep mode

andrelmbraga, 2019-04-26:
Hello guys,
I have a project with ESP32 WROOM in my own pcb board where I control one transistor to connect the ground my device sensors when my ESP32 is wake and when it goes to sleep it must have a GPIO in low lvl in order to put the transistor in cut mode.

The problem that I figured out is that the GPIO's are floating instead of stay in low mode during the sleep mode.

First I was trying to do this in arduino IDE, that can be the best option for me as my project was original made over there. However I didnt find any function to make the GPIO keep the same state during the deep sleep.

Looking in ESP-IDF i found these function:
gpio_deep_sleep_hold_en();
gpio_hold_en(gpio_num_t BLINK_GPIO);

I understood that they were made for this keep the gpio in previous state during the deep sleep, however i have tried without success.

Following the code so you can easily reproduce this issue:

Code to reproduce this issue
```cpp

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "esp_sleep.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#define BLINK_GPIO 25

void app_main()
{
    printf("Hello world!\n");
	gpio_pad_select_gpio(BLINK_GPIO);
	gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);

        /* Blink off (output low) */
	printf("Turning off the LED\n");
        gpio_set_level(BLINK_GPIO, 0);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
        /* Blink on (output high) */
	printf("Turning on the LED\n");
        gpio_set_level(BLINK_GPIO, 1);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
  
  
    printf("Restarting now.\n");
	fflush(stdout);
	gpio_deep_sleep_hold_en();
	gpio_hold_en(gpio_num_t BLINK_GPIO);
	esp_sleep_enable_timer_wakeup(10000000);
	esp_deep_sleep_start();

   

}
```
Therefore, I would like your help to understand why this is not working and if is possible to make it in arduino IDE.

Thanks in advance!!!

lbernstone, 2019-04-26:
You might need to enable the power on the rtc gpios. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#power-down-of-rtc-peripherals-and-memories

andrelmbraga, 2019-04-26:
I make it works in esp idf using 	
gpio_hold_en(BLINK_GPIO);
gpio_deep_sleep_hold_en();

however i really would like to do that in arduino IDE. Does anyone has a clue?

me-no-dev, 2019-04-26:
the code above will not run in Arduino. what code are you running in Arduino?

andrelmbraga, 2019-04-26:
Actually I'm running a much bigger code in arduino, I ll put here something that you can reproduce the problem. 

`TC_DATA_ATTR int bootCount = 0;
#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  10       /* Time ESP32 will go to sleep (in seconds) */

void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Going HIGH");
  pinMode(GPIO_NUM_25,OUTPUT);
  digitalWrite(GPIO_NUM_25, HIGH);
  delay(2000);
  Serial.println("Going LOW");
  digitalWrite(GPIO_NUM_25, LOW);
  delay(2000);
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This is not going to be called
}`

In this code you can see that the gpio 25 ll be floating in sleep mode, and I want it fix on low level.

me-no-dev, 2019-04-26:
why not add the missing
```cpp
gpio_hold_en(GPIO_NUM_25);
gpio_deep_sleep_hold_en();
```

andrelmbraga, 2019-04-26:
I can use gpio_hold_en() declaring my pin like this #define PIN_LORA GPIO_NUM_25
However the  'gpio_deep_sleep_hold_en'  doesnt work, I dont know if there is such thing in arduino IDE.

The error is:

exit status 1
'gpio_deep_sleep_hold_en' was not declared in this scope

me-no-dev, 2019-04-26:
you are probably missing an include... 
```cpp
#include "esp_sleep.h"
```

andrelmbraga, 2019-04-29:
I try to include this lib, but still doesn't works. 

I found this function **gpio_deep_sleep_hold_en** at:

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/driver/driver/gpio.h

I'm not sure if I can use all these functions on arduino IDE, also I tried to #include "gpio.h" without succeed.  
"C:\Users\andre\Documents\Arduino\Projetos\_07_-_Sleep\_07_-_Sleep.ino:2:18: fatal error: gpio.h: No such file or directory"

This gpio.h can be used at arduino IDE or is just for esp-idf? 
Please help me this is hold me for more than 2 weeks =/.

lbernstone, 2019-04-29:
The code in your last post compiles for me.  The functions in the headers at tools/sdk/include should all be compiled into the esp-idf libraries that are available to arduino-esp32.  So, you should be able to use the exact same code snippets in arduino as in esp-idf.  What version of arduino-esp32 are you using?  You might want to update or uninstall/reinstall to ensure your files are not damaged.

rene6262, 2019-05-09:
Hi, i have the same issues where 'gpio_deep_sleep_hold_en' was not declared in this scope and the eps32 has its libraries updated.  @lbernstone did you compile it on arduino ide or another software?

closed 2019-08-14 by stale[bot]

ClemensGruber, 2020-02-04:
Any solution for this since 2019?

beegee-tokyo, 2020-02-05:
I am doing this (but this is holding them HIGH, not LOW):
```cpp
#include <driver/rtc_io.h>
...
	// Keep the GPIOs on required levels
	rtc_gpio_set_direction((gpio_num_t)PIN_LORA_DIO_1, RTC_GPIO_MODE_INPUT_ONLY);
	rtc_gpio_pulldown_en((gpio_num_t)PIN_LORA_DIO_1);
	rtc_gpio_set_direction((gpio_num_t)PIN_LORA_RESET, RTC_GPIO_MODE_OUTPUT_ONLY);
	rtc_gpio_set_level((gpio_num_t)PIN_LORA_RESET, HIGH);
	rtc_gpio_set_direction((gpio_num_t)PIN_LORA_NSS, RTC_GPIO_MODE_OUTPUT_ONLY);
	rtc_gpio_set_level((gpio_num_t)PIN_LORA_NSS, HIGH);
```
Works on most of the GPIO's. A list of GPIO's can be found [here at page 7](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf). Look for the GPIO's named as RTC_GPIOxx

Above code works for me to keep a LoRa transceiver active while the ESP32 is in deep sleep, waiting for an interrupt coming on DIO1 line from the transceiver.

ClemensGruber, 2020-02-05:
I tried out an code example like 

```
  gpio_hold_en(GPIO_NUM_25);
  gpio_deep_sleep_hold_en();
```

but got the same error message as mentione above yesterday. So I updated the Arduino Core for ESP32 to the latest version and after that the error message was gone!

niallmcauley, 2020-03-11:
Hi I am using an ESP32 and LoRa chip pcb. When I put it into deep sleep I am still drawing 10mA. Could this function be used to prevent pull up resistors from drawing current while in sleep mode?

plusmartin, 2020-10-14:
> I am doing this (but this is holding them HIGH, not LOW):
> 
> ```c++
> #include <driver/rtc_io.h>
> ...
> 	// Keep the GPIOs on required levels
> 	rtc_gpio_set_direction((gpio_num_t)PIN_LORA_DIO_1, RTC_GPIO_MODE_INPUT_ONLY);
> 	rtc_gpio_pulldown_en((gpio_num_t)PIN_LORA_DIO_1);
> 	rtc_gpio_set_direction((gpio_num_t)PIN_LORA_RESET, RTC_GPIO_MODE_OUTPUT_ONLY);
> 	rtc_gpio_set_level((gpio_num_t)PIN_LORA_RESET, HIGH);
> 	rtc_gpio_set_direction((gpio_num_t)PIN_LORA_NSS, RTC_GPIO_MODE_OUTPUT_ONLY);
> 	rtc_gpio_set_level((gpio_num_t)PIN_LORA_NSS, HIGH);
> ```
> 
> Works on most of the GPIO's. A list of GPIO's can be found [here at page 7](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf). Look for the GPIO's named as RTC_GPIOxx
> 
> Above code works for me to keep a LoRa transceiver active while the ESP32 is in deep sleep, waiting for an interrupt coming on DIO1 line from the transceiver.

Hello, were you able to put the ESP32 to sleed and wake up with the LoRa interrupt? Then receive the payload and go back to sleep?

andrewclapham, 2020-12-26:
I'm also using Arduino IDE, and I'm also trying to keep a pin's state held (high, in my case).

I have GPIO16, GPIO17, GPIO18 (which I believe are all non-RTC GPIO pins). They're connected to three P-channel MOSFETs, which power an RGB LED. So my three pins need to be held HIGH while sleeping.

When my ESP32 goes into deep sleep, the RGB LED slowly fades up to full white brightness 😅

I've tried the below, but no luck.

Arduino IDE:
```  
gpio_hold_en((gpio_num_t) 16);
gpio_hold_en((gpio_num_t) 17);
gpio_hold_en((gpio_num_t) 18);
gpio_deep_sleep_hold_en();
delay(10000);
esp_deep_sleep_start();
```

Strangely though, now the blue LED (on GPIO 18) doesn't come on, just the other two (16 and 17) snap to full brightness as soon as I call deep sleep.

Does it matter that they're GPIO vs RTC GPIO?

Is it possible to hold them HIGH during deep sleep? I've also tried INPUT_PULLUP but no luck either.

jessgade, 2021-01-24:
The following code works for me (PlatformIO). But NOT on PIN 17! 

I'm driving a motor via a IRLZ44N MOSFET. The output pin stays HIGH during deepsleep.

I have a 100k resistor connected from the output pin to GND.

`
#include <Arduino.h>
#include "soc/rtc_cntl_reg.h"
#include "soc/rtc.h"
#include "driver/rtc_io.h"
#include "driver/gpio.h"

#define OUTPUT_PIN 25 // NOT ALL PINS WORK. Eks. pin 17 doesent work!
#define LED_PIN 5

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  20        /* Time ESP32 will go to sleep (in seconds) */
#define AWAKE 10 /* Avake in seconds */

// Demonstration of an ESP32 holding a pin HIGH in deepsleep - making it possible to keep a IRLZ44N Mosfet turned on during deepsleep.

void deepSleep() {
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for " + String(TIME_TO_SLEEP) +
  " Seconds");
  Serial.flush();
  esp_deep_sleep_start();
}

void setup() {
Serial.begin(115200);
Serial.println("Started! Set output pin = LOW (and turn on LED on the LOLIN32)");
Serial.flush();      
gpio_hold_dis((gpio_num_t)OUTPUT_PIN);  // Required to turn of the pin (if thats what you want) after deepsleep
pinMode(OUTPUT_PIN, OUTPUT);
pinMode(LED_PIN, OUTPUT);
digitalWrite(OUTPUT_PIN, LOW);
digitalWrite(LED_PIN, LOW);

delay(4000);

Serial.println("Set Pin = HIGH. Then wait 3 seconds before going to deepsleep...");
Serial.flush();
digitalWrite(OUTPUT_PIN, HIGH);
digitalWrite(LED_PIN, HIGH);
gpio_hold_en((gpio_num_t)OUTPUT_PIN); // Hold the high state during deepsleep.

delay(3000);

Serial.println("Sleep now !");
Serial.flush();
deepSleep();
}

void loop() {
  // put your main code here, to run repeatedly:
}
`

ppotter100, 2021-04-06:
@jessgade I have the exact same problems, with also leds on the pins. The behaviour seems quite random. Sometimes the pin is held high, other times it is not. And I even go into light sleep. How did you solve the problem?

ppotter100, 2021-04-06:
I also have an RGB led connected to pin 16, 17, 18. When setting pin 17 high, the state is sometimes held before going into light sleep, other times it is not. It seems quite random. Anyone found a solution for this?

jessgade, 2021-04-06:
@ppotter100 Have you tried using pin 26 or pin 27? 
I specifically had a problem with pin 17

ppotter100, 2021-04-07:
No I haven't, since these are occupied for other functions. But it seems I have resolved the issue by adding    gpio_hold_dis(PIN); when going out of sleep (even in light sleep). When not doing this, the pin would sometimes not be held high in light sleep anymore.

Innocentius-alt, 2022-01-23:
Probably, im late, but i made gpios keep its state `LOW.
`#include "esp_camera.h"
#include <WiFi.h>
#include "esp_timer.h"
#include "img_converters.h"
#include "Arduino.h"
#include "fb_gfx.h"
#include "soc/soc.h" //disable brownout problems
#include "soc/rtc_cntl_reg.h"  //disable brownout problems
#include "esp_http_server.h"
#include "driver/adc.h"
#include "driver/gpio.h"
#include "esp_sleep.h"
#include <PubSubClient.h>

//Replace with your network credentials
const char* ssid = "xxxxx";
const char* password = "xxxxxxx";
const char* mqtt_server = "192.168.x.xx";
const char* HostName = "xxxx";
const char* mqttUser = "xxxx";
const char* mqttPassword = "xxxxt";
#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define MINUTES_TO_SECONDS(x) (60*(x))
#define TIME_TO_SLEEP  MINUTES_TO_SECONDS(1)       /* Time ESP32 will go to sleep (in seconds) */

#define forwardLeft 12
#define backwardLeft 13
#define forwardRight 15
#define backwardRight 14

WiFiClient espClient;
PubSubClient client(espClient);

typedef struct {
  httpd_req_t *req;
  size_t len;
} jpg_chunking_t;

#define PART_BOUNDARY "123456789000000000000987654321"

// This project was tested with the AI Thinker Model, M5STACK PSRAM Model and M5STACK WITHOUT PSRAM
#define CAMERA_MODEL_AI_THINKER
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WITHOUT_PSRAM

// Not tested with this model
//#define CAMERA_MODEL_WROVER_KIT

#if defined(CAMERA_MODEL_WROVER_KIT)
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    21
#define SIOD_GPIO_NUM    26
#define SIOC_GPIO_NUM    27

#define Y9_GPIO_NUM      35
#define Y8_GPIO_NUM      34
#define Y7_GPIO_NUM      39
#define Y6_GPIO_NUM      36
#define Y5_GPIO_NUM      19
#define Y4_GPIO_NUM      18
#define Y3_GPIO_NUM       5
#define Y2_GPIO_NUM       4
#define VSYNC_GPIO_NUM   25
#define HREF_GPIO_NUM    23
#define PCLK_GPIO_NUM    22

#elif defined(CAMERA_MODEL_M5STACK_PSRAM)
#define PWDN_GPIO_NUM     -1
[... 368 lines cut ...]
}
void BACKWARD() {
  digitalWrite(forwardLeft, LOW);
  digitalWrite(forwardRight, LOW);
  digitalWrite(backwardLeft, HIGH);
  digitalWrite(backwardRight, HIGH);
}
void RIGHT() {
  digitalWrite(forwardLeft, HIGH);
  digitalWrite(backwardRight, HIGH); 
  digitalWrite(forwardRight, LOW);
  digitalWrite(backwardLeft, LOW); 
}
void LEFT() {
  digitalWrite(forwardLeft, LOW);
  digitalWrite(backwardRight, LOW); 
  digitalWrite(forwardRight, HIGH);
  digitalWrite(backwardLeft, HIGH); 
}
void STOP() {
  digitalWrite(forwardLeft, LOW);
  digitalWrite(backwardRight, LOW); 
  digitalWrite(forwardRight, LOW);
  digitalWrite(backwardLeft, LOW); 
}`

AliferousWolf, 2023-04-28:
> Probably, im late, but i made gpios keep its state LOW. ESP32 CAM.

I am also late, but thank you for this. I got it working with this code, but I had to add a delay in between setting the pins low. If I didn't have the delay in, it would never work. I put the delay in, and it worked. One probably doesn't need all the libraries I used, or even the tft.enableSleep line....but I'm leaving it in there haha
...libraries:
#include "driver/rtc_io.h"
#include "soc/rtc_cntl_reg.h"
#include "soc/rtc.h"
#include "esp_sleep.h"

//pinMode was set here and configured for the tft screen.

... in setup:

rtc_gpio_hold_dis((gpio_num_t) TFT_LIT);

... in code loop somewhere:

analogWrite(TFT_LIT, 0); // set LIT low (tft backlight)      
delay(100); //must have this display or else the screen backlight will stay on       
rtc_gpio_hold_en((gpio_num_t) TFT_LIT); //hold the TFT_LIT pin at its last value, even during deep sleep
tft.enableSleep(true);        
esp_deep_sleep_start(); //Initiate deep sleep mode