espressif/arduino-esp32 #2714: Cannot make flash while using arduino as an esp-idf component

happynet95, 2019-04-26:
Hi. I'm using Arduino as esp-idf components, and I'm pretty sure that I'd followed this instruction. https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

However, it fails when I make flash on it (Even though I did not change the code in blink.c)
My environment is imac, and here is the result when I type make flash in terminal.

please help.

GENCONFIG
Arduino libraries in build: ArduinoOTA AsyncUDP AzureIoT BLE BluetoothSerial DNSServer EEPROM ESP32 ESPmDNS FFat FS HTTPClient HTTPUpdate NetBIOS Preferences SD SD_MMC SPI SPIFFS SimpleBLE Ticker Update WebServer WiFi WiFiClientSecure Wire
App "blink" version: v4.0-dev-311-g70eda3d22-dirty
jeonghaemins-iMac-3:blink junghaemin$ make flash
Toolchain path: /Users/junghaemin/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/efuse/esp32/esp_efuse_table.o
CC build/bootloader/efuse/src/esp_efuse_api.o
CC build/bootloader/efuse/src/esp_efuse_fields.o
CC build/bootloader/efuse/src/esp_efuse_utility.o
AR build/bootloader/efuse/libefuse.a
LD build/bootloader/bootloader.elf
esptool.py v2.7-dev
Python requirements from /Users/junghaemin/esp/esp-idf/requirements.txt are satisfied.
/Users/junghaemin/esp/esp-idf/components/partition_table/Makefile.projbuild:60: warning: undefined variable `quote'
Building partitions from /Users/junghaemin/esp/esp-idf/examples/get-started/blink/components/arduino/tools/partitions/default.csv...

CC build/app_trace/app_trace.o
CC build/app_trace/app_trace_util.o
CC build/app_trace/heap_trace_tohost.o
CC build/app_trace/host_file_io.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
App "blink" version: v4.0-dev-311-g70eda3d22-dirty
CC build/app_update/esp_app_desc.o
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
Arduino libraries in build: ArduinoOTA AsyncUDP AzureIoT BLE BluetoothSerial DNSServer EEPROM ESP32 ESPmDNS FFat FS HTTPClient HTTPUpdate NetBIOS Preferences SD SD_MMC SPI SPIFFS SimpleBLE Ticker Update WebServer WiFi WiFiClientSecure Wire
CC build/arduino/cores/esp32/libb64/cdecode.o
make[1]: *** No rule to make target `cores/esp32/libb64/cencode.o', needed by `libarduino.a'.  Stop.
make: *** [component-arduino-build] Error 2

me-no-dev, 2019-04-26:
just run make again :)

closed 2019-04-26 by me-no-dev

alexqzd, 2020-06-25:
I have the same issue. Running make again as @me-no-dev suggests indeed works. But why do I have to do that and how can I fix it so I can build it by running make just once?

me-no-dev, 2020-06-26:
you can't. it's an issue in the build system/arduino requirements. Good thing is that it does not exist when using `idf.py build`, but that comes in newer IDF (and soon to arduino)