espressif/arduino-esp32 #2716: error: 'WebServer::_chunked' will be initialized after

rickysiah, 2019-04-26:
Hi, I'm currently building the Arduino as esp-idf components, and I was following the guide from the espressif/arduino-esp32 to build it. But however, I had uncounted the above problem. Please advice and thanks. 

====================================================
happyday@LAPTOP-2F3P2FNV MINGW32 ~/esp/esp-idf/examples/esp-idf-arduino
$ make
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/requirements.txt are satisfied.

App "app-template" version: df62230-dirty
Arduino libraries in build: ArduinoOTA AsyncUDP AzureIoT BLE BluetoothSerial DNSServer EEPROM ESP32 ESPmDNS FFat FS HTTPClient HTTPUpdate NetBIOS Preferences SD SD_MMC SPI SPIFFS SimpleBLE Ticker Update WebServer WiFi WiFiClientSecure Wire
CXX build/arduino/libraries/WebServer/src//WebServer.o
In file included from D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.cpp:29:0:
D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.h: In constructor 'WebServer::WebServer(IPAddress, int)':
D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.h:198:20: error: 'WebServer::_chunked' will be initialized after [-Werror=reorder]
   bool             _chunked;
                    ^
D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.h:169:15: error:   'boolean WebServer::_corsEnabled' [-Werror=reorder]
   boolean     _corsEnabled;
               ^
D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.cpp:41:1: error:   when initialized here [-Werror=reorder]
 WebServer::WebServer(IPAddress addr, int port)
 ^
In file included from D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.cpp:29:0:
D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.h: In constructor 'WebServer::WebServer(int)':
D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.h:198:20: error: 'WebServer::_chunked' will be initialized after [-Werror=reorder]
   bool             _chunked;
                    ^
D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.h:169:15: error:   'boolean WebServer::_corsEnabled' [-Werror=reorder]
   boolean     _corsEnabled;
               ^
D:/Win_ProgramFiles/msys32/home/happyday/esp/esp-idf/examples/esp-idf-arduino/components/arduino/libraries/WebServer/src/WebServer.cpp:62:1: error:   when initialized here [-Werror=reorder]
 WebServer::WebServer(int port)
 ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1plus.exe: some warnings being treated as errors
make[1]: *** [/home/happyday/esp/esp-idf/make/component_wrapper.mk:290: libraries/WebServer/src//WebServer.o] Error 1
make: *** [D:\Win_ProgramFiles\msys32\home\happyday\esp\esp-idf/make/project.mk:582: component-arduino-build] Error 2

lbernstone, 2019-04-26:
Should be fixed in https://github.com/espressif/arduino-esp32/commit/0202ba7c21cceda180217dad5ffa58ba73b58baf

rickysiah, 2019-04-26:
Thanks, it is working!

closed 2019-04-26 by rickysiah