espressif/arduino-esp32 #2717: Import WiFi.h indicates error on WiFiServer.h

shedyhs, 2019-04-26:
What you are trying to do?
Use WiFi.h lib to connect on my wifi
Indicates error in WifiServer.h when i include Wifi lib

### Hardware:
[WiFi.zip](https://github.com/espressif/arduino-esp32/files/3122368/WiFi.zip)

Board:							ESP32 Dev Module ESP32 WROOM
Installation version:			                1.0.0 end i tried 1.2.7
IDE name:						Arduino IDE on version 1.8.8 and 1.8.9
Flash Frequency:					240Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
When i include WiFi.h and try to compile it returned error on lib archives

### Sketch:
```cpp

#include <WiFi.h>

void setup() {
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
Arduino: 1.8.9 (Windows 10), Placa:"ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 80MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

In file included from C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:32:0,

                 from C:\Users\Shedy\Documents\Arduino\wifi_test\wifi_test.ino:1:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiServer.h:37:3: error: invalid abstract return type 'WiFiClient'

   WiFiClient available(uint8_t* status = NULL);

   ^

In file included from C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:31:0,

                 from C:\Users\Shedy\Documents\Arduino\wifi_test\wifi_test.ino:1:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiClient.h:27:7: note:   because the following virtual functions are pure within 'WiFiClient':

 class WiFiClient : public Client {

       ^

In file included from C:\Users\Shedy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:157:0,

                 from sketch\wifi_test.ino.cpp:1:

C:\Users\Shedy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:31:17: note: 	virtual int Client::connect(IPAddress, uint16_t, int)

     virtual int connect(IPAddress ip, uint16_t port, int timeout) =0;

                 ^

C:\Users\Shedy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:32:17: note: 	virtual int Client::connect(const char*, uint16_t, int)

     virtual int connect(const char *host, uint16_t port, int timeout) =0;

                 ^

exit status 1
Erro compilando para a placa ESP32 Dev Module

```

me-no-dev, 2019-04-26:
It's building against the wrong WiFi library: `C:\Program Files (x86)\Arduino\libraries\WiFi\src\WiFi.h`. Also Are you using 1.0.2 or 1.0.0?

shedyhs, 2019-04-26:
Should I use which local to compile the library? documents?

my wifi.h is 1.2.7

me-no-dev, 2019-04-26:
not sure what you are asking to be honest. You have a wrong library in the path I quoted above. ESP32 has it's own WiFi library

shedyhs, 2019-04-26:
thanks man, i tried redownload version 1.0.2 and i resolve this, sry about take our time

closed 2019-04-26 by shedyhs