espressif/arduino-esp32 #2718: BluetoothSerial Restart crash

dobraMorda, 2019-04-27:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:	                1.0.1
IDE name:						Platform.io
Flash Frequency:					240Mhz
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
ESP32 crash after restart BT. When main loop run first time it's ok, but when loop starts again after BT.end() then system crash. 

### Sketch:
```cpp

#include <Arduino.h>
#include "BluetoothSerial.h"

BluetoothSerial BT;

void setup() {
}

void loop() {
    BT.begin("BT test");
    delay(1000);
    BT.end();
    delay(1000);
}
```

### Debug Messages:
```
ASSERT_PARAM(2048 0), in rwbt.c at line 311
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084f21  PS      : 0x00060034  A0      : 0x80089294  A1      : 0x3ffbe310  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000800  A5      : 0x60008054  
A6      : 0x3ffbdd48  A7      : 0x60008050  A8      : 0x80084f21  A9      : 0x3ffbe2f0  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40084e59  LEND    : 0x40084e60  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084f21

Backtrace: 0x40084f21:0x3ffbe310 0x40089291:0x3ffbe330 0x400896a3:0x3ffbe350 0x400813a5:0x3ffbe370 0x4000bfed:0x3ffdf770 0x4008efd5:0x3ffdf780 0x40090d6f:0x3ffdf7a0 0x400913f4:0x3ffdf7c0 0x40084818:0x3ffdf7e0 0x40084abb:0x3ffdf800 0x40084b27:0x3ffdf820 0x400ef971:0x3ffdf870 0x400d59f9:0x3ffdf890 0x400d5acd:0x3ffdf8b0 0x401173fd:0x3ffdf8d0 0x40117410:0x3ffdf8f0 0x400d6ecb:0x3ffdf910 0x40117119:0x3ffdf930 0x4008d735:0x3ffdf960

Core 1 register dump:
PC      : 0x40184f56  PS      : 0x00060f34  A0      : 0x8014519a  A1      : 0x3ffbc590  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060a20  A7      : 0x00000000  A8      : 0x00060023  A9      : 0x3ffbc104  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x8008d948  A13     : 0x3ffc8e40  
A14     : 0x00000000  A15     : 0x3ffbc280  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40184f56:0x3ffbc590 0x40145197:0x3ffbc5b0 0x4008f64d:0x3ffbc5d0 0x4008d735:0x3ffbc5f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
 
```

dobraMorda, 2019-04-29:
Anybody can help me?

DQOmni, 2019-05-10:
Reported that two months ago (#2601), also got no useful reply. My solution was to call 'begin' only once and only call 'end' before reseting. To force a disconnect, use 'esp_spp_disconnect'. If you want to change the device name, easiest way is to reset the ESP32.

closed 2019-08-14 by stale[bot]

tito, 2019-10-26:
Could you reopen? This is still here and we hit the issue as well easily.

rgot-org, 2019-11-10:
I had the same issue. this is because the BT state stay ESP_BT_CONTROLLER_STATUS_INITED after the end() method. 
i have modified the **esp32-hal-bt.c** file 
line 57 and next
(i have insert the esp_bt_controller_deinit() function so the controller go into Idle state)
```c++
bool btStop(){
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){
		log_i("bt stopped");
        return true;
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
		log_i("bt enabled");
        if (esp_bt_controller_disable()) {
            log_e("BT Disable failed");
            return false;
        }
        while(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED);
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED){
		log_i("inited");
		if (esp_bt_controller_deinit()) {
			log_e("BT deint failed");
			return false;
		}
		while (esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED);
        return true;
    }
    log_e("BT Stop failed");
    return false;
}
```

Saph123, 2019-12-09:
> 
> 
> I had the same issue. this is because the BT controller remains in state ESP_BT_CONTROLLER_STATUS_INITED instead of state ESP_BT_CONTROLLER_STATUS_IDLE after the end() method.
> in file esp_bt.h it is specified
> 
> > @brief Enable BT controller.
> > Due to a known issue, you cannot call esp_bt_controller_enable() a second time
> > to change the controller mode dynamically. To change controller mode, call
> > esp_bt_controller_disable() and then call esp_bt_controller_enable() with the new mode.
> 
> after **esp_bt_controller_disable()** the controller remains in state INITED so we do call the **esp_bt_controller_deinit()** function to put the controller into state IDLE.
> 
> i have modified the **esp32-hal-bt.c** file
> line 57 and next
> (i have insert the esp_bt_controller_deinit() function so the controller go into Idle state)
> 
> ```c++
> bool btStop(){
>     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){
> 		log_i("bt stopped");
>         return true;
>     }
>     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
> 		log_i("bt enabled");
>         if (esp_bt_controller_disable()) {
>             log_e("BT Disable failed");
>             return false;
>         }
>         while(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED);
>     }
>     if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED){
> 		log_i("inited");
> 		if (esp_bt_controller_deinit()) {
> 			log_e("BT deint failed");
> 			return false;
> 		}
> 		while (esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED);
>         return true;
>     }
>     log_e("BT Stop failed");
>     return false;
> }
> ```

Wow thanks a lot! That worked for me!

zimiao815, 2020-03-18:
@Saph123 
I also modified the corresponding code. I found that no code didn't work. What should I do？

zimiao815, 2020-03-18:
@Saph123   I got it
This code is great   Thank you
but  feed  watchdog  is  5  second。