espressif/arduino-esp32 #2719: Preferences crashes ESP32 when same value is overwritten

malebuffy, 2019-04-27:
I am using the ESP32 with both the IRremote Library and the EEPROM and Preferences Library.

When I use EEPROM I can overwrite data on the same addresses, but only when disbaling the IRRemote receiver. 

With Preferences I also disable the IRreceiver but can only use a Preference name one time. When I try to update the value of the Preference I get a crash.

For instance
`  irrecv.enableIRIn(false); // Stop the receiver
    preferences.putString("Device200", "TEST2");
    irrecv.enableIRIn(true); // Start the receiver
    String traps = preferences.getString("Device200", "");
    Serial.println(traps);`

This works one time and Prints out Device200 in the Serial Tewrminal, but when this part of the code is triggered again, it crashes with the error message:

`13:25:11.390 -> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
13:25:11.390 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:25:11.390 -> Memory dump at 0x401a6dcc: bad00bad bad00bad bad00bad
13:25:11.390 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:25:11.424 -> Memory dump at 0x401a6dcc: bad00bad bad00bad bad00bad
13:25:11.424 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:25:11.424 -> Memory dump at 0x401a6dcc: bad00bad bad00bad bad00bad
13:25:11.424 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:25:11.424 -> Memory dump at 0x401a6dcc: bad00bad bad00bad bad00bad
13:25:11.458 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:25:11.458 -> Memory dump at 0x401a6dcc: bad00bad bad00bad bad00bad
13:25:11.458 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:25:11.458 -> Memory dump at 0x401a6dcc: bad00bad bad00bad bad00bad
13:25:11.458 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:25:11.458 -> Memory dump at 0x401a6dcc: bad00bad bad00bad bad00bad
`
If I change the name of the preference to Device201, it works again until I try to update it and I get the same error message. Any ideas?

atanisoft, 2019-04-27:
This is likely a bug in the IRremote library, it likely uses an ISR in an incorrect manner. When the ISR is triggered any access to flash it data that is not in IRAM/DRAM will cause this sort of failure.

malebuffy, 2019-04-27:
Thanks that's probably true since I had the exact same problem with EEPROM writing. However I thought I fixed it buy modifying the IRremote library to be able to disable it like in the code. However, for some reason it doesn't work for Preferences. Will have to find out why...

atanisoft, 2019-04-27:
Both preferences and eeprom use the same underlying storage type. They will both fail in the same manner when and ISR is running.

closed 2019-08-14 by stale[bot]

helestrange, 2020-06-16:
For those interested, I experienced this exact issue, but found this IR library to be friendlier with the Preferences library (on the ESP32 at least):
https://github.com/crankyoldgit/IRremoteESP8266