espressif/arduino-esp32 #2720: I2c doesn't work, when SPI is enabled.

Flo082002, 2019-04-27:
### Hardware:
Board:							Dev Module
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                         Windows 10

I have a problem with when I use i2c and SPI together.
The i2c comunication only works when I us no SPI.

```

#include "SPI.h"
#include <Wire.h>

#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Wire.begin();
  Wire.beginTransmission(0x60);
  Serial.println(String(Wire.endTransmission()) + " Error");
  delay(200);
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);

}

void loop() {
  // put your main code here, to run repeatedly:
  Wire.beginTransmission(0x60);
  Serial.println(String(Wire.endTransmission()) + " Error");
  delay(1000);
}
```

The first i2c transmition works fine.
But after enabeling SPI and the i2c is broken and `Serial.println(String(Wire.endTransmission()) + " Error");` returns 5.

me-no-dev, 2019-04-27:
they have nothing in common and I am using them together just fine. did you try without calling SPI.begin?

Flo082002, 2019-04-27:
> did you try without calling SPI.begin?

Then I have no error, but the Spi is not working.
In my actual code I use this library vor the vs1053 : https://github.com/baldram/ESP_VS1053_Library
I get this error and the esp restarts:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dd544  PS      : 0x00060d30  A0      : 0x800d5e34  A1      : 0x3ffd2820  
A2      : 0x00000000  A3      : 0x3ffc181e  A4      : 0x00000020  A5      : 0x00000040  
A6      : 0x00000010  A7      : 0xdb4593f0  A8      : 0x00000008  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00241001  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0xff000000  A15     : 0x00000001  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400dd5f8  LEND    : 0x400dd65a  LCOUNT  : 0x00000002  

Backtrace: 0x400dd544:0x3ffd2820 0x400d5e31:0x3ffd2840 0x400d5ff5:0x3ffd2860 0x400d9f76:0x3ffd2880 0x400d21e6:0x3ffd28d0 0x4008917d:0x3ffd28f0

Rebooting...
```

stickbreaker, 2019-04-27:
@Flo082002 What are the values of your pullup resistors?

Try this code to display the error message:
```c++
Wire.beginTransmission(0x60);
uint8_t err = Wire.endTransmission();
Serial.printf("endTransmission() returned status as: %d  %s\n",err, Wire.getErrorText(err));
```

Error 5 is Busy, Which means that either SCL or SDA is being held low by something.  If you are using ESP32 Dev module, SCL and SDA are using pins:
```
static const uint8_t SDA = 21;
static const uint8_t SCL = 22;
```
from [pins_arduino.h](https://github.com/espressif/arduino-esp32/blob/master/variants/esp32/pins_arduino.h)
Chuck.

Flo082002, 2019-04-27:
> Try this code to display the error message:

`endTransmission() returned status as: 5  BUSY`

> static const uint8_t SDA = 21;
> static const uint8_t SCL = 22;

I use these pins. And they are not pulled low. I measure 3.3V when I send nothing.

Flo082002, 2019-04-27:
When I call `Wire.begin();` after `Spi.begin();` I get error 1

stickbreaker, 2019-04-27:
Set CORE DEBUG LEVEL to Verbose, recompile and lets look at the error reports.

Flo082002, 2019-04-28:
Core Debug Level : debug
```

[D][esp32-hal-i2c.c:1336] i2cProcQueue():  Busy Timeout start=0xa5c, end=0xa8e, =50, max=50 error=1
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbecdc
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb864c
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb86d0 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8700
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffafeac
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=1, scl(22)=0
[D][esp32-hal-i2c.c:1419] i2cCheckLineState(): Recovered after 1 Cycles
[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0

```

stickbreaker, 2019-04-28:
>[D][esp32-hal-i2c.c:1336] i2cProcQueue():  Busy Timeout start=0xa5c, end=0xa8e, =50, max=50 error=1
...
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=1, scl(22)=0
[D][esp32-hal-i2c.c:1419] i2cCheckLineState(): Recovered after 1 Cycles
[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0

These are telling you the I2C hardware is seeing activity on the I2C bus that it did not initiate.  Something it holding SCL low.  I would look for a hardware fault. Try using different pins for the I2C bus.

`Wire.begin( int SDAPin, int SCLPin, uint32_t frequency);`

SDA and SCL can be any Input/output pin, remember pins 34..39 are input only.Try 25,26,27,32,33

Chuck.

chegewara, 2019-04-28:
For sure try to use pullup resistors. I have max30100 sensor which have its own pullup resistors, but they are too weak. I had to use additional resistors to make it works with esp32 on arduino ide.

Flo082002, 2019-04-28:
> For sure try to use pullup resistors.

I use 2,2kΩ pull ups.
I will test other pins now.

Flo082002, 2019-04-29:
Thanks a lot for your help. I don't know what happened to pin 22.
Now I use pin 15 as scl and everything works fine. 👍

Flo S

closed 2019-05-02 by Flo082002