espressif/arduino-esp32 #2721: Incorrect Content Length on browser while using send_P function of WebServer

ayushsharma82, 2019-04-28:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.0
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     yes
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
I am trying to send large webpage (around 55kb ~ 110kb) via PROGMEM of ESP32 using `send_P` function of WebServer. WebServer's `send_P` function doesn't act properly, It is always sending incorrect content length when trying to send large files via progmem. 

### Sketch: 
Upload this branch of my Library. You will reproduce the 'CONTENT_LENGTH_MISMATCH' error on browser when trying to access `/update` url. https://github.com/ayushsharma82/ElegantOTA/tree/esp32-dev 

### Debug Messages:
```
CONTENT_MISMATCH_ERROR on browser
```

me-no-dev, 2019-04-28:
ESP32 does not have PROGMEM. It's all aliases, therefore you do not need to use the _P functions at all. Just use the regular ones.

closed 2019-04-28 by me-no-dev

ayushsharma82, 2019-04-28:
@me-no-dev  Okay. I will try `send` function.