espressif/arduino-esp32 #2722: How to disable touch pad interrupts?

baoshuai9527, 2019-04-28:
hi，Open the touch function can wake up the device io，how to disable touch pad interrupts?I need a function like detachInterrupt.
----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			last git
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 7

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <Arduino.h>

#define PIN_TOUCH1 12
#define PIN_TOUCH2 13
#define PIN_TOUCH3 14
#define PIN_TOUCH4 15

int threshold = 40;

void start_sleep()
{
  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_sleep_enable_touchpad_wakeup();
  esp_deep_sleep_start();
}

void touch1_cb()
{
  Serial.println("touch1_cb");
}

void touch2_cb()
{
  Serial.println("touch2_cb");
}

void touch3_cb()
{
  Serial.println("touch3_cb");
}

void touch4_cb()
{
  Serial.println("touch4_cb");
}

void print_wakeup_reason()
{
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason)
  {
  case ESP_SLEEP_WAKEUP_EXT0 :
    Serial.println("Wakeup caused by external signal using RTC_IO");
    break;
  case ESP_SLEEP_WAKEUP_EXT1 :
    Serial.println("Wakeup caused by external signal using RTC_CNTL");
    break;
  case ESP_SLEEP_WAKEUP_TIMER :
    Serial.println("Wakeup caused by timer");
    break;
  case ESP_SLEEP_WAKEUP_TOUCHPAD :
    Serial.println("Wakeup caused by touchpad");
    break;
  case ESP_SLEEP_WAKEUP_ULP :
    Serial.println("Wakeup caused by ULP program");
    break;
  default :
    Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason);
    break;
  }
}

void print_wakeup_touchpad() {
  touch_pad_t touchPin;

  touchPin = esp_sleep_get_touchpad_wakeup_status();

  switch (touchPin)
  {
  case 0  : Serial.println("Touch detected on GPIO 4"); break;
  case 1  : Serial.println("Touch detected on GPIO 0"); break;
  case 2  : Serial.println("Touch detected on GPIO 2"); break;
  case 3  : Serial.println("Touch detected on GPIO 15"); break;
  case 4  : Serial.println("Touch detected on GPIO 13"); break;
  case 5  : Serial.println("Touch detected on GPIO 12"); break;
  case 6  : Serial.println("Touch detected on GPIO 14"); break;
  case 7  : Serial.println("Touch detected on GPIO 27"); break;
  case 8  : Serial.println("Touch detected on GPIO 33"); break;
  case 9  : Serial.println("Touch detected on GPIO 32"); break;
  default : Serial.println("Wakeup not by touchpad"); break;
  }
}

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println("Starting...");

  print_wakeup_reason();
  print_wakeup_touchpad();

  touchAttachInterrupt(PIN_TOUCH1, touch1_cb, threshold);
  touchAttachInterrupt(PIN_TOUCH2, touch2_cb, threshold);
  touchAttachInterrupt(PIN_TOUCH3, touch3_cb, threshold);
  touchAttachInterrupt(PIN_TOUCH4, touch4_cb, threshold);

  // pinMode(PIN_TOUCH1, OUTPUT);
  // pinMode(PIN_TOUCH2, OUTPUT);
  // pinMode(PIN_TOUCH3, OUTPUT);
  // touch_pad_intr_disable();
  // touch_pad_clear_status();
  start_sleep();
}

void loop()
{

}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4

Starting...
Wakeup was not caused by deep sleep: 0
Wakeup not by touchpad
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4

Starting...
Wakeup caused by touchpad
Touch detected on GPIO 12
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4

Starting...
Wakeup caused by touchpad
Touch detected on GPIO 14
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4

Starting...
Wakeup caused by touchpad
Touch detected on GPIO 12
Going to sleep now
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4

Starting...
Wakeup caused by touchpad
Touch detected on GPIO 13
Going to sleep now
```

Jeroen88, 2019-04-28:
Is [this](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/touch_pad.html) of any help?

rudsomlima, 2019-06-25:
touch_pad_intr_disable( );

[https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/touch_pad.html#_CPPv422touch_pad_intr_disablev](url)

PR #6194 (2022-02-03): Touch Sensor IDF Refactoring
## Summary
Touch Sensor refactoring to be based on IDF 4.4.
Adds support to ESP32S2 and future ESP32S3.

#### Adds some new APIs:
#### For all chips:
```void touchAttachInterruptArg(uint8_t pin, void (*userFunc)(void*), void *arg, uint32_t threshold);```
This function allows the user to add and pass a void* parameter to the ISR user callback.

```void touchDetachInterrupt(uint8_t pin);```
This function detaches ISR call back for the touch pad pin.

#### Only ESP32 chip
```void touchInterruptSetThresholdDirection(bool mustbeLower);```
This function allows the user t [... cut]
commit 0ea485e5 (2022-02-03): Touch Sensor IDF Refactoring (#6194)

Summary

Touch Sensor refactoring to be based on IDF 4.4.
Adds support to ESP32S2 and future ESP32S3.

Adds some new APIs:

For all chips:

void touchAttachInterruptArg(uint8_t pin, void (*userFunc)(void*), void *arg, uint32_t threshold);
This function allows the user to add and pass a void* parameter to the ISR user callback.

void touchDetachInterrupt(uint8_t pin);
This function detaches ISR call back for the touch pad pin.

Only ESP32 chip

void touchInterruptSetThresholdDirection(bool mustbeLower);
This function allows the user to set if the ISR callba [... cut]
closed 2022-02-03 by me-no-dev