espressif/arduino-esp32 #2725: #ifdef / #elif not working as expected

bill-orange, 2019-04-29:
I am programming an ESP32 using the Arduino IDE with a fairly lengthy sketch .  I have a second version of the sketch written to run on an EPS8266.  maintaining two versions of the same sketch is awkward to say the least.  I decided it was probably best to merge the two sketches by defining the processor and using #ifdef / #elif, etc throughout the code.

I can not get it to behave as I expected.  I don't know if this is bug or I am doing something wrong that should be obvious. 

```
#define ESP32

#ifdef ESP8266
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266Ping.h>
#elif define ESP32
#include <WiFi.h>
#include "SPIFFS.h"
#include <HTTPClient.h>
#include  "my_esp32_ping.h"
#endif
```
Compiler errors indicate that the include statements under the #elif were not included.

Thoughts?

stickbreaker, 2019-04-29:
@bill-orange are you sure `ESP8266` is not defined?
```
#undef ESP8622
```
Just because you defined `ESP32` doesn't mean `ESP8266` wasn't defined somewhere else.

Chuck.

bill-orange, 2019-04-29:
I added `#undef ESP8266` and that seemed to do the trick, so far.  So, the compiler will pick up definitions from libraries in #include?

stickbreaker, 2019-04-29:
Define can also come from cmdline.

me-no-dev, 2019-04-29:
esp32 definitely does not define 8266 anywhere. You failed to properly fill the issue form, so I have no idea what versions and software you are running.

bill-orange, 2019-04-29:
@me-no-dev 

No worries, @stickbreaker answered my question and all is well now.  I was trying to use #ifdef to define chipset and unbeknownst to be one one of the libraries I was including defined ESP8266 causing my ESP32 code not to compile. Adding `#undef ESP8266` to my ESP32 code fixed the problem.  Complicating the issue, I did not know that `#undef` existed.   I think on this basis I can close this now.

closed 2019-04-29 by bill-orange