espressif/arduino-esp32 #2729: Unable to exit bootloader mode on ESP32

mimansamaheshwari, 2019-05-01:
Hardware:
Board:							        ESP32 Dev Module
IDE name:							Visual Studio Code
Flash Frequency:					        80Mhz
PSRAM enabled:                                              no
Upload Speed:						115200
Computer OS:                                                  Windows 

I am trying to upload a getChipId.ino file example file to ESP32 from Terminal of VSCode. I succeded in doing that but as soon as I reset my board it goes in Boot mode. I get the following message on my terminal:

`rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))waiting for download`
Can anyone explain what is the possible issue?

What I have done is installed Node.js support on Visual Studio Code and made two .js scripts one for uploading code and one for reading data of Serial from ESP32. I have used "serialport" dependency to read data from serial. Both the scripts are  as stated below :

# Code Upload
```
const path = require('path');
const spawn = require('child_process').spawn;
const tmpdir = require('os').tmpdir();

const arduinoHome = path.join('C:','Program Files (x86)', 'Arduino');
const espressif = path.resolve(arduinoHome, 'hardware', 'espressif');
const buildPath = path.resolve(tmpdir); 

const uploadArgs = {
    'esp32': [
        '--chip', 'esp32',
        '-p', 'COM9',
        '--baud', '921600',
        '--before', 'default_reset',
        '--after', 'hard_reset',
        'write_flash', '-z',
        '--flash_mode', 'dio',
        '--flash_freq', '80m',
        '--flash_size', 'detect',
        '0xe000', path.join(espressif, 'esp32', 'tools', 'partitions', 'boot_app0.bin'),
        '0x1000', path.join(espressif, 'esp32', 'tools', 'sdk', 'bin', 'bootloader_qio_80m.bin'),
        '0x10000', path.join(buildPath, 'getchip.ino.bin'),
        '0x8000', path.join(buildPath, 'getchip.ino.partitions.bin')
    ]
};

const compileCode = () => {
    const build = spawn(path.join(arduinoHome, 'arduino-builder'), [
        '-compile',
        '-logger=machine',
        '-hardware', path.join(arduinoHome, 'hardware'),
        '-tools', path.join(arduinoHome, 'tools-builder'),
        '-tools', path.join(arduinoHome, 'hardware', 'tools', 'avr'),
        '-built-in-libraries', path.join(arduinoHome, 'libraries'),
        '-build-path', buildPath,
        '-fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none',
        '-prefs=build.warn_data_percentage=75',
        '-verbose',
        path.join(__dirname, 'getchip.ino')
    ]);
    build.stdout.on('data', (rawdata) => {
        console.log(`stdout: ${rawdata.toString()}`);
    });

    build.stderr.on('data', (data) => {
        console.log(`err: ${data.toString()}`);
    });

    build.on('close', (code) => {
        console.log(code);
        if(code === 0) {
            upload();
        } else {
            console.log('Error');
        }
    });
};

const upload = () => {
    const build = spawn(`${path.resolve(espressif, 'esp32', 'tools', 'esptool', 'esptool.exe')}`, uploadArgs['esp32']);
    build.stdout.on('data', (data) => {
        console.log(data.toString());
    });

    build.stderr.on('data', (data) => {
        console.log(data.toString());
    });

    build.on('close', (code) => {
        console.log(`child process exited with code ${code}`);
    });
};

compileCode();

```
# Reading Data From Serial
```
const SerialPort = require('serialport');

const port = new SerialPort('COM9', {
  baudRate: 115200
});

port.on('data', function (data) {
    console.log(data.toString());
});
```
Here the code upload process works fine but as soon as I run serialport script and reset my board the module goes in the boot mode. And if I close this application and open Serial Monitor in Arduino IDE things work fine.

atanisoft, 2019-05-01:
The problem is likely gpio 0 and 2 are being pulled to GND during startup.

Also since you are using vscode, you might want to try the PlatformIO extension for it as it implements the full build stack for arduino.

mimansamaheshwari, 2019-05-02:
Yes I observed that gpio 0  is remaining LOW hence I am facing this issue.  Actually what I have done is a part for sub process that I am working on for making software so I have to understand what is the issue in the above mentioned method. Is there any change that Serialport dependency used is creating issue? Because I am able to upload the code successfully.

atanisoft, 2019-05-02:
I don't know for certain but it could be playing a factor if it interacts with CTS/RTS signals.

mimansamaheshwari, 2019-05-03:
When I ran same script on Linux things were working fine. I think problem is related to dependency only. Now how to find that out is an issue though.

mimansamaheshwari, 2019-05-08:
I observed that after uploading code and resetting it again and again I get stuck with the following message:
`rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))waiting for download`

But on disconnecting from here if I open port on Serial Monitor in Arduino IDE and perform the hardware reset operation
I get following data:

ets Jun  8 2016 00:22:57

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9400
ho 0 tail 12 room 4
load:0x40080400,len:5928
entry 0x40080698
```
However if I upload same 'GetChip' example code from Arduino IDE I get the normal reset data:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10208
load:0x40080400,len:6460
entry 0x400806a4
ESP32 Chip ID = F4E46BA4AE30
ESP32 Chip ID = F4E46BA4AE30
ESP32 Chip ID = F4E46BA4AE30
```

chegewara, 2019-05-08:
Hi, recently i had some similar very weird behavior with my esp32 boards. With help of @me-no-dev i figured out its putty causing all my problems. With putty my board behaved very very strange, but with arduino-ide serial monitor it worked as expected.
I dont know why but some settings in my putty are not correct and does something not expected on CTS/RTS lines.

mimansamaheshwari, 2019-05-09:
I also tried on Putty however, I didn't face any issue there. I think the problem is because of the serial port dependency used here. But don't know if there is an alternate solution.

mimansamaheshwari, 2019-05-17:
I tried to the same procedure of using Serial port dependency on Linux and Mac as well. I found no issue in resetting board there. But the problem on Windows is still the same. Any idea why?

mimansamaheshwari, 2019-05-22:
Any hints on how to proceed further? What type of analysis should I do to understand the root cause of the problem?

closed 2019-05-30 by mimansamaheshwari

Viktorlvkl, 2019-11-05:
> I observed that after uploading code and resetting it again and again I get stuck with the following message:
> `rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))waiting for download`
> 
> But on disconnecting from here if I open port on Serial Monitor in Arduino IDE and perform the hardware reset operation
> I get following data:
> 
> ets Jun 8 2016 00:22:57
> 
> ```
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:928
> ho 0 tail 12 room 4
> load:0x40078000,len:9400
> ho 0 tail 12 room 4
> load:0x40080400,len:5928
> entry 0x40080698
> ```
> 
> However if I upload same 'GetChip' example code from Arduino IDE I get the normal reset data:
> 
> ```
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1100
> load:0x40078000,len:10208
> load:0x40080400,len:6460
> entry 0x400806a4
> ESP32 Chip ID = F4E46BA4AE30
> ESP32 Chip ID = F4E46BA4AE30
> ESP32 Chip ID = F4E46BA4AE30
> ```
> 
> Any idea what issue can be?

Try change Flash Frequency: from 80Mhz to 40Mhz

sonpython, 2024-01-01:
same issue here

 after trying many firmware not working (esphome, micropython, adruino...)

Jut stuck at

waiting for download

Darren777, 2025-05-22:
I found 2 solutions to the problem which seems to be cause by a small amount of code being regarded as no code by the esp32 so it just stays in boot mode thinking that it has no code to run.

Firstly, holding the boot button on the ESP32 down prior to connecting the USB cable, and then only releasing it once the sketch starts uploading. this solved the problem.

Secondly, this solution only works in Arduino IDE, before uploading the sketch go to the upload options and choose the "large app" option