espressif/arduino-esp32 #2730: BLE max connections in arduino IDE : Why is it limited to 3?

happynet95, 2019-05-01:
Hello. I'm trying to connect four BLE Servers to a single BLE Client.
I know that max BLE connections are limited to 3, because of esp-idf's default settings.
But it still doesnt work even though I changed the settings of [/arduino-esp32/tools/sdk/sdkconfig.
](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig)
Is anything other than sdkconfig file I should consider? Does Arduino BLE library itself have a limitation?

> 
> #
> # Bluetooth
> #
> CONFIG_BT_ENABLED=y
> 
> #
> # Bluetooth controller
> #
> CONFIG_BTDM_CONTROLLER_MODE_BLE_ONLY=y
> CONFIG_BTDM_CONTROLLER_MODE_BR_EDR_ONLY=
> CONFIG_BTDM_CONTROLLER_MODE_BTDM=
> CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN=4
> CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF=4
> CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_EFF=0
> CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_EFF=0
> CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_0=y
> CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_1=
> CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
> CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI=y
> CONFIG_BTDM_CONTROLLER_HCI_MODE_UART_H4=
>

lbernstone, 2019-05-01:
The sdkconfig is only used when a release is created from the esp-idf sources.  If you want to change the core libraries, you will need to [compile your own libraries](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md)

happynet95, 2019-05-02:
@lbernstone thank you!

closed 2019-05-03 by happynet95

GianCann, 2020-02-26:
@lbernstone i tryed this method (with build successfull result)
https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md  but I am not clear how to replace the original libraries with those compiled by me.
Can you explain how to do? I'm using Arduino on Windows 10.