espressif/arduino-esp32 #2731: Face Detection in CameraWebServer Example Not Working in 1.0.2

xgarb, 2019-05-01:
Faces aren't detected in the CameraWebServer example in 1.0.2. I tested with an Ai-Thinker and an ESP-EYE and there is nothing in the serial monitor or in the stream.

Works great when I rolled back to 1.0.1 with the previous example.

crackpot1977, 2019-05-10:
Hi.
Not your solution, but:
I have the same problem :-(

sconwayaus, 2019-05-11:
I was able to repeat the OP issue.

Hardware: AI-Thinker ESP32-CAM
Built the CameraWebServer example in the Ardunio IDE using 1.0.2 and 1.0.1.
1.0.2, face detection was not working. 
1.0.1, face detection working

me-no-dev, 2019-05-13:
Hey, can you please install 1.0.3-rc1 from the development package manager link and give it a try?

crackpot1977, 2019-05-13:
Hi.
Unfortunately no solution for me.
1.0.1 => working
1.0.2 & 1.0.3-rc1 => not working

eggnot, 2019-05-18:
[image]

1.0.1 - working
1.0.2 - not working

MohamadBay, 2019-05-24:
ESP-eye  Face detection and recognition is not working  
1.0.1 not working
1.0.2 not working
a other member has the same Problem : https://github.com/espressif/arduino-esp32/issues/2817

any idea??

rikki78, 2019-07-27:
Same here. I have an AI thinker board. Is it possible to use that with the previous release 1.0.1 somehow? I'm using platformIO.

rikki78, 2019-07-28:
I've got it to work by switching to Arduino IDE 1.0.1. The AI thinker board is not supported, but the ESP32 Wrover module also works

andytion, 2019-08-05:
hey,guts,i think i figure out how to fix this.
in the old version mtmn_config init code is something like

`mtmn_config.min_face = 80;
		mtmn_config.pyramid = 0.8;
		mtmn_config.p_threshold.score = 0.1;
		mtmn_config.p_threshold.nms = 0.9;
		mtmn_config.r_threshold.candidate_number = 10;
		mtmn_config.r_threshold.score = 0.2;
		mtmn_config.r_threshold.nms = 0.8;
		mtmn_config.r_threshold.candidate_number = 5;
		mtmn_config.o_threshold.score = 0.3;
		mtmn_config.o_threshold.nms = 0.7;
		mtmn_config.o_threshold.candidate_number = 3;`

and in the new version.there is some new values had been add in the struct mtmn_config_t.
and a new function.
`mtmn_init_config()`

and i use this function to init my mtmn_config.  face_detect() work finaly.

omathurin, 2019-08-10:
I confirm andytion's solution worked for me: I'm on 1.0.2 and I just added a call to mtmn_init_config() to initialize the mtmn_config variable in the file _app_httpd.cpp_:
`mtmn_config = mtmn_init_config()`

And I managed to see faces detected in the stream (green rectangle around the face) - as expected.

[image]

privet971, 2019-08-10:
Thanks to andytion and omathurin..... face detection is ok even in 1.0.3rc1

rvdbreemen, 2019-08-28:
First big thanks, that fix by andytion works for me too. 

But then after flashing and running the software, the face recoginition at least is happening. Then just like privet971, the same thing happens here, after a few frame it just died with a corrupt heap error. 

Logs:
..
WiFi connected
Starting web server on port: '80'
Starting stream server on port: '81'
Camera Ready! Use 'http://192.168.88.35' to connect
MJPG: 9291B 305ms (3.3fps), AVG: 305ms (3.3fps), 136+91+0+0=228 0
MJPG: 9316B 244ms (4.1fps), AVG: 274ms (3.6fps), 136+88+0+0=224 0
MJPG: 9333B 250ms (4.0fps), AVG: 266ms (3.8fps), 139+89+0+0=229 0
MJPG: 9323B 285ms (3.5fps), AVG: 271ms (3.7fps), 138+129+0+0=268 0
MJPG: 9298B 297ms (3.4fps), AVG: 276ms (3.6fps), 139+129+0+0=269 0
CORRUPT HEAP: Bad head at 0x3ffded74. Expected 0xabba1234 got 0x00000000
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400dd6cb on core 1

Backtrace: 0x40091b44:0x3ffdc8d0 0x40091d75:0x3ffdc8f0 0x400dd6cb:0x3ffdc910 0x400917b9:0x3ffdc940 0x4008840a:0x3ffdc960 0x400899b5:0x3ffdc980 0x4000bec7:0x3ffdc9a0 0x400e9bb1:0x3ffdc9c0 0x400d2f96:0x3ffdc9e0 0x4012b445:0x3ffdcb90 0x4012a699:0x3ffdcbc0 0x4012a775:0x3ffdcc50 0x4012ab80:0x3ffdcc70 0x40129cb6:0x3ffdcc90 0x40129d08:0x3ffdccd0 0x4008dae1:0x3ffdccf0

Rebooting...

But it just rebooted, and then it toke samples again and it started working. There must be something going on, it happened after I tried to add my first face (myself) and then it .

Now it's been running for a few minutes without any issue. Just noticed that the ESP is getting quite hot, maybe running it on 3.3V is better, but it initially would not work at that voltage, so I switched to 5V.

Anyway, just wanted to let you all know!

rvdbreemen, 2019-08-28:
And there it is again:
CORRUPT HEAP: Bad tail at 0x3ffdfcc4. Expected 0xbaad5678 got 0x00000000
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400dd6cb on core 0

Backtrace: 0x40091b44:0x3ffd41d0 0x40091d75:0x3ffd41f0 0x400dd6cb:0x3ffd4210 0x400917b9:0x3ffd4240 0x4008840a:0x3ffd4260 0x400899b5:0x3ffd4280 0x4000bec7:0x3ffd42a0 0x4008e96b:0x3ffd42c0 0x4010dcb5:0x3ffd42e0 0x4010f499:0x3ffd4300 0x4010fec3:0x3ffd4330 0x400fd379:0x3ffd4350 0x4008dae1:0x3ffd4380

Rebooting...

So the code still has bugs that needs killing ;-)

andytion, 2019-08-29:
That happened to me,too. First of all, i suspect is something wrong in my code? But after i debug and test around for a while. I doubt that is there something wrong in the official code? And i try the lastest master branch. After that the annouying multi_heap_free abort() gone finaly.You guys can give it a try~~;-P

pfeerick, 2019-10-13:
Both face detection and face recognition work fine for me with the provided `CameraWebServer` using 1.0.4 of this core on both the Arduino IDE and PlatformIO... so perhaps this has now actually been fixed?

I'm using an [AI-Thinker ESP32-CAM](https://www.aliexpress.com/item/4000000845941.html), so my configuration was as follows:

In the main sketch, set my WiFI SSID and Password, comment out `//#define CAMERA_MODEL_WROVER_KIT` and uncomment `#define CAMERA_MODEL_AI_THINKER`

On the Arduino IDE, I have the board configured as an `ESP32 Wrover Module` with a `HUGE APP (3MB NO OTA/1MB SPIFFs)` partition scheme. Everything else at defaults. On PlatformIO, I used the `esp32-cam` board type which worked as-is. 

Unless faults can be reproduced against the 1.0.4 core, with the stock example (with configuration changes only), hopefully this and the related #2817 can be closed now.

me-no-dev, 2019-10-14:
all should be good now :) was not working before on 1.0.2 because we did not notice a change in esp-face that made the app compile but not work at the end.

closed 2019-10-14 by me-no-dev

eKselent-Tech, 2021-10-06:
ESP32 Reboots, i have been facing this issue. i switched to older board version of ESP32 Board under Boards Manager. This Solved my issue.

VishalGMeshram37, 2021-11-16:
Hello Everyone
I am Vishal, i am facing the problem with ESp32 cam. I have installed ESP32 version2.0.1, but was not able to detect or recognise the face,  I have used the ESp32 Cam model with FTDI convertor, and Blue Downloading cable which is recomended, i have also installed every version of ESP32 ( fromver 1.0.1 to 2.0.1),  but apart from video streaming nothing is happening, please suggest the corrections ,  Thanking you

xgarb, 2021-11-16:
There's no detection or recognition in 2.0.1. Definitely works in 1.0.4. Maybe you need more light on your face or a different distance from the camera.

me-no-dev, 2021-11-16:
something have changed in the API of the face recognition lib. Will update the example and make it work

VishalGMeshram37, 2021-11-20:
When rolled back to 1.0.4 i got this msg. 

[image]

VishalGMeshram37, 2021-11-20:
And No error with 2.0.1 versions

me-no-dev, 2021-11-20:
did you run an example from newer Arduino on older version? That would explain the error :)

VishalGMeshram37, 2021-11-24:
NO everything is latest

xgarb, 2021-11-24:
You need to use v 1.0.4 (or 5, probably) of the ESP32 Hardware libraries and load the CameraWebServer example as here: https://robotzero.one/esp32-cam-arduino-ide/ You can't use the examples from a different hardware library.

Pretty sure  fb->timestamp doesn't exist in the v1.0.4 so you must be using the new example with the old library.

bigowash, 2021-11-27:
Does face detection not work with version 2.0.1 of ESP32?

xgarb, 2021-11-27:
Nope: https://github.com/espressif/arduino-esp32/issues/5878

1.0.4 definitely works.

tro-nic, 2022-07-27:
> Nope: #5878
> 
> 1.0.4 definitely works.

Yeeeeeees...! FD and FR work 1.0.4

Gabriel-create-code, 2023-05-02:
I am having a similar issue, instead of the intruder alert text I see in the tutorials I just get yellow dots on my face and i am unable to enroll my face can anyone help.