espressif/arduino-esp32 #2732: Using nkolban/esp32-snippets vs espressif/arduino-esp32

ASL07, 2019-05-02:
Hi,

I am developing a BLE app using PlatformIO and ESP32, and atm I am using nkolban/esp32-snippets.

However, readme for the repo nkolban/esp32-snippets says that it is not maintained anymore and that the BLE code is included directly in Arduino, so, I am assuming that this repository already includes nkolban/esp32-snippets ?

Also, how can I configure Platformio to use this repo instead of the Platformio one https://github.com/platformio/platform-espressif32.git?
Let me know if I am right or I am missing something

h2zero, 2019-05-02:
I personally use the snippets BLE code and just copy/paste it into the arduino BLE folder as it's got some changes that haven't been made in the arduino repo yet that I prefer (better multi-client process etc..) but I also heavily modded mine so maybe I'm just partial :). 
Also it looks like @chegewara just merged my PR there so I guess he is maintaining it.

Never used platformio, so can't help there, maybe you can just copy/paste the files?

atanisoft, 2019-05-02:
PIO uses this repo by default when you select the platform as espressif32 in platformio.ini. You should be able to run any snippet from esp32-snippets with the available built-in libraries provided by arduino-esp32 in the PIO env.

closed 2019-08-14 by stale[bot]