espressif/arduino-esp32 #2733: SSL - An invalid SSL record was received MbedTLS message code: -29184

thedim-witted, 2019-05-02:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

thedim-witted, 2019-05-02:
A more detailed error log:

> 15:25:36.881 -> }connecting to requestbin.fullcontact.com
> 15:25:36.881 -> [V][ssl_client.cpp:53] start_ssl_client(): Free internal heap before TLS 214996
> 15:25:36.881 -> [V][ssl_client.cpp:55] start_ssl_client(): Starting socket
> 15:25:36.916 -> [V][ssl_client.cpp:90] start_ssl_client(): Seeding the random number generator
> 15:25:36.916 -> [V][ssl_client.cpp:99] start_ssl_client(): Setting up the SSL/TLS structure...
> 15:25:36.916 -> [I][ssl_client.cpp:153] start_ssl_client(): WARNING: Use certificates for a more secure communication!
> 15:25:36.916 -> [V][ssl_client.cpp:177] start_ssl_client(): Setting hostname for TLS session...
> 15:25:36.916 -> [V][ssl_client.cpp:192] start_ssl_client(): Performing the SSL/TLS handshake...
> 15:25:36.950 -> [V][ssl_client.cpp:213] start_ssl_client(): Verifying peer X.509 certificate...
> 15:25:36.950 -> [V][ssl_client.cpp:222] start_ssl_client(): Certificate verified.
> 15:25:36.950 -> [V][ssl_client.cpp:237] start_ssl_client(): Free internal heap after TLS 178476
> 15:25:36.950 -> [D][ssl_client.cpp:353] verify_ssl_fingerprint(): pos:59 len:59 fingerprint too short
> 15:25:36.950 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:36.984 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:36.984 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:36.984 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:36.984 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:36.984 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:36.984 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:37.018 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:37.018 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:37.018 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:37.018 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:37.018 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
> 15:25:37.018 -> [E][ssl_client.cpp:33] handle_error(): SSL - An invalid SSL record was received
> 15:25:37.053 -> [E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -29184
> 15:25:37.053 -> [V][ssl_client.cpp:245] stop_ssl_socket(): Cleaning SSL connection.
>

thedim-witted, 2019-05-02:
The interesting part is that this is not happening always but only sometimes in between, randomly. I haven't yet seen any pattern regarding this.

chegewara, 2019-05-02:
`15:25:36.950 -> [D][ssl_client.cpp:353] verify_ssl_fingerprint(): pos:59 len:59 fingerprint too short`

thedim-witted, 2019-05-02:
This is because I am using the fingerprint of a different server on this test one. However, even if the fingerprints don't match the data is still getting sent and is received on the other end.

chegewara, 2019-05-02:
If you think its not important then good luck.

thedim-witted, 2019-05-02:
Ok. I'll try it again with a valid fingerprint and check that too. However, one other issue the code not working in case of no debug level lower than "Verbose". Any suggestions for that?

Thanks.

thedim-witted, 2019-05-13:
Hey. Sorry for being late. So I did try with a valid certificate and private key verification but unfortunately, it is getting stuck again. This time I am not even getting any error messages as it is just restarting

closed 2019-08-14 by stale[bot]

Yuron5, 2020-05-08:
Hi! I have similar problem with ESP-WROOM-32 and HTTPSRedirect library. Problem can be in Maximum TLS message length (in bytes) supported by mbedTLS. 16384 is the default value. If the value is set too low, symptoms are a failed TLS handshake or a return value of MBEDTLS_ERR_SSL_INVALID_RECORD (-0x7200).  This variable sets on sdkconfig.h and you can check it. In my case the way is C:\Users\<Local User>\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\config\sdkconfig.h  [C:\Users\Navitek\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\config\sdkconfig.h](url)

caipifrosch, 2023-06-01:
There is a similar problem on my end. Log is reporting invalid SSL record and ESP reboots due to heap problem.

Issue happens very random. Sometime within hours, but also happens after days of operation.

I am using PlatformIO with latest ESP framework release 2.0.9.

Anyone solved this problem or may have ideas how to takle this?

Thanks a lot in advance for your replies!
 

15:06:00.360 > SYSTEM: Runtime: 0d 01h 20min 05sec Free Heap: 90.86 KB
15:06:00.362 >
15:06:09.845 > [4814632][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-29184) SSL - An invalid SSL record was received
15:06:09.854 > [4814633][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-29184) SSL - An invalid SSL record was received
15:06:09.864 > CORRUPT HEAP: Bad head at 0x3ffd9240. Expected 0xabba1234 got 0x3ffcd9ec
15:06:09.870 >
15:06:09.873 > assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)
15:06:09.879 >
15:06:09.879 >
15:06:09.879 > Backtrace: 0x40083e45:0x3ffbb400 0x40095301:0x3ffbb420 0x4009af69:0x3ffbb440 0x4009abdb:0x3ffbb570 0x40084309:0x3ffbb590 0x4008d3e1:0x3ffbb5b0 0x401dc122:0x3ffbb5d0 0x401cd8b2:0x3ffbb5f0 0x400f656a:0x3ffbb610 0x400f5268:0x3ffbb630 0x400f5349:0x3ffbb650 0x400f52b4:0x3ffbb670 0x4022c34a:0x3ffbb690 0x4022bd55:0x3ffbb6c0 0x400e0a05:0x3ffbb6e0 0x400d3a43:0x3ffbb700 0x4012edcf:0x3ffbb720
15:06:09.912 >
15:06:09.912 >
15:06:09.912 >
15:06:09.912 >
15:06:09.912 > ELF file SHA256: 89e4abb211f07ab3
15:06:09.916 >
15:06:10.328 > Rebooting...

tigramaan, 2024-12-22:
this happens if you use ssl client in different contexts. for example from loop, ticker or freertos car to run tasks using the same resources. there will be competition for resources and damage to the heap and connection break and maybe reboot. and problems will just arise occasionally.