espressif/arduino-esp32 #2740: Arduino board support/definition for TTGO T1 (v1.3+)

probonopd, 2019-05-03:
A board variant for TTGO-T1 V1.3 is needed.

TTGO-T1 V1.3 is a board based on ESP32 with a TF (microSD) card slot on the board. It is similar to the existing `hardware/esp32/1.0.2/variants/ttgo-lora32-v1/pins_arduino.h` but instead of LoRa it has microSD on board.

[image]

Looking at the pinout diagram my question is, how can we get `SD.begin()` to use GPIO13 for CS, GPIO15 for MOSI, GPIO2 for MISO, GPIO14 for SCK 

```
static const uint8_t SD_SS    = 13;
static const uint8_t SD_MOSI  = 15;
static const uint8_t SD_MISO  = 2;
static const uint8_t SD_SCK   = 14;
```

while keeping, for all other devices,

```
static const uint8_t SS    = 18;
static const uint8_t MOSI  = 23;
static const uint8_t MISO  = 19;
static const uint8_t SCK   = 5;
```

so that the microSD card can be used while other SPI devices can still be attached to the broken out pins?

luc-github, 2019-05-03:
you just call 
`SPI.begin(14, 2, 15, 13);`
before 
`SD.begin();

check ` void begin(int8_t sck=-1, int8_t miso=-1, int8_t mosi=-1, int8_t ss=-1);`
in
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h#L55

probonopd, 2019-05-03:
Thanks @luc-github

atanisoft, 2019-05-03:
There is also the i2c pins, the built-in LED is on pin 22 and from my experience causes problems for i2c on that default pin.

probonopd, 2019-05-03:
Reopening, since having a board definition file would probably still be a good thing.

atanisoft, 2019-05-03:
I'll see if I can create a variant this weekend for this.

atanisoft, 2019-05-05:
Submitted a PR for the board definition, opted for override SS/MISO/MOSI/SCK with the SD defaults to make it easier for end users. SS needs to be overridden for all SPI devices usually anyway so it doesn't hurt to share the other pins.

closed 2019-05-11 by me-no-dev