espressif/arduino-esp32 #2741: ESP32: can't upload code when using WiFi.begin(ssid, password)

Golku, 2019-05-04:
I'm working on a project and ran into a problem.

**Description**

I'm trying to use a esp32 as a websocket server for a project. To achive this I'm followig this tutorial http://shawnhymel.com/1675/arduino-websocket-server-using-an-esp32/. I provided the code with my SSID and password then uploaded it to the esp32.

I tested this using the google chrome extension "Simple WebSocket Client" and everything was working fine. I was able to send messages to the esp32 and receive them back in my browser. After I was convinced that everything was working fine I changed some parts of the code to suit my needs. Then when I wanted to upload the modify code I got this error message: [Error message](https://i.stack.imgur.com/u1yLu.png)

After a few unsuccessful google searches for a solution I tried uploading the sample code that had worked before. But I got the exact same error. After this I tried uploading the sample code peace by peace until I got the error. I can upload the complete code to the esp32 except this line WiFi.begin(ssid, password); in the setup. For some reason when I use WiFi.begin(ssid, password); in any sketch I'm unable to upload code to the esp32, I get the same error shown above.

This esp32 is also brand new and has only been connected to my pc with a usb cable.

**Question**

Does someone knows how to solve this and why it's happening?

Many thanks in advance.

### Hardware:
Board:							DOIT ESP32 Devkit v1
Core Installation version:			?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     ?
Upload Speed:					921600
Computer OS:                                        Windows 10

lbernstone, 2019-05-04:
Usually a bad USB cable.  Try a different cable, reduce your baud rate, set flash capabilities to lowest rate (40MHz/DIO).

Golku, 2019-05-04:
> Usually a bad USB cable. Try a different cable, reduce your baud rate, set flash capabilities to lowest rate (40MHz/DIO).

Tried all the above and no luck, same error.

closed 2019-08-14 by stale[bot]