espressif/arduino-esp32 #2742: GPIO Interrupt and Functional Interrupt not working as intended

PaladinEng, 2019-05-04:
I've tried running both provided examples on my ESP32 Dev board from SparkFun. I would expect that when I press "Button 1", I would only get an indication that Button 1 was pressed; instead, when either button is pressed, the ISR for both buttons is triggered. 
The only change I've made to the example is to switch to pin 38 and 39.

GPIOInterrupt Output: (only Button 1 pressed 1 time)
`Button 1 has been pressed 1 times
Button 2 has been pressed 1 times`

FunctionalInterrupt Output: (only Button 1 pressed 1 time)
`Button on pin 38 has been pressed 1 times
Button on pin 39 has been pressed 1 times`

atanisoft, 2019-05-04:
Pin 38 doesn't exist.

PaladinEng, 2019-05-04:
it does on this board: 
[https://cdn.sparkfun.com/assets/learn_tutorials/5/0/7/esp32-thing-graphical-datasheet-v02.png](url)

atanisoft, 2019-05-04:
Pins 38 and 39 are internally connected, same for 37 and 36. Any board that exposes either 38 or 37 are very likely mislabeled.

PaladinEng, 2019-05-04:
I switched one button to pin 32, per the advice above and that's solved it.

closed 2019-05-04 by PaladinEng