espressif/arduino-esp32 #2746: #ifdef conditional for specific ESP32 board

cweinhofer, 2019-05-06:
If I want to do a logical test for a specific board in the ESP32 list, how do I figure out what to put in the #ifdef conditional?

I'll give my board as an example, although I'm hoping for information that can be used to make a conditional for any specific board in the ESP32 family.

In my case, I need to assign specific pin when my sketch is loaded onto a LOLIN D32 PRO. So, I need to create a conditional similar to what is shown in [](https://arduino.stackexchange.com/a/21246). I am assuming that the define would be based on something in the ESP32 boards.txt. Based on the D32 Pro's entry which is
`d32_pro.build.board=LOLIN_D32_PRO`
I have tried
`#ifdef LOLIN_D32_PRO`
`#ifdef d32_pro`
and
`#ifdef ESP_LOLIN_D32_PRO`
but none of them seem to work. Which leads me to believe I am not using the right thing for the define.

Can someone tell me how to figure out the proper define based on what's in boards.txt?

PS - If it matters, I'm using the Arduino IDE.

dok-net, 2019-05-06:
Please try ARDUINO_LOLIN_D32_PRO

cweinhofer, 2019-05-06:
Thanks @dok-net That seems to work.

I would appear then that you put "ARDUINO_" in front of the the build.board name.

This seems strange to me since these are not Arduino boards, but I don't know much about this stuff. Can you share any more about why it works like that? Is it because I'm compiling on the Arduino IDE?

closed 2019-08-14 by stale[bot]