espressif/arduino-esp32 #2751: can't maintain i2c connection heart rate sensor

vikbessems, 2019-05-06:
### Hardware:
Board:								Adafruit Huzzah32
Core Installation/update date:			
IDE name:							Arduino IDE with epressif installed
Flash Frequency:						80MHz
PSRAM enabled:						no idea?
Upload Speed:						115200
Computer OS:							Windows 10

### Description:
Hi there!

I'm trying to get my heart rate sensor from seeed to work with my huzzah32.
The sensor works on the regular arduino with the grove shield. I have connected the module properly to the huzzah at the dedicated SDA and SCL pins (23,22), and added 4.7K ohm pull-up resistors to both lines. For now I'm just testing it with the example code on the huzzah32.
I seem to get some value's but after 5-7 seconds it just stops. I'm not a pro at debugging or at electronics but I'll do my best to understand and learn.

The debug messages kept repeating the same message over and over so I only recorded 1 second from when the error started occuring. I'll provide my example code and debug messages below.

### Sketch:
```cpp

#include <Wire.h>
void setup() {
    Serial.begin(115200);
    Serial.println("heart rate sensor:");
    Wire.begin();
}
void loop() {
  Wire.requestFrom(0xA0 >> 1, 1);    // request 1 bytes from slave device
    while(Wire.available()) {          // slave may send less than requested
        unsigned char c = Wire.read();   // receive heart rate value (a byte)
        Serial.println(c, DEC);         // print heart rate value
        
    }
    delay(500);
}
```

### Debug Messages:
```
heart rate sensor:
0
0
0
0
0
0
0
0
0
68
68
[D][esp32-hal-i2c.c:1345] i2cProcQueue():  Gross Timeout Dead start=0x159e, end=0x15d0, =50, max=50 error=1
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbebe0
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb843c
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb84c0 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb84f0
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb849c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(23)=0, scl(22)=1
[D][esp32-hal-i2c.c:1419] i2cCheckLineState(): Recovered after 1 Cycles
[D][esp32-hal-i2c.c:1345] i2cProcQueue():  Gross Timeout Dead start=0x181d, end=0x184f, =50, max=50 error=1
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbebe0
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb843c
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb84c0 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb84f0
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb849c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(23)=1, scl(22)=0
[D][esp32-hal-i2c.c:1419] i2cCheckLineState(): Recovered after 1 Cycles
[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
[D][esp32-hal-i2c.c:1336] i2cProcQueue():  Busy Timeout start=0x1c99, end=0x1ccb, =50, max=50 error=1
```

stickbreaker, 2019-05-06:
@vikbessems The "gross" timeout is telling you that the slave stretched SCL (paused communications while sampling) longer than the ESP32 can handle (approximately 13.1ms).  It is a fatal error.

The cascade of errors is because the ESP32 aborted the READ transaction when it timed out. But the Sensor is still in read mode.  It is trying to output the sensor value, but the ESP32 is trying to begin a new transaction.  The bus is in use (Slave is driving SDA, waiting for the Master(ESP32) to cycle the SCL Clock. Except, the ESP32 is waiting for the Bus to Clear (SCL & SDA high).

Check the data sheet for your heartrate sensor, see if there is a polling mode instead of the SCL stretching.

I would change your code to something like this:
```c++
void loop() {
  uint8_t count = Wire.requestFrom(0xA0 >> 1, 1);    // request 1 bytes from slave device
    if(count==1) {          // slave may send less than requested
        unsigned char c = Wire.read();   // receive heart rate value (a byte)
        Serial.println(c, DEC);         // print heart rate value
        }
    else { // error
       Serial.printf("request from sensor failed =%d(%s)\n",Wire.lastError(), Wire.getErrorText(Wire.lastError()));
    }
    delay(500);
}

Chuck.

vikbessems, 2019-05-06:
@stickbreaker It has a reserved SWD Interface for programming to STM32 (Integrated the STM32F103). I noticed it has some extra pins on the little pcb (GND,3V3,SWCLK and SWDIO). Thats all the info I got from the seeed wiki page. I still have to figure out how to reprogram it without breaking everything.

closed 2019-08-14 by stale[bot]