espressif/arduino-esp32 #2753: Missing f_utime() in latest arduino-esp32 update?

cbpdk, 2019-05-07:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			1.0.2/ apr 16 2019
IDE name:						Arduino IDE 1.8.9
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:			                921600
Computer OS:                                          Ubuntu 18.04

### Description:
In the latest update of arduino-esp32 it seems that the f_utime() function is missing.
I used it to set sd card file creation date using the time from a rtc3231 module.
Otherwise it looks silly to see all files created 01/01/1980 01.00.

Searching the sdk 1.0.2 for f_utime shows:
./tools/sdk/include/fatfs/ff.h
./tools/sdk/include/fatfs/ffconf.h

Searchig the previous version gave:
./tools/sdk/include/fatfs/ffconf.h
./tools/sdk/include/fatfs/ff.h
./tools/sdk/lib/libfatfs.a

### Sketch:  
```cpp
int set_timestamp (
  char *obj,     /* Pointer to the file name */
  int year,
  int month,
  int mday,
  int hour,
  int min,
  int sec
)
{
  FILINFO fno;

  fno.fdate = (WORD)(((year - 1980) * 512U) | month * 32U | mday);
  fno.ftime = (WORD)(hour * 2048U | min * 32U | sec / 2U);

  return f_utime(obj, &fno);
}
```

### Debug Messages:
/tmp/arduino_build_565431/sketch/ESP32_Receiver_Wifi_AP.ino.cpp.o: In function `set_timestamp(char*, int, int, int, int, int, int)':
/home/xx/lcdsketchbook/esp32rx-tx/ESP32_Receiver_Wifi_AP/ESP32_Receiver_Wifi_AP.ino:427: undefined reference to `f_utime'
```

me-no-dev, 2019-05-11:
this is caused because Arduino is tracking the 3.2 branch of IDF :)

cbpdk, 2019-05-15:
As an aside, once I found out to set the system time to UTC using a ds3231, the file stamps work.
And shows the correct time when using the sd card on a PC.
However, I have not figured out how to use time zones to update the file stamp correctly(UTC time) except by using UTC system time with no time zone.
Fortunately I have algorithms from e.g. Arduino Uno for other time keeping.

closed 2019-05-15 by cbpdk