espressif/arduino-esp32 #2755: Virtual functions in client.h are out of line with standard Arduino API

SRGDamia1, 2019-05-07:
### Hardware:
Board:							ESP32
Core Installation version:			1.0.2
IDE name:						PlatformIO
Flash Frequency:					N/A
PSRAM enabled:                                     N/A
Upload Speed:					N/A
Computer OS:                                        Linux (TravisCI)

### Description:
I am working on TinyGSM - a cross-platform GSM client.  Because the client.h file in the current ESP32 core does not match the standard Arduino API (https://github.com/arduino/ArduinoCore-API/blob/master/api/Client.h) the library can no longer be compiled by esp32 users.

NOTE:  The ESP8266 core introduced the same problem in version 2.5.0.  They are fixing it for 2.5.1 (https://github.com/esp8266/Arduino/pull/5969)

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <Arduino.h>

#define TINY_GSM_MODEM_SIM800
#include <TinyGsmClient.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
Unable to compile because of undefined virtual functions:

```
/tmp/tmpgkfUuk/src/test_build.ino:11:15: error: cannot declare variable 'client' to be of abstract type 'TinyGsmA6::GsmClient'
TinyGsmClient client(modem);
^
In file included from lib/TinyGSM/src/TinyGsmClient.h:65:0,
from /tmp/tmpgkfUuk/src/test_build.ino:8:
lib/TinyGSM/src/TinyGsmClientA6.h:44:7: note:   because the following virtual functions are pure within 'TinyGsmA6::GsmClient':
class GsmClient : public Client
^
In file included from /home/travis/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:157:0,
from /tmp/tmpdTm4GS:1:
/home/travis/.platformio/packages/framework-arduinoespressif32/cores/esp32/Client.h:31:17: note: 	virtual int Client::connect(IPAddress, uint16_t, int)
virtual int connect(IPAddress ip, uint16_t port, int timeout) =0;
^
/home/travis/.platformio/packages/framework-arduinoespressif32/cores/esp32/Client.h:32:17: note: 	virtual int Client::connect(const char*, uint16_t, int)
virtual int connect(const char *host, uint16_t port, int timeout) =0;
^
```

This was also mentioned in #2704.

maniekq, 2019-05-14:
I've got the same problem. After downgrading to 1.0.1 it works again.

thujer, 2019-05-15:
### Hardware
Board: WeMos WiFi and Bluetooth Battery
Platform: Espressif 32
Configuration: ESP32 240MHz 320KB RAM (4MB Flash)
IDE: VSCode 1.33.1
OS: Linux / XUbuntu

### Description
I working on multi communication device (Wifi, Ethernet, RS232, Bluetooth). Last week I cannot compile source codes, because there is some errors in Ethernet library, described bellow.

I tried to compile some basic Arduino examples (WebClient) in Arduino IDE and I got same resolution.

```
In file included from src/EthernetController/EthernetController.h:5:0,
from src/App/App.h:11,
from src/App/App.cpp:2:
.piolibdeps/Ethernet_ID872/src/Ethernet.h:258:2: error: invalid abstract return type 'EthernetClient'
EthernetClient available();
^
.piolibdeps/Ethernet_ID872/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
class EthernetClient : public Client {
^
```

[image]

[image]

This was also mentioned in Arduino Libraries Issues
https://github.com/arduino-libraries/Ethernet/issues/97

misak113, 2019-05-23:
We're dealing with the exactly same issue. Could you please, please :+1: look at it? :)

digitalwm, 2019-05-26:
Same issue here.

brentru, 2019-05-30:
Same issue with ArduinoHTTPClient and Adafruit IO Arduino, @me-no-dev will this change be reverted in a future 1.0.3 release?

thujer, 2019-06-10:
We fixed this incompatibility by downgrading Espressif32 platform core from 1.8.0 to 1.7.0.

Just put line bellow into platformio.ini
```
platform = espressif32@1.7.0
```

If some other libraries was updated and some errors will be show, then try downgrade libraries by the same way or put the old libraries code directly into the project directory and update #include directives.

Good Luck

> ### Hardware
> Board: WeMos WiFi and Bluetooth Battery
> Platform: Espressif 32
> Configuration: ESP32 240MHz 320KB RAM (4MB Flash)
> IDE: VSCode 1.33.1
> OS: Linux / XUbuntu
> 
> ### Description
> We are working on multi communication device (Wifi, Ethernet, RS232, Bluetooth). Last week we cannot compile source codes, because there is some errors in Ethernet library, described bellow.
> 
> I tried to compile some basic Arduino examples (WebClient) in Arduino IDE and I got same resolution.
> 
> ```
> In file included from src/EthernetController/EthernetController.h:5:0,
> from src/App/App.h:11,
> from src/App/App.cpp:2:
> .piolibdeps/Ethernet_ID872/src/Ethernet.h:258:2: error: invalid abstract return type 'EthernetClient'
> EthernetClient available();
> ^
> .piolibdeps/Ethernet_ID872/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
> class EthernetClient : public Client {
> ^
> ```
> 
> [image]
> 
> [image]
> 
> This was also mentioned in Arduino Libraries Issues
> [arduino-libraries/Ethernet#97](https://github.com/arduino-libraries/Ethernet/issues/97)

brentru, 2019-06-10:
@thujer Yep, you can downgrade to retain the old feature. Point is - the function signatures should be in-line with the Arduino API in a new release of the ESP32 BSP.

brentru, 2019-06-13:
@me-no-dev Any updates on this?

eabase, 2019-07-19:
Hi Guys!
Why not try to fix this? It just doesn't make sense to have to downgrade, since who knows what other crazy 2ndary bugs would get (re)introduced by doing such a thing. 

We're also trying to implement an IoT solution that require an updated library when using [TinyGSM](https://github.com/vshymanskyy/TinyGSM), so it would be a bloody shame to have to abandon 2 libraries (and perhaps even more) just because *this one* won't play ball. Especially since several other things already depend on this.  

I hope you can reconsider and prioritize this issue.

Cheers,
:+1:

xvinny, 2019-08-12:
I had to implement my own HttpClient (it is very simple, with support only for GET). Like @eabase said, why don't you prioritize this issue? It is simple to solve and it would help a lot of users!!

brentru, 2019-08-12:
@me-no-dev  Any updates on this issue?

hardmax, 2019-08-30:
Hi,

ArduinoHttpClient library has the same problem, please review to correct this..are a lot a libraries with problems

kugelkopf123, 2019-09-08:
The Problem still exists with 1.0.3-rc3

me-no-dev, 2019-09-09:
Sorry guys! I get hundreds of emails daily and it's not possible to read them all. Please, stop by gitter (link in the readme) and drop a message if something is seriously wrong. I try to keep an eye there. Issue is fixed and will be in 1.0.3 final.

closed 2019-09-09 by me-no-dev

xvinny, 2019-09-18:
@SRGDamia1 the 1.0.3 final update was released!!!

brentru, 2019-09-18:
1.0.3 works with Adafruit IO Arduino and Adafruit MQTT, thanks @me-no-dev.

per1234, 2024-12-17:
There was a regression of this bug in https://github.com/espressif/arduino-esp32/pull/10202

Example of breakage caused: https://github.com/arduino-libraries/ArduinoHttpClient/issues/182

pennam, 2024-12-17:
It breaks also https://github.com/arduino-libraries/ArduinoMqttClient/pull/112