espressif/arduino-esp32 #2756: SPIFFS memory leak

bill-orange, 2019-05-08:
Make your question, not a Statement, inclusive. Include all pertinent information:

I am using SPIFFS to store log data.  I am running identical code on an ESP32 and an ESP8266.  In the ESP32, I have a memory leak eating up about 500 bytes on eachSPIFFS write function call.  After about 24 hours of run time the memory loss is sever enough to crash other functions.Is there something I can do to recover lost memory (or cache) at the end of the function shown below?

My system is a Wemos clone ESP32 using the 1.0.2 core release compiled in the Arduino IDE 1.8.8.

Here is a code snippet (SPIFFS.begin(true); is in Startup):

```
Serial.print("freeMemory()= "); Serial.print(system_get_free_heap_size()); Serial.println();
  File tempLog = SPIFFS.open("/temp.csv", "w");

if (numberOfFoundSensors == 2) {

    tempLog.print(olderSensorTemp_st); tempLog.print(','); tempLog.println(olderSensorTemp, 1);
    Serial.print ("olderSensorTemp: "); Serial.print(olderSensorTemp, 1); Serial.println(  "°F" );
    Serial.print ("oldSensorTemp: "); Serial.print(oldSensorTemp, 1); Serial.println( "°F" );

    tempLog.print(actualTime_st); tempLog.print(','); tempLog.println(temp_1, 1);
    Serial.print ("actual current: "); Serial.print(temp_1, 1); Serial.println ( "°F" );

    tempLog.print(nearestFutureSensorTemp_st); tempLog.print(','); tempLog.println(nearestFutureSensorTemp, 1);
    Serial.print ("forecast actual half hour ahead: "); Serial.print (nearestFutureSensorTemp); Serial.println ( "°F" );

    tempLog.print(nearFutureSensorTemp_st); tempLog.print(','); tempLog.println(nearFutureSensorTemp, 1);
    Serial.print ("forecast actual one hour ahead: ");  Serial.print (nearFutureSensorTemp, 1); Serial.println ( "°F" );

    tempLog.print(futureActualTime_st); tempLog.print(','); tempLog.println(futureSensorTemp , 1);
    Serial.print ("forecast two hours ahead: ");  Serial.print (futureSensorTemp, 1); Serial.println ( "°F" );

    Serial.print ("OWM corrected forecast: "); Serial.print (prettyTime(unix_time)); Serial.print (", "); Serial.print (forecastTemp, 1); Serial.println ( "°F" );

    tempLog.print(actualTime_st); tempLog.print(','); tempLog.print("null"); tempLog.print(','); tempLog.println(temp_2);
    Serial.print ("actual second sensor temperature: "); Serial.print (temp_2, 1); Serial.println( "°F" );

  }
```

Here's the worsening result:
freeMemory()= 205908
olderSensorTemp: 61.7°F
oldSensorTemp: 61.7°F
actual current: 61.7°F
forecast actual half hour ahead: 61.30°F
forecast actual one hour ahead: 60.6°F
forecast two hours ahead: 59.0°F
OWM corrected forecast: 11:00 PM, 57.9°F
actual second sensor temperature: 75.2°F
freeMemory()= 205516
uptime (minutes) = 3

As mentioned above, the problem does not exist with the same code running on and ESP8266 which I believe uses a different SPIFFS.h.  I have check the rest of the sketch for memory leaks and it seems limited to the use of spiffs.  

What can I do to plug this memory leak?

beegee-tokyo, 2019-05-08:
Is the above code in the loop() function?

I see you open the file with `File tempLog = SPIFFS.open("/temp.csv", "w");` but you never close it. 
Not sure, but could this be the problem?

bill-orange, 2019-05-14:
Found the problem. It was not within the SPIFFS. Ping for the ESP32 was the source of the leak. SPIFFS.end recover some of the lost memory confusing me as to the cause.

closed 2019-05-14 by bill-orange

adynis, 2019-12-28:
@bill-orange : you saved me with your hint regarding "ping" ... I was investigating and investigating ... "who is the guilty" in my code... After I've seen you comment, I tried to remove ESP32Ping library .. and ... solved ! :) Thanks !!!