espressif/arduino-esp32 #2761: OTA Guru Meditation Error StoreProhibited

Shuptuu, 2019-05-09:
Hi,
In my project, I have 2 ESP32 working in parrallel, comunicating together through Serial1.
The first esp32 is with wifi on. The second is with wifi off, acting as a dedicated microcontroller for a specific task.
I implemented OTA firmware update on the first Esp, works well. 
Trying now to implement OTA firmware update for the second esp, through the first one as long as I don't have wifi access on the second. 
The idea: the first one receive the bin image and forward to the second through serial1.
It almost works! The transfer starts correctly, but after a while, I get a  Guru Meditation Error: Core  1 panic'ed (StoreProhibited)...
Seems to be linked to the uart performance...
Any suggestion to fix that problem?
Here my code on the first esp:
`server.on("/updateB", HTTP_POST, []() {
    server.sendHeader("Connection", "close");
    server.send(200, "text/plain", "OK");
  }, []() {
    HTTPUpload& upload = server.upload();
    // Check if file seems valid .bin file
    if(!upload.filename.endsWith(F(".bin"))){
      server.send(500, F("text/plain"), F("ONLY BIN FILES ALLOWED\n"));
      return false;
    }
    int bufsize=0;
    if (upload.status == UPLOAD_FILE_START) {
      Serial.printf("Update B: %s\r\n", upload.filename.c_str());
      SerialEspB.write(FUPDATE); // send update order to espB
      SerialEspB.write(filesize.b,4); // send size of update image         
    } else if (upload.status == UPLOAD_FILE_WRITE) {
      bufsize=upload.currentSize;
      SerialEspB.write(highByte(bufsize));  // send buffer size to espB
      SerialEspB.write(lowByte(bufsize));
      SerialEspB.write(upload.buf, upload.currentSize); 
    } else if (upload.status == UPLOAD_FILE_END) {
      Serial.printf("Total sent = %u\r\n", upload.totalSize );
    } else {
      Serial.printf("Update Failed Unexpectedly (likely broken connection): status=%d\r\n", upload.status);
    }
  });`

on the second esp:
`case FUPDATE:{
      long filesize=getlongfromEspA();
      Serial.printf("Starting update (%u) \r\n", filesize );
      Serial.setDebugOutput(true);
      if (!Update.begin()) { //start with max available size
        Update.printError(Serial);
      }
      uint8_t buf[2048];
      unsigned long starttime=millis();
      int i=0;
      int j=0;
      int bsize=0;
      boolean success=1;
      boolean bready=0;
      while (i<filesize && success && (millis()-starttime)<60000){
        if (SerialEspA.available()){
          if (bready==0){
            bsize=0;
            bsize=SerialEspA.read();
            bsize=bsize << 8;
            bsize+=SerialEspA.read();
            bready=1;
          } else {
            buf[j]=SerialEspA.read();
            i++;
            j++;
            if (j==bsize || i==filesize){
              if (Update.write(buf, j) != j) {
                Update.printError(Serial);
                success=0;
              }
              Serial.print(".");
              j=0;
              bready=0;
            }
          }
        }
        delay(0);
      }
      Serial.println();
      Serial.printf("Total received = %u\r\n", i );
      if (Update.end(true)) { //true to set the size to the current progress
        Serial.printf("Update Success: %u\nRebooting...\r\n", i);
        ESP.restart();
      } else {
        Update.printError(Serial);
      }
      Serial.setDebugOutput(false);
    }break;`

Serial1 setup on the second esp:
`SerialEspA.begin(115200, SERIAL_8N1,RX1,TX1);
 SerialEspA.setRxBufferSize(2048);`

Guru Meditation Message I get:
.......................Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008b66a  PS      : 0x00050033  A0      : 0x4008b573  A1      : 0x3ffb1630  
A2      : 0xb3a83f40  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x4008b4f4  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x80082fa2  A9      : 0x3ffbe790  
A10     : 0x3ffbe7c8  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xb3a83f40  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x4008b66a:0x3ffb1630 0x4008b570:0x3ffb1640
`
Thanks for your help

Guillaume57, 2019-05-09:
Try replace the while loop by a state machine
Add more Serial prints for debugging
60000 should be 60000UL
Also try increase serial speed to 921600

Shuptuu, 2019-05-09:
Thanks Guillaume,
Increasing serial speed make it worst! (tried 921600 and 512000)
I think it's more related to my buf[2048] as long as before to write in this local buffer, I made a test  sending data received on SerialEspA to Serial:

>  Serial.write(SerialEspA.read())

And capturing Serial on my computer, I was able to receive my bin image.
Using this buffer, I made the test again, sending to Serial instead of sending to Update: I have the same Guru Meditation error.

Anyway, I found an other way to do, using Update.writeStream(SerialEspA), based on updateFromFS function found in SD_Update.ino and it works!!! And that's easier!

closed 2019-05-09 by Shuptuu