espressif/arduino-esp32 #2769: WiFi Strange behavior about connection/reconnection

asetyde, 2019-05-12:
### Hardware:
Board:							ESP32 WROOM 32D
Core Installation version:			1.0.2 stable 
IDE name:						Arduino IDE/ Visual Micro on Visual Studio
Flash Frequency:					80cpu/80flash
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
In this month we 've fond many issue that weren't on 1.0.0, but we need temperature reduction with low core, therefore we must use 1.0.2.
WiFi fail sometimes to reconnect, also watchdog sometimes fail to reboot board , we not find any issue about our code.
We ve see many issue but no response from espressif coders, we ve also use ESP-IDF base sdk but issue is hide on WiFi core working and we not investigate inside.

We hope a fast resolving bugfix because devices are not stable and not usable.

### Debug Messages:
Always verbose logs about fail reconnection says "AUTH_EXP" looping
about watchdog , says not triggered many times continuosly, core 0 seems live but core 1 freezed and try rebooting but nothing do, board freeze.

asetyde, 2019-05-12:
If not fast resolution coming we must come back to 1.0.0 but we ve some problems with i2c, millis, core temperature and some big bug

me-no-dev, 2019-05-12:
there is a fix coming for the wifi issues.

asetyde, 2019-05-13:
Many Thanks

me-no-dev, 2019-05-13:
I have merged updated IDF lib. It's only available through github. Could you try them and see if your issues are resolved?

asetyde, 2019-05-13:
Yes now I'm loading on devices, test require 24/48h max .

asetyde, 2019-05-13:
can use 1.0.3 rc  from board manager ? git give me always strange issue

me-no-dev, 2019-05-13:
yes :) I tagged it so you and other people with wifi issues can test

asetyde, 2019-05-13:
not download , give error on board manager

me-no-dev, 2019-05-13:
something went wrong the first time. I have updated the release. Please try again :)

asetyde, 2019-05-13:
ok it go !

asetyde, 2019-05-13:
Report:   (connection go )

[wifi_start] Connessione fallita
[wifi_start] Tentativo di connessione (EEPROM) 2 di 3
Smartconfig
maggio2014
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.1.163, MASK: 255.255.255.0, GW: 192.168.1.1
[wifi_start] Connesso!
 --------------------------------------- 
 - Wifi SSID : Smartconfig
 - Wifi ip : 192.168.1.163
 - Size of structt : 3812
 - Wifi Mac : 30:AE:A4:D4:20:C0
- WIFI signal : -58
- Signal level is  < -67 Very Good 
 - Firm.Chip :v3.2-111-g524308c98
 - EHI SW : 1.1
 - IP configuration : system on DHCP
 - IP on dhcp : 192.168.1.163
 - IPGateway on dhcp : 192.168.1.1
 - IPmask on dhcp : 255.255.255.0
 - Server : 192.168.1.89
 - Serial : 192.168.1.201
 - Broadcast : 192.168.1.255
 --------------------------------------- 
[INFO] Entrata in iamaliveSQLforecast
	Connessione...
Connected to server version 5.5.5-10.1.38-MariaDB-0+deb9u1
Disconnected.
	Query eseguita
[SETUP] WiFi inizializzato
[INFO] PIN TOUCHX :0

[INFO] Avvio Server UDP...

[SISTEMA] Fase 4 - Avvio sistema
[INFO] Stati resettati

[INFO] Setup canali
[INFO] Elenco canali input configurati:
[INFO] Diagnostica canali terminata
SHTX TEST test!
Found model SHT25 Rev(1) Serial #[INFO] Gestione sensore Termperatura: ON
[INFO] Gestione sensore luce: ON
[INFO] Gestione sensore Humi : ON

[SISTEMA] Setup completato
[SISTEMA] Inizio loop.

[INFO] Led backgroud Full
[SETUP] Invio del messaggio di diagnostica
[INFO] Calibrazione Touch

asetyde, 2019-05-13:
Now I test it, first connection not go (always), second go , now I test many situations

asetyde, 2019-05-13:
example : 

Port open
[wifi_start] Stato WiFi: WL_NO_SSID_AVAIL
[wifi_start] Tentativo di connessione (EEPROM) 2 di 3
Smartcondfig
maggio2014
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[wifi_start] Stato WiFi: WL_NO_SSID_AVAIL
[wifi_start] Tentativo di connessione (EEPROM) 3 di 3
Smartcondfig
maggio2014
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[wifi_start] Stato WiFi: WL_NO_SSID_AVAIL
[wifi_start] Tentativo di connessione (Recovery) 1 di 3
Smartconfig
maggio2014
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
[wifi_start] Connessione fallita
[wifi_start] Tentativo di connessione (Recovery) 2 di 3
Smartconfig
maggio2014
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.1.163, MASK: 255.255.255.0, GW: 192.168.1.1

asetyde, 2019-05-13:
now we test all days , I report as soon as possible

asetyde, 2019-05-13:
Note : We new library 1.0.3 i start to use our chip of memory , eprom is not use anymore

asetyde, 2019-05-14:
About some devices : 3 lost connection and 2 of this after 30/40 min re-connected. Our code not use autoconnect and disabile it, also we use olle manual begin with user and password, we disconnect every time we see lost connection and our code begin our code routine to connect again

asetyde, 2019-05-14:
We se only auth exp but we see many logs inside  esp-idf this library not report it

asetyde, 2019-05-14:
About SSID we underline with router with wireless access point , seems esp not choose best signal but choose randomlly, also we see on ESP MDF  where channel is fixed and esp not capace to change it 
Hope advice you, we continue test on it

about watchdog , one device it blocks on loop wifi, watchdog unable to reset it

asetyde, 2019-05-14:
a question , your update is not 3.3 beta IDF correct ? I see many improvements to wifi 
0acf19a Update IDF to v3.2 977854975 (#2771)

me-no-dev, 2019-05-15:
Yup we are on the latest IDF 3.2 with whatever bugfixes were added to it.
How is WiFi doing?

asetyde, 2019-05-15:
WiFi seems has same problems, but now I test it in STATIC mode, it seems work without reconnection, I suppose two reason , Arduino library not manage correctly or not manage a WL_STATUS, or ISSUE of re-connection is a mask for a issue of DHCP , I give to you an example of our routine , but I think we restrict issue to this two ways, DHCP or WL-STATUS (after reconnections. ? )

asetyde, 2019-05-15:
wl_status_t wifi_connect(boolean recovery)
{
	static unsigned long conn_timeout = 5000;
	static unsigned long t0, t1 = 0;
	char wssid[30], wpasswd[30];

	if (WiFi.status() == WL_CONNECTED)
	{
		WiFi.disconnect();
	}

	WiFi.mode(WIFI_MODE_STA);
	WiFi.setHostname("Electric Labs Device");

	if (recovery)
	{
		snprintf(wssid, sizeof(wssid), "Smartconfig");
		snprintf(wpasswd, sizeof(wpasswd), "maggio2014");
	}
	else
	{
		if (memory.iptype == 100)
		{
			WiFi.config(convertfromStringSTD(memory.ipset), convertfromStringSTD(memory.ipgateway), convertfromStringSTD(memory.ipmask));
		}
		snprintf(wssid, sizeof(wssid), memory.ssid);
		snprintf(wpasswd, sizeof(wpasswd), memory.password);
	}

	WiFi.begin(wssid, wpasswd);

	t0 = t1 = millis();
	while ((WiFi.status() != WL_CONNECTED) && ((t1 - t0) < conn_timeout))
	{
		delay(500);            // internal memory                         // revovery memory
		if (!recovery){ledcommands(false, false, 0, 2);} else { ledcommands(false, false, 0, 2);}
		delay(500);
		if (!recovery){ledcommands(true, false, 0, 2);} else { ledcommands(false, true, 0, 2); }
		t1 = millis();
	}

	return WiFi.status();
}

asetyde, 2019-05-15:
wl_status_t wifi_connect(boolean recovery)
{
	static unsigned long conn_timeout = 5000;
	static unsigned long t0, t1 = 0;
	char wssid[30], wpasswd[30];

	if (WiFi.status() == WL_CONNECTED)
	{
		WiFi.disconnect();
	}

	WiFi.mode(WIFI_MODE_STA);
	WiFi.setHostname("Electric Labs Device");

	if (recovery)
	{
		snprintf(wssid, sizeof(wssid), "Smartconfig");
		snprintf(wpasswd, sizeof(wpasswd), "maggio2014");
	}
	else
	{
		if (memory.iptype == 100)
		{
			WiFi.config(convertfromStringSTD(memory.ipset), convertfromStringSTD(memory.ipgateway), convertfromStringSTD(memory.ipmask));
		}
		snprintf(wssid, sizeof(wssid), memory.ssid);
		snprintf(wpasswd, sizeof(wpasswd), memory.password);
	}

	WiFi.begin(wssid, wpasswd);

	t0 = t1 = millis();
	while ((WiFi.status() != WL_CONNECTED) && ((t1 - t0) < conn_timeout))
	{
		delay(500);            // internal memory                         // revovery memory
		if (!recovery){ledcommands(false, false, 0, 2);} else { ledcommands(false, false, 0, 2);}
		delay(500);
		if (!recovery){ledcommands(true, false, 0, 2);} else { ledcommands(false, true, 0, 2); }
		t1 = millis();
	}

	return WiFi.status();
}
boolean wifi_start()
{
	WiFi.setAutoReconnect(false);

	int RETRY_MAX = 3;
	int retry = 0;
	boolean memory_connected = false;

	wdtreset();
	while (retry < RETRY_MAX)
	{
		debug("[wifi_start] Tentativo di connessione (EEPROM) "); debug((retry + 1)); debug(" di "); debugn(RETRY_MAX);
		wl_status_t result = wifi_connect(false);
		int rssi = 0;
		int value2 = 0;

		switch (result)
		{
			case WL_CONNECT_FAILED:
				debugn("[wifi_start] Connessione fallita");
				break;

			case WL_CONNECTED:
				currentIP.fromString(WiFi.localIP().toString());
				server[0] = currentIP[0];
				server[1] = currentIP[1];
				server[2] = currentIP[2];
[... 176 lines cut ...]
			debugn("WL_NO_SHIELD");
			break;
		case WL_CONNECTED:
			debugn("WL_CONNECTED");
			break;
		case WL_IDLE_STATUS:
			debugn("WL_IDLE_STATUS");
			break;
		case WL_NO_SSID_AVAIL:
			debugn("WL_NO_SSID_AVAIL");
			break;
		case WL_SCAN_COMPLETED:
			debugn("WL_SCAN_COMPLETED");
			break;
		case WL_CONNECT_FAILED:
			debugn("WL_CONNECT_FAILED");
			break;
		case WL_CONNECTION_LOST:
			debugn("WL_CONNECTION_LOST");
			break;
		case WL_DISCONNECTED:
			debugn("WL_DISCONNECTED");
			break;
	}
}

asetyde, 2019-05-15:
void printWiFiStatus(wl_status_t state)
{
	switch (state)
	{
		case WL_NO_SHIELD:
			debugn("WL_NO_SHIELD");
			break;
		case WL_CONNECTED:
			debugn("WL_CONNECTED");
			break;
		case WL_IDLE_STATUS:
			debugn("WL_IDLE_STATUS");
			break;
		case WL_NO_SSID_AVAIL:
			debugn("WL_NO_SSID_AVAIL");
			break;
		case WL_SCAN_COMPLETED:
			debugn("WL_SCAN_COMPLETED");
			break;
		case WL_CONNECT_FAILED:
			debugn("WL_CONNECT_FAILED");
			break;
		case WL_CONNECTION_LOST:
			debugn("WL_CONNECTION_LOST");
			break;
		case WL_DISCONNECTED:
			debugn("WL_DISCONNECTED");
			break;
	}
}`

me-no-dev, 2019-05-15:
please read how to paste code :)

asetyde, 2019-05-15:
sorry I do with cell phone, it's a bit confusing, thanks for reformat

asetyde, 2019-05-16:
UPDATE :  With static configuration and this library, no issue, therefore I thin something with disconnection and DHCP is wrong. We can not investigate inside, we not have other information about AUTH_EXP

asetyde, 2019-05-17:
Other test and continuos operative test on STATIC IP not has issue, I think we can close circle on DHCP

asetyde, 2019-05-22:
Continues test and no issue, uses watchdog with while(1) blocking function it works.
We have find some strange casual issue about data from .RSSI sometimes values is corrupted .. this can say something but I don't know

asetyde, 2019-06-27:
Now it working so better without freeze, but I see sometimes reconnection and sometimes reboot of ESP32 WROOM D. I hope soon @me-no-dev  you will upgrade to new IDF with new bug fix about wifi, I see on logs many correction , we need stable working library. Hope to improve signal performance, it's lower than first firmware.

GeorgeFlorian, 2019-07-18:
I am experiencing something similar: https://github.com/espressif/arduino-esp32/issues/2999

Any luck with this ?

asetyde, 2019-07-18:
now I see new IDF loaded .. I'm waiting new RC to test it .. at now go better but signal so so .. sometimes not freeze but reboot .. i wait to test it

CircuitSetup, 2019-07-18:
I'm wondering if this has to do with this issue: https://github.com/espressif/arduino-esp32/issues/2899

However, in the last IDF update CONFIG_ESP32_WIFI_RX_BA_WIN was changed to 16 from 6

CircuitSetup, 2019-07-19:
I'm curious if the problems people are having with reconnecting to wifi have anything to do with the power being provided? This seems to only happen to me when running a smaller power supply. 

When it tries to reconnect to the AP, it fails and eventually freezes. It seems to not disconnect in the first place with more power, but if it does, it reconnects easier.

CircuitSetup, 2019-07-20:
> now I see new IDF loaded .. I'm waiting new RC to test it .. at now go better but signal so so .. sometimes not freeze but reboot .. i wait to test it

You could just copy the current files in the repository. I did this, but unfortunately it didnt solve my issue.

closed 2019-10-02 by stale[bot]