espressif/arduino-esp32 #2770: AD5933 - i2C stuck at Initialization

sanjum89, 2019-05-12:
### Hardware:
Board:							ESP32 Dev Module WROOM32
Core Installation version:			1.0.2
IDE name:						Arduino IDE 1.8.9
Flash Frequency:					80Mhz
PSRAM enabled:                                      No
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I am using the AD5933 Code and Libraries provided by  https://github.com/mjmeli/arduino-ad5933. This is working on an Arduino UNO.

Uploaded the same code on an ESP32 Devkit module. The code hangs at initialization step. Doesn't even show failed Initialization.
I saw that using stickbreaker (https://github.com/stickbreaker/arduino-esp32#esp32dev-board-pinmap) solved i2c issues.

Using stickbreaker fixed initialization issues and now it initializes correctly but gets stuck in Calibrations step. I commented our calibrations step and then it fails at frequency sweep with the following Guru meditation Error.

### Sketch:
```
/*
ad5933-test
    Reads impedance values from the AD5933 over I2C and prints them serially.
*/

#include <Wire.h>
#include "AD5933.h"

#define START_FREQ  (80000)
#define FREQ_INCR   (1000)
#define NUM_INCR    (40)
#define REF_RESIST  (10000)

double gain[NUM_INCR+1];
int phase[NUM_INCR+1];

void setup(void)
{
  // Begin I2C
  Wire.begin();

  // Begin serial at 9600 baud for output
  Serial.begin(9600);
  Serial.println("AD5933 Test Started!");

  // Perform initial configuration. Fail if any one of these fail.
  if (!(AD5933::reset() &&
        AD5933::setInternalClock(true) &&
        AD5933::setStartFrequency(START_FREQ) &&
        AD5933::setIncrementFrequency(FREQ_INCR) &&
        AD5933::setNumberIncrements(NUM_INCR) &&
        AD5933::setPGAGain(PGA_GAIN_X1)))
        {
            Serial.println("FAILED in initialization!");
            while (true) ;
        }

  // Perform calibration sweep
  if (AD5933::calibrate(gain, phase, REF_RESIST, NUM_INCR+1))
    Serial.println("Calibrated!");
  else
    Serial.println("Calibration failed...");
}

void loop(void)
{
  // Easy to use method for frequency sweep
  frequencySweepEasy();

  // Delay
  delay(5000);

  // Complex but more robust method for frequency sweep
  frequencySweepRaw();

  // Delay
  delay(5000);
}

// Easy way to do a frequency sweep. Does an entire frequency sweep at once and
// stores the data into arrays for processing afterwards. This is easy-to-use,
// but doesn't allow you to process data in real time.
void frequencySweepEasy() {
    // Create arrays to hold the data
    int real[NUM_INCR+1], imag[NUM_INCR+1];

    // Perform the frequency sweep
    if (AD5933::frequencySweep(real, imag, NUM_INCR+1)) {
      // Print the frequency data
      int cfreq = START_FREQ/1000;
      for (int i = 0; i < NUM_INCR+1; i++, cfreq += FREQ_INCR/1000) {
        // Print raw frequency data
        Serial.print(cfreq);
        Serial.print(": R=");
        Serial.print(real[i]);
        Serial.print("/I=");
        Serial.print(imag[i]);

        // Compute impedance
        double magnitude = sqrt(pow(real[i], 2) + pow(imag[i], 2));
        double impedance = 1/(magnitude*gain[i]);
        Serial.print("  |Z|=");
        Serial.println(impedance);
      }
      Serial.println("Frequency sweep complete!");
    } else {
      Serial.println("Frequency sweep failed...");
    }
}

// Removes the frequencySweep abstraction from above. This saves memory and
// allows for data to be processed in real time. However, it's more complex.
void frequencySweepRaw() {
    // Create variables to hold the impedance data and track frequency
    int real, imag, i = 0, cfreq = START_FREQ/1000;

    // Initialize the frequency sweep
    if (!(AD5933::setPowerMode(POWER_STANDBY) &&          // place in standby
          AD5933::setControlMode(CTRL_INIT_START_FREQ) && // init start freq
          AD5933::setControlMode(CTRL_START_FREQ_SWEEP))) // begin frequency sweep
         {
             Serial.println("Could not initialize frequency sweep...");
         }

    // Perform the actual sweep
    while ((AD5933::readStatusRegister() & STATUS_SWEEP_DONE) != STATUS_SWEEP_DONE) {
        // Get the frequency data for this frequency point
        if (!AD5933::getComplexData(&real, &imag)) {
            Serial.println("Could not get raw frequency data...");
        }

        // Print out the frequency data
        Serial.print(cfreq);
        Serial.print(": R=");
        Serial.print(real);
        Serial.print("/I=");
        Serial.print(imag);

        // Compute impedance
        double magnitude = sqrt(pow(real, 2) + pow(imag, 2));
        double impedance = 1/(magnitude*gain[i]);
        Serial.print("  |Z|=");
        Serial.println(impedance);

        // Increment the frequency
        i++;
        cfreq += FREQ_INCR/1000;
        AD5933::setControlMode(CTRL_INCREMENT_FREQ);
    }

    Serial.println("Frequency sweep complete!");

    // Set AD5933 power mode to standby when finished
    if (!AD5933::setPowerMode(POWER_STANDBY))
        Serial.println("Could not set to standby...");
}
```

stickbreaker, 2019-05-12:
@sanjum89 two things.
* Do you have pullup resistors on SCL and SDA, the ESP32's internal pullups are too weak for reliable communications.  I recommend using 3.3k resistors this will provide 1mA of pullup allowing full speed 400k bus speeds.
* Second, I would modify the library, add some error reporting.  Currently the library detects I2C fails, but does not report what failed:

Change these low level function in the library to report the problems.
```c++
int AD5933::getByte(byte address, byte *value) {
    // Request to read a byte using the address pointer register
    Wire.beginTransmission(AD5933_ADDR);
    Wire.write(ADDR_PTR);
    Wire.write(address);

    // Ensure transmission worked
    if (byte res = Wire.endTransmission() != I2C_RESULT_SUCCESS) {
        *value = res;
        log_e("Selecting register(0x%02x) failed=%d(%s)",address,Wire.lastError(),Wire.getErrorText(Wire.lastError()));
        return false;
    }

    // Read the byte from the written address
    Wire.requestFrom(AD5933_ADDR, 1);
    if (Wire.available()) {
        *value = Wire.read();
         log_i("Read register(0x%02x)=%d",address,*value);
    return true;
    } else {
        *value = 0;
        log_e("Reading register(0x%02x) failed=%d(%s)",address,Wire.lastError(),Wire.getErrorText(Wire.lastError()));
        return false;
    }
}

/**
 * Write a byte to a register on the AD5933.
 *
 * @param address The register address to write to
 * @param value The byte to write to the address
 * @return Success or failure of transmission
 */
bool AD5933::sendByte(byte address, byte value) {
    // Send byte to address
    Wire.beginTransmission(AD5933_ADDR);
    Wire.write(address);
    Wire.write(value);

    // Check that transmission completed successfully
    if (byte res = Wire.endTransmission() != I2C_RESULT_SUCCESS) {
        log_e("Writing register(0x%02x)=%d failed=%d(%s)",address,value,Wire.lastError(),Wire.getErrorText(Wire.lastError()));

        return false;
    } else {
        log_i("Wrote register(0x%02x)=%d",address,value);
        return true;
    }
}
```

Set the CORE DEBUG LEVEL to ERROR to see the fails, set it to INFO to see all fails and successful communications.  Then recompile and upload.

After you have these logs we can work through the problems.

Chuck.

sanjum89, 2019-05-13:
Hi Chuck,

Thank you for your time.

- Yes, there are pull up resistors on the i2c lines. It's 10K instead of 3.3K as that was the recommended value on most of the AD5933 designs I was looking at.

- I made the changes you mentioned to AD5933.cpp file
CORE DEBUG LEVEL was set to Error, the code was compiled and uploaded.
Serial Monitor didn't show any errors and the code was still stuck at Calibration
###Output :
'''
AD5933 Test Started!
'''
Set the DEBUG LEVEL to INFO, compiled and uploaded again and this was the output -

###Output :
''' 
AD5933 Test Started!
[I][esp32-hal-i2c.c:1076] i2cProcQueue(): create ISR
[I][AD5933.cpp:39] getByte(): Read register(0x81)=0
[I][AD5933.cpp:67] sendByte(): Wrote register(0x81)=16
[I][AD5933.cpp:67] sendByte(): Wrote register(0x81)=0
[I][AD5933.cpp:67] sendByte(): Wrote register(0x82)=39
[I][AD5933.cpp:67] sendByte(): Wrote register(0x83)=16
[I][AD5933.cpp:67] sendByte(): Wrote register(0x84)=185
[I][AD5933.cpp:67] sendByte(): Wrote register(0x85)=0
[I][AD5933.cpp:67] sendByte(): Wrote register(0x86)=125
[I][AD5933.cpp:67] sendByte(): Wrote register(0x87)=2
[I][AD5933.cpp:67] sendByte(): Wrote register(0x88)=0
[I][AD5933.cpp:67] sendByte(): Wrote register(0x89)=40
[I][AD5933.cpp:39] getByte(): Read register(0x80)=33
[I][AD5933.cpp:67] sendByte(): Wrote register(0x80)=33
[I][AD5933.cpp:39] getByte(): Read register(0x80)=33
[I][AD5933.cpp:67] sendByte(): Wrote register(0x80)=177
[I][AD5933.cpp:39] getByte(): Read register(0x80)=177
[I][AD5933.cpp:67] sendByte(): Wrote register(0x80)=17
[I][AD5933.cpp:39] getByte(): Read register(0x80)=17
[I][AD5933.cpp:67] sendByte(): Wrote register(0x80)=33
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
[I][AD5933.cpp:39] getByte(): Read register(0x8f)=112
'''
and it goes on showing [I][AD5933.cpp:39] getByte(): Read register(0x8f)=112

Later, I added a print statement after initializing so that I am sure it is getting Initialized and is stuck at Calibration. Then I commented out calibration and the entire thing gets stuck just before frequency sweep.

stickbreaker, 2019-05-13:
@sanjum89 
so, you have a hardware problem, or the library is not configuring the sensor correctly,  Verify the configuration of the sensor is correct.  If it is configured correctly, then the only other possibility is that the sensor hardware has failed.

The I2C communications subsystem seems to be functioning correctly, but the sensor is not responding as expected. 

Chuck.

stickbreaker, 2019-05-13:
@sanjum89 
You might post this issue against the libraries github [ADS5933 lib by @mjmeli](https://github.com/mjmeli/arduino-ad5933).  He might have an idea why the sensor is not responding.

Chuck.

sanjum89, 2019-06-02:
Sorry for the late reply.
I tried the same code with Wemos D1 Mini and it calibrated correctly.
Another issue that came up - During the frequency sweep, it would trigger a Software reset in between.
Used ESP.wdtDisable(); but it doesn't seem to work all the time and sometimes, the unit resets at start. Will try figuring out what's happening.

closed 2019-06-02 by sanjum89

u1973295, 2024-04-09:
Hello, I am a student who is studying the fourth year of biomedical engineering. I am doing a project to design a system that is capable of measuring the impedance of the system. The previous code would be very good for me to advance my project, I was wondering if you have been able to fix the error or perfect the code. And if you would be so kind as to provide the link to the ESP32 kit that you have used specifically for this project. Thank u!!