espressif/arduino-esp32 #2773: 1.0.3 Download error form Arduino IDE

joseitor, 2019-05-13:
Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.3-rc1/esp32-1.0.3-rc1.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.3-rc1/esp32-1.0.3-rc1.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.3-rc1/esp32-1.0.3-rc1.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more

me-no-dev, 2019-05-13:
can you please retry?

joseitor, 2019-05-13:
Fixed! :)

closed 2019-05-13 by joseitor

skavinvarnan, 2019-07-22:
This error still persists

Nicolas-Lakes, 2020-07-11:
I have tried many times but no progress, am receiving the same error again and again