espressif/arduino-esp32 #2774: library libfrmn.a missing

anna-seppala, 2019-05-13:
### Hardware:
Board:							featheresp32
Core Installation version:			1.8.7
IDE name:						arduino-cmake
Computer OS:                                        Ubuntu

I'm trying to compile an esp32 example provided [here](https://github.com/Pro/arduino-cmake/tree/feature/esp32)
But my compilation fails because of a linking error:

    /usr/share/arduino-1.8.7/hardware/espressif/arduino-esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lfrmn
    /usr/share/arduino-1.8.7/hardware/espressif/arduino-esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lfr_coefficients
    /usr/share/arduino-1.8.7/hardware/espressif/arduino-esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lmtmn
    /usr/share/arduino-1.8.7/hardware/espressif/arduino-esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lfd_coefficients
    collect2: error: ld returned 1 exit status
    example_esp32/CMakeFiles/example_esp32.dir/build.make:84: recipe for target 'example_esp32/example_esp32.elf' failed
    make[2]: *** [example_esp32/example_esp32.elf] Error 1
    CMakeFiles/Makefile2:424: recipe for target 'example_esp32/CMakeFiles/example_esp32.dir/all' failed
    make[1]: *** [example_esp32/CMakeFiles/example_esp32.dir/all] Error 2
    Makefile:83: recipe for target 'all' failed
    make: *** [all] Error 2

Looking into this, I found that the library libfrmn.a is not in the tools/sdk/lib/ folder. Same problem with the other libraries mentioned in the error message above.

Are these files supposed to be autogenerated or should they be there from the start?

atanisoft, 2019-05-14:
this library was renamed as part of an ESP-IDF update for the 1.0.3-rc1 builds. You will need to do a clean build and make sure you are using the latest ESP-IDF 3.2 branch code.

judge2005, 2019-06-02:
How? I cloned arduino-esp32 latest and I ran get.exe in tools. Still get this error. So how do I get the latest ESP-IDF 3.2 branch code?

atanisoft, 2019-06-02:
Perhaps you should move this issue over to the repo you pulled the example from. Their example is broke due to version differences and is not compatible with esp-idf v3.2. this is not a failure of arduino-esp32.

judge2005, 2019-06-02:
I'm not using this example. I'm just trying to build a project that uses arduino-esp32

atanisoft, 2019-06-02:
@judge2005 I'd suggest opening a new issue with details for your usage and exact failure.

The original issue reported here appears to be due to a broken example that hasn't been maintained in over a year and is very likely broken due to changes in esp-idf during that period.

closed 2019-08-15 by stale[bot]