espressif/arduino-esp32 #2776: Why I can't read the characteristic value?

vcmorini, 2019-05-13:
Hi guys, would you know why I cannot read this characteristic? I can't understand where is my mistake. Need a light!

Objective:  Read the value of a characteristic. Product: Thermostat eq-3 Bluetooth.

### Hardware:
Board:							ESP32 Dev Module
Upload Speed:					921600
CPU Frequency:                                       240MHz(Wifi/BT)
Flash Frequency:					80Mhz
Flash Mode:                                             "QIO"
Flash Size:                                                "4MB" (32Mb)
IDE name:						Arduino 1.8.9 (windows store 1.8.21.0)
PSRAM enabled:                                    Disabled
Computer OS:                                        Windows 10 Pro

### Description:
Describe what is failing:
      Following the provided example by nkolban:
              file->examples->esp32 ble arduino -> ble_client
      I get the value @ or nothing at all.

```
  if (connected) {    
    // Set the characteristic's value to be the array of bytes that is actually a string.
    pRemoteCharacteristic->writeValue(0x4040);

    std::string value_to_be_read = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value_to_be_read.c_str());

  }else if(doScan){
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }

![image](https://user-images.githubusercontent.com/44900293/57658210-5b82d080-75de-11e9-9e12-66dbcde6ac5d.png)

Thanks a lot!

vcmorini, 2019-05-22:
**Follow-Up:**
I am sniffing EQ-3 with the following setup: Adafruit Bluefruit LE Sniffer V1 + Nordic's nRF Sniffer Utility (Windows only) + Wireshark version 1.12.1 -> spoiler alert, using the current stable release of Wireshark is 3.0.1 will not work.

I am using espressif/arduino-esp32 library: [](https://github.com/espressif/arduino-esp32)

Right after my ESP32 connects to the EQ-3, the sniffer shows the connection packets, but after that, it shows no more packets. Writing: nothing sniffed. Reading: nothing sniffed.
It only goes alive, if I disconnect and connect to it again.

Next steps: will I be able to make it work with nkolban snippets? For now, still a question.

chegewara, 2019-05-22:
In arduino there is a bug that has been fixed in kolban's library. There is also a lot of opened issue similar to this one and already answered. 
In BLERemoteService in getcharacteristics function is a bug.

vcmorini, 2019-05-23:
Thanks for your feedback! It is an honor to talk to you chegewara :)

I've installed the ESP32 BLE Arduino library via Arduino IDE (Figure 1), as described here: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md
And deleted the version of espressif.
Figure 1:
https://user-images.githubusercontent.com/44900293/58248239-e9d42080-7d5b-11e9-9aea-61ff4dc65c91.png

Until now I had no success.

A small progress was made when I added a delay(5000) in the Arduino loop (Figure 2) - then I started receiving more sniffer packets (only from write commands).
Figure 2:
https://user-images.githubusercontent.com/44900293/58247547-3f0f3280-7d5a-11e9-889c-cc0d5cd5e43e.png

Read the characteristic triggers nothing in the sniffer, and no answers in the Serial Monitor of Arduino IDE.
As with the write command, with the delay in the loop, it triggers some sniffing packets. The Wireshark finds: "Master	Slave ATT	34 Rcvd Write Command, Handle: 0x0411",  "Slave	Master	LE LL	26	Empty PDU".

The handle 0x0411 is correct (Figure 3), extracted from [Reverse Engineering BLE devices - Sergio Alberti](https://buildmedia.readthedocs.org/media/pdf/reverse-engineering-ble-devices/latest/reverse-engineering-ble-devices.pdf) - Thanks, Sergio Alberti!

Figure 3:
https://user-images.githubusercontent.com/44900293/58247998-4b47bf80-7d5b-11e9-8cd9-64bfa7b538a3.png

During connection request, I receive several error responses in the sniffing Wireshark packets.
Examples: 
59	28.227611000	Slave	Master	ATT	35	Rcvd Error Response - Attribute Not Found, Handle: 0xff08
63	28.258828000	Slave	Master	ATT	35	Rcvd Error Response - Attribute Not Found, Handle: 0x0100
...(many similars)
102	28.603703000	Slave	Master	ATT	35	Rcvd Error Response - Attribute Not Found, Handle: 0x0211
and also:
182	29.234101000	Slave	Master	ATT	35	Rcvd Error Response - **Insufficient Authentication**, Handle: 0x0220

Here I provide the Wireshark sniffer output: [wireshark.txt](https://github.com/espressif/arduino-esp32/files/3211842/wireshark.txt).

**### With nRF Connect:**
I was also curious about how [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) would behave. **It works!** Reading and Writing on EQ-3 works! I can actually control the device. Sniffing its packets, I've just received the:
41	35.127635000	Slave	Master	LE LL	60	CONNECT_REQ
Only! No reading or writing request.
On my second try: it worked. Although writing or reading requests were treated as L2CAP fragments.

### **Partial conclusions:**
**about the sniffing:**
Is the setup:  Adafruit Bluefruit LE Sniffer V1 + Nordic's nRF Sniffer Utility (Windows only) + Wireshark version 1.12.1 reliable? I don't think so.
It didn't pick up the nRF commands all the times.
Returning strange packets. e.g.:
1) 424	14.803079000	Slave	Master	LE LL	63	ADV_NONCONN_IND
2) not recognizing the read and write packets, and treating them as L2CAP fragment - and I could not find the hexadecimal I've sent inside it.

How to solve? -> I disconnected and connected again the  Adafruit Bluefruit LE Sniffer V1 and started the SW again or pressing Restart the running live script.

**about the esp32 ble arduino**
I kindly ask for the expertise of the community! Because I don't know where is the problem. I hope my description was complete, if not, just write me a comment, and I will provide it.
Will this be a problem of pairing? I don't think so, because nRF was not paired as well, and was able to work.
I also tried with ESP32 Feather, but as there is no partition scheme (Figure 4), I could not upload it to the board, since there is not enough memory.
https://user-images.githubusercontent.com/44900293/58252023-67505e80-7d65-11e9-98dc-e0714d8656b2.png

All the best!

chegewara, 2019-05-23:
> Thanks for your feedback! It is an honor to talk to you chegewara :)

I hope its not irony.

I think you are trying to re-invent wheel. There is a lot examples, i see you know how to work with nRF and ble. Just start with simplest examples and nRF connect. If its not working then maybe its hardware issue (antenna).

vcmorini, 2019-05-24:
Hi @chegewara, you were right. And no, definitively not an irony!

I think the problem lies in the fact that I am not able to send hexadecimal values, nor read hexadecimal values.
The caracteristic value is 0x4040 for example.

vcmorini, 2019-06-11:
@chegewara thanks to your support I was able to make it work. And to learn a lot about BLE debugging.
To read the callback:
```
    for (int i = 0; i < length; i++)
    {
      Serial.print("[");
      Serial.print(i);
      Serial.print("]");
      Serial.print(": ");
      Serial.print(pData[i], HEX);
      Serial.print("; ");
    }
```
To write to the device:
`      pRemoteCharacteristic->writeValue(value, 2, true); // this "true" is important to make a write request(opcode 0x12) instead of write command(opcode 0x52)`

closed 2019-06-11 by vcmorini

Koenmani, 2021-02-28:
@vcmorini I was searching for the same and stumbled, through the web, on your lines above. Could you elaborate abit more on how you adjusted the code to control the EQ3 BT valves over bluetooth? Which values are you writing for example? 
If i write "pRemoteCharacteristic->writeValue(0x40-0x40, 2, true);" nothing happens.
Also i can't understand where you put the code snippet to read the value