espressif/arduino-esp32 #2777: CORRUPT HEAP: Bad Tail error

GeorgeFlorian, 2019-05-14:
OS: Linux Mint 19.1
Board: ESP32-Wrover-B DevKitV4
IDE: PlatformioIDE

Hello !

I just got this error:
```
CORRUPT HEAP: Bad tail at 0x3ffde428. Expected 0xbaad5678 got 0x20736563
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 245, function: multi_heap_realloc
abort() was called at PC 0x400e7f9f on core 1

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40089048:0x3ffdc700 0x400892c5:0x3ffdc720 0x400e7f9f:0x3ffdc740 0x400923b1:0x3ffdc770 0x40084914:0x3ffdc790 0x40084979:0x3ffdc7b0 0x40085e19:0x3ffdc7d0 0x4000bedd:0x3ffdc7f0 0x400e109a:0x3ffdc810 0x400e10fc:0x3ffdc830 0x400e129a:0x3ffdc850 0x400e1403:0x3ffdc870 0x400d323d:0x3ffdc890 0x400d32b2:0x3ffdc8c0 0x40179e79:0x3ffdc8e0 0x400dce9d:0x3ffdc900 0x400dd0e1:0x3ffdc990 0x400dbc3e:0x3ffdc9e0 0x400da762:0x3ffdca20 0x400db1fd:0x3ffdca60 0x400d2f71:0x3ffdcac0 0x400d2fd1:0x3ffdcb10 0x400dd587:0x3ffdcb30 0x400db3ef:0x3ffdcb70 0x400db495:0x3ffdcbb0 0x400db6ed:0x3ffdcc00 0x4013b4c1:0x3ffdcc20 0x4013b509:0x3ffdcc60 0x4013b822:0x3ffdcc80 0x4008eff5:0x3ffdccb0
```

```
0x40089048: vTaskPriorityInherit at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 4079
0x400892c5: xTimerCreateTimerTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/timers.c line 259
0x400e7f9f: _vfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1405
0x40084914: esp_rom_spiflash_write_status at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/spi_flash_rom_patch.c line 284
0x40084979: esp_rom_spiflash_unlock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/spi_flash_rom_patch.c line 70
0x40085e19: rtc_clk_bbpll_enable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/rtc_clk.c line 445
0x400e109a: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1063
0x400e10fc: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 962
0x400e129a: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1046
0x400e1403: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1118
0x400dce9d: spiffs_object_truncate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 1751
0x400dd0e1: spiffs_object_truncate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 1844
0x400dbc3e: spiffs_cb_object_event at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 1134
0x400da762: SPIFFS_rename at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_hydrogen.c line 879
0x400db1fd: spiffs_obj_lu_find_entry_visitor at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 154
0x400dd587: spiffs_object_read at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 2044
0x400db3ef: spiffs_erase_block at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 244
0x400db495: spiffs_obj_lu_scan at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 361
0x400db6ed: spiffs_obj_lu_find_id_and_span at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spiffs/spiffs/src/spiffs_nucleus.c line 537
```

The error happened while I was trying out the web-pages and "testing" them: changing some input values, refreshing... trivial stuff.
My project relies heavily on ESPAsyncWebServer and SPIFFS. It also has a HTTP Client that GETs the body of a URL. On top of all of that I'm using SmartMatrix library to drive a 64x32 RGB Matrix.
```
#include <Arduino.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>
#include <HTTPClient.h>
#include <Adafruit_GFX.h>
#include <SmartMatrix_GFX.h>
#include <SmartMatrix3.h>
#include <Fonts/FreeSansBold18pt7b.h>
#include <Fonts/FreeSans9pt7b.h>
```

```
server.on("/register", HTTP_GET, [](AsyncWebServerRequest *request){              
    request->send(SPIFFS, "/indexAP.html", "text/html");
  });
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){              
    request->redirect("/register");
  });
  server.on("/register", HTTP_POST, [](AsyncWebServerRequest * request){
    int params = request->params();
    String values_user[2];
    for(int i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isPost()){
          logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
          values_user[i] = p->value();            
        } else {
[... 114 lines cut ...]
    for(int i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isPost()){
          logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
          values_dhcp[i] = p->value();            
        } else {
            logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
          }
    }
    if(values_dhcp[0] != NULL && values_dhcp[0].length() != 0 &&
      values_dhcp[1] != NULL && values_dhcp[1].length() != 0) {
      File inputsWrite = SPIFFS.open("/inputs.txt", "w");
      if(!inputsWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write DHCP IP credentials !");
      inputsWrite.println(values_dhcp[0]);  // SSID
      inputsWrite.println(values_dhcp[1]);  // Password
      inputsWrite.close();
      logOutput("Configuration saved !");
      logOutput("Restarting in 5 seconds...");
      request->redirect("/logs");
      restartFlag = true;
    } else request->redirect("/dhcpIP");
  
  }); // server.on("/dhcpLogin", HTTP_POST, [](AsyncWebServerRequest * request)
  server.begin();
```

me-no-dev, 2019-05-14:
Known issue in the Async lib. Have tried to fix and failed. When I have more spare time I'll try some more.

closed 2019-05-14 by me-no-dev

GeorgeFlorian, 2019-05-14:
> Known issue in the Async lib. Have tried to fix and failed. When I have more spare time I'll try some more.

But can I limit the number of clients to just 1 ?

GeorgeFlorian, 2019-05-14:
> Known issue in the Async lib. Have tried to fix and failed. When I have more spare time I'll try some more.

@me-no-dev 

But can I limit the number of connections to the server to just 1 ?

Is there something similar to this ? 
`WiFi.softAPgetStationNum() == 1`

me-no-dev, 2019-05-14:
limiting is not the way to go.

joaolrc, 2020-05-18:
Hello all. Any updates on this one?

thbl, 2020-11-16:
getting the same error :( @me-no-dev  could you please help