espressif/arduino-esp32 #2779: ESP32 EEPROM Example "EEPROM Fail to initialise"  - Default CSV doenst have EEPROM line - similar to Problem #569

andrelmbraga, 2019-05-14:
Hello guys,

I would like to know why my default.csv doesnt have the eeprom line. 

I'm trying to use the eeprom example however I'm getting the following msg:
"EEPROM Fail to initialise" 

Is similar to the Problem #569, however I did the update of my repo  and my default.csv still doesnt have the eeprom line.

Tutorial that I used to update:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

Following the default.csv  file: 

# Name,   Type, SubType,   Offset,  Size, Flags
--
nvs,      data, nvs,     0x9000,    0x5000,
otadata,  data, ota,     0xe000,    0x2000,
app0,     app,    ota_0,   0x10000, 0x140000,
app1,     app,    ota_1,   0x150000,0x140000,
spiffs,   data, spiffs,  0x290000,0x170000,

I'll be really glad if you can give me a tip.

atanisoft, 2019-05-14:
Eeprom no longer depends on a partition for storage. This would be a bug in the example.

andrelmbraga, 2019-05-14:
Thanks! You can close the issue.

closed 2019-05-14 by andrelmbraga