espressif/arduino-esp32 #2780: Guru Meditation Error: Core  1 panic'ed (LoadProhibited)

GeorgeFlorian, 2019-05-14:
Hello !

I am getting the following error:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dfc96  PS      : 0x00060d30  A0      : 0x800dfd24  A1      : 0x3ffb1e00  
A2      : 0x3ffb1f14  A3      : 0x00000000  A4      : 0x3f4043e5  A5      : 0x3ffb1f14  
A6      : 0x00000005  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1de0  
A10     : 0x3ffb1e0c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffdda5d  
A14     : 0x3ffb1e04  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xfffffffe  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400dfc96:0x3ffb1e00 0x400dfd21:0x3ffb1e30 0x400dfddf:0x3ffb1e70 0x400d47a5:0x3ffb1e90 0x400e2491:0x3ffb1fb0 0x4008eff5:0x3ffb1fd0

PC: 0x400dfc96: nvs::Page::mLoadEntryTable() at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/compressed_enum_table.hpp line 39
EXCVADDR: 0x00000000

Decoding stack results
0x400dfc96: nvs::Page::mLoadEntryTable() at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/compressed_enum_table.hpp line 39
0x400dfd21: nvs::Page::mLoadEntryTable() at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp line 586
0x400dfddf: nvs::Page::mLoadEntryTable() at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp line 620
0x400d47a5: __register_frame_info_bases at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind-dw2-fde.c line 97
0x400e2491: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1411
```
from the following code:
```
if (WiFi.status() == WL_CONNECTED && SPIFFS.exists("/configFile.txt")) {  //Check the current connection status 
      if(URL.length() > 15) {
        HTTPClient http;
        http.begin(URL); //Specify the URL
        int httpCode = http.GET(); //Make the request  
        if (httpCode > 0) { //Check for the returning code
          logOutput((String)"[HTTP] GET... code: " + httpCode);
          if(httpCode == HTTP_CODE_OK) {
              String payload = http.getString();
              logOutput(payload);
              signPlace = getPlaces(payload);
              oldSignPlace = signPlace;
            }
        } else {
            logOutput((String)"[HTTP] GET... failed, error: " + http.errorToString(httpCode).c_str());
          }  
        http.end(); //Free the resources
        } else {
            logOutput("URL Link is invalid ! Please enter another URL");
        }
    } // if (WiFi.status() == WL_CONNECTED)
```

It appears randomly whenever I change the URL.

This is how I change the URL from a webpage
```
server.on("/home", HTTP_POST, [](AsyncWebServerRequest * request){
        int params = request->params();
        String urlCopy = "";
        for(int i=0;i<params;i++){
          AsyncWebParameter* p = request->getParam(i);
            if(p->isPost()){
              logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
              if(p->name() == "getURL" && p->value() != NULL) urlCopy = p->value();
              if(p->name() == "getBrightness" && p->value() != NULL)  Brightness = p->value();
              if(p->name() == "getRefreshRate" && p->value() != NULL)  urlRefreshRate = p->value(); 
            } else {
                logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n"); 
              }
        } // for(int i=0;i<params;i++)
        
        // Serial.println("_TEST");
        // if(Brightness.length()>2) {
        //   Brightness = "5";
        // } 
        // if(urlRefreshRate.length()>2) {
        //   urlRefreshRate = "3";
        // }

        if(urlCopy != NULL && urlCopy.length() != 0) {
          URL = urlCopy;
          File urlWrite = SPIFFS.open("/configFile.txt", "w");
          if(!urlWrite) logOutput((String)"ERROR_INSIDE_URL_POST ! Couldn't open file to write URL !");
          urlWrite.println(urlCopy);
          urlWrite.close();
          request->redirect("/home");
        } else {
          request->redirect("/home");
        }
      });
```

atanisoft, 2019-05-14:
It looks like you are ending up with a null value for URL or urlCopy under some circumstances. from the provided code though it doesn't look like the exception being raised is coming from this code though. The exception appears to be related to NVS (or EEPROM) with a null key name or attempt to read an invalid flash location possibly.

GeorgeFlorian, 2019-05-15:
> It looks like you are ending up with a null value for URL or urlCopy under some circumstances. from the provided code though it doesn't look like the exception being raised is coming from this code [...]

This is the handler for the page in which I change the URL:
```
      server.on("/home", HTTP_POST, [](AsyncWebServerRequest * request){
        int params = request->params();
        for(int i=0;i<params;i++){
          AsyncWebParameter* p = request->getParam(i);
            if(p->isPost()){
              logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
              if(p->name() == "getURL" && p->value() != NULL) URL = p->value();
              if(p->name() == "getBrightness" && p->value() != NULL)  Brightness = p->value();
              if(p->name() == "getRefreshRate" && p->value() != NULL)  urlRefreshRate = p->value(); 
            } else {
                logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n"); 
              }
        } // for(int i=0;i<params;i++)
        
        Serial.println((String)"Brightness: " + Brightness);
        Serial.println((String)"urlRefreshRate: " + urlRefreshRate);
        if(URL != "INVALID_URL" && URL != NULL && URL.length() > 15) {
          File urlWrite = SPIFFS.open("/configFile.txt", "w");
          if(!urlWrite) logOutput((String)"ERROR_INSIDE_URL_POST ! Couldn't open file to write URL !");
          urlWrite.println(URL);
          urlWrite.close();
          request->redirect("/home");
        } else {
          request->redirect("/home");
        }
      });
```

`File urlWrite = SPIFFS.open("/configFile.txt", "w");` should be safe, because `"w"` discards the previous content, if there is one, and the file is treated as a new empty file.
[image]

I'm also checking for the existence of the SPIFFS file in which I have stored the URL before starting `HTTP Client`.
```
if (WiFi.status() == WL_CONNECTED && SPIFFS.exists("/configFile.txt"))
```

Also, I've previously added `Serial.println()` throughout `void loop()` :
```
if (WiFi.status() == WL_CONNECTED && SPIFFS.exists("/configFile.txt")) {  //Check the current connection status
      if(URL.length() > 15) {
        HTTPClient http;
        http.begin(URL); //Specify the URL
        // Serial.println("http.begin(URL);");
        int httpCode = http.GET(); //Make the request  
        // Serial.println("int httpCode = http.GET();");
        if (httpCode > 0) { //Check for the returning code
          // Serial.println("if (httpCode > 0)");
          logOutput((String)"[HTTP] GET... code: " + httpCode);
          if(httpCode == HTTP_CODE_OK) {
            // Serial.println("if(httpCode == HTTP_CODE_OK)");
            String payload = http.getString();
            logOutput(payload);
            signPlace = getPlaces(payload);
            oldSignPlace = signPlace;
          }
        } else {
            logOutput((String)"[HTTP] GET... failed, error: " + http.errorToString(httpCode).c_str());
          }  
        http.end(); //Free the resources
        } else {
            logOutput("URL Link is invalid ! Please enter another URL");
        }
    } // if (WiFi.status() == WL_CONNECTED)
```
And it went through all of them and then crashed.
It looked like that after I introduced the URL, the first loop iteration worked, but the second crashed.

closed 2019-05-17 by GeorgeFlorian