espressif/arduino-esp32 #2782: esp_deep_sleep.h is depreciated, where is esp_deep_sleep_pd_config defined?

nevercast, 2019-05-15:
### Hardware:
Board:							Custom board, uses WROVER
Core Installation version:			https://github.com/espressif/arduino-esp32/commit/fd5a2f08f95429234179453a8d0c44fcb4770843
IDE name:						IDF Component, text editor.
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:					921600
Computer OS:                                        Ubuntu 18.04

### Description:
I am wanting to call `esp_deep_sleep_pd_config` because I wish to keep the RTC Fast Memory on when in deepsleep. I currently include esp_deep_sleep.h and call the function before sleeping. 

I have updated Arduino to https://github.com/espressif/arduino-esp32/commit/fd5a2f08f95429234179453a8d0c44fcb4770843

I have updated ESP-IDF to https://github.com/espressif/esp-idf/commit/977854975

I am using arduino-esp32 as an esp-idf component.

If I include esp_sleep.h instead of esp_deep_sleep.h, esp_deep_sleep_pd_config is undefined. I will ignore the depreciated warning for the moment. Please advise me on the appropriate migration.

atanisoft, 2019-05-15:
esp_sleep_pd_config is the replacement per the esp_deep_sleep.h file, esp_deep_sleep_pd_config is an inline function calling this method which is present in esp_sleep.h

nevercast, 2019-05-16:
Thank you. I will use `esp_sleep_pd_config`

closed 2019-05-16 by nevercast