espressif/arduino-esp32 #2792: How to reduce Tx WiFi power

bill-orange, 2019-05-16:
I have not found a way to reduce WiFi power on an ESP32.  `esp_wifi_set_max_tx_power(1);` does not work.

Can it be done?

I am running a Wemos ESP32 with the battery and OLED (probably counterfeit).  Also connected to the I2C bus is a CCS811 environmental sensor.  It is spitting our incorrect readings more often that not.  It runs fine in the sample sketches.  To find the problem with my sketch, I eliminated functions one-by-one.  I removed OLED display, aRest, DHT22, server and finally Wifi.  It appears WiFi is the problem.  This leaves me with the question, is the WiFi TX jamming the sensor or is there some problem between the WiFi drivers and and the sensor's driver.  Thus my desire to set Tx power at the minimum.

Oddly, I am not getting any sort of I2C errors, just bad data.  If VOCs were as high as it reads with WiFi running, I would be dead.

This would be a tough problem for someone to duplicate since a CCS811 sensor would be needed.

So back to the question, for test purposes can Tx power be reduced?  Any other troubleshooting  advice I may not have thought of would also be appreciated.

jandegr, 2019-05-17:
Hi

I am using `WiFi.setTxPower(WIFI_POWER_MINUS_1dBm);`

and for verification `WiFi.getTxPower();`

Regards.

bill-orange, 2019-05-17:
@jandegr thanks.

I will see if reducing power solves my sensor problem. Might hyst be a bad sensor.

lbernstone, 2019-05-17:
That sounds more like interference. Don't run your wires over/under the antenna.

bill-orange, 2019-05-17:
I agree. I have run the wires every way that I can think of, added extra pull-ups to the I2C bus, put a tantalum cap on the power supply, and twisted the 4” wire bundle to make a poor man’s shield. No luck so far.

stickbreaker, 2019-05-17:
@bill-orange , you say that i2c comms are working, so just shield the sensor,  It measure gasses, so put it inside a Faraday Cage of Aluminum Foil. [A Faraday cage for 2.4ghz signals only has to have holes smaller than 1/10 of wavelength](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=3&cad=rja&uact=8&ved=2ahUKEwjqh4zuxqPiAhW1ITQIHS_BAQMQFjACegQIDBAL&url=https%3A%2F%2Fforum.allaboutcircuits.com%2Fthreads%2Fholes-in-a-faraday-cage.69512%2F&usg=AOvVaw0Ur__cvr1QF0sTLACcL6Wm). So, as long as the openings in the shield are smaller no WiFi will intrude.  Anything smaller than 1cm will block WiFi but let atmosphere in.

Chuck.

bill-orange, 2019-05-17:
I will give that a try, but a couple of things first. Lots of people on the Adafruit and SparkFun forums have experienced this problem but I think I am the first to try turning WiFi off. 

The CCS811 has a self generated calibration factor stored in its own EEPROM. Right now I am serial printing the calibration bytes. I want to see if that changes when the failure occurs. It it is getting corrupted, I could just write back the correct value. We shall see.

bill-orange, 2019-05-18:
It does not appear that the calibration register is being corrupted.

Armed with the new-found knowledge of how to turn down ESP32 power from   @jandegr  , I set power with:

`if (LOWPOWER) WiFi.setTxPower(WIFI_POWER_7dBm);` 

It made some difference.  If I cupped my hands around the ESP32 the sensor it would behave. 

 So, @stickbreaker  following your suggestion (somewhat) I put a piece of scotch tape on the backside of the little sensor board and and painted it with conductive paint which I then bridged to the ground pin.  So far this seems to be doing the trick.  I am a ham so I happened to have conductive paint.

Thanks Guys, I will close this.

closed 2019-05-18 by bill-orange