espressif/arduino-esp32 #2794: I2C issue with touch on standard I2C lines for ESP32-PICO-D4

shr1Khr, 2019-05-17:
So I'm getting following error on the ESP32 trying to read the touch points on a FT6336G touch based LCD. Please note, I have tested this LCD's touch on other lines, and it does work. I have to make changes to the code and I chose the standard lines for LCD and touch and since then this issue started appearing.

Error:

`[E][esp32-hal-i2c.c:1104] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init
[E][esp32-hal-i2c.c:1023] i2cProcQueue():  Busy Timeout start=0x1b2d, end=0x1b5f, =50, max=50 error=1
[E][esp32-hal-i2c.c:1413] i2cDumpI2c(): i2c=0x3ffc1108
[E][esp32-hal-i2c.c:1414] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[E][esp32-hal-i2c.c:1416] i2cDumpI2c(): lock=0x3ffe2e40
[E][esp32-hal-i2c.c:1418] i2cDumpI2c(): num=0
[E][esp32-hal-i2c.c:1419] i2cDumpI2c(): mode=1
[E][esp32-hal-i2c.c:1420] i2cDumpI2c(): stage=3
[E][esp32-hal-i2c.c:1421] i2cDumpI2c(): error=1
[E][esp32-hal-i2c.c:1422] i2cDumpI2c(): event=0x3ffe2b9c bits=0
[E][esp32-hal-i2c.c:1423] i2cDumpI2c(): intr_handle=0x3ffdea70
[E][esp32-hal-i2c.c:1424] i2cDumpI2c(): dq=0x3ffe2dcc
[E][esp32-hal-i2c.c:1425] i2cDumpI2c(): queueCount=1
[E][esp32-hal-i2c.c:1426] i2cDumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:1427] i2cDumpI2c(): byteCnt=0
[E][esp32-hal-i2c.c:1384] i2cDumpDqData(): [0] 71 R STOP buf@=0x3ffc4f24, len=16, pos=0, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:1402] i2cDumpDqData(): 0x0000: ................                 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:1445] i2cDumpInts(): enable Core Debug Level "Error"
[E][esp32-hal-i2c.c:1104] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init
0, 0
[E][esp32-hal-i2c.c:1023] i2cProcQueue():  Busy Timeout start=0x1c87, end=0x1cb9, =50, max=50 error=1
[E][esp32-hal-i2c.c:1413] i2cDumpI2c(): i2c=0x3ffc1108
[E][esp32-hal-i2c.c:1414] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[E][esp32-hal-i2c.c:1416] i2cDumpI2c(): lock=0x3ffe2e40
[E][esp32-hal-i2c.c:1418] i2cDumpI2c(): num=0
[E][esp32-hal-i2c.c:1419] i2cDumpI2c(): mode=1
[E][esp32-hal-i2c.c:1420] i2cDumpI2c(): stage=3
[E][esp32-hal-i2c.c:1421] i2cDumpI2c(): error=1
[E][esp32-hal-i2c.c:1422] i2cDumpI2c(): event=0x3ffe2b9c bits=0
[E][esp32-hal-i2c.c:1423] i2cDumpI2c(): intr_handle=0x3ffdea70
[E][esp32-hal-i2c.c:1424] i2cDumpI2c(): dq=0x3ffe2dcc
[E][esp32-hal-i2c.c:1425] i2cDumpI2c(): queueCount=1
[E][esp32-hal-i2c.c:1426] i2cDumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:1427] i2cDumpI2c(): byteCnt=0
[E][esp32-hal-i2c.c:1384] i2cDumpDqData(): [0] 70 W STOP buf@=0x3ffc4faa, len=1, pos=1, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:1402] i2cDumpDqData(): 0x0000: .                                02 
[E][esp32-hal-i2c.c:1445] i2cDumpInts(): enable Core Debug Level "Error"
[E][esp32-hal-i2c.c:1023] i2cProcQueue():  Busy Timeout start=0x1d0d, end=0x1d3f, =50, max=50 error=1
[E][esp32-hal-i2c.c:1413] i2cDumpI2c(): i2c=0x3ffc1108
[E][esp32-hal-i2c.c:1414] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[E][esp32-hal-i2c.c:1416] i2cDumpI2c(): lock=0x3ffe2e40
[E][esp32-hal-i2c.c:1418] i2cDumpI2c(): num=0
[E][esp32-hal-i2c.c:1419] i2cDumpI2c(): mode=1
[E][esp32-hal-i2c.c:1420] i2cDumpI2c(): stage=3
[E][esp32-hal-i2c.c:1421] i2cDumpI2c(): error=1
[E][esp32-hal-i2c.c:1422] i2cDumpI2c(): event=0x3ffe2b9c bits=0
[E][esp32-hal-i2c.c:1423] i2cDumpI2c(): intr_handle=0x3ffdea70
[E][esp32-hal-i2c.c:1424] i2cDumpI2c(): dq=0x3ffe2dcc
[E][esp32-hal-i2c.c:1425] i2cDumpI2c(): queueCount=1
[E][esp32-hal-i2c.c:1426] i2cDumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:1427] i2cDumpI2c(): byteCnt=0
[E][esp32-hal-i2c.c:1384] i2cDumpDqData(): [0] e0 W STOP buf@=0x3ffc4faa, len=1, pos=1, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:1402] i2cDumpDqData(): 0x0000: .                                02 
[E][esp32-hal-i2c.c:1445] i2cDumpInts(): enable Core Debug Level "Error"

 (    0,    0) p=  1 r=  0 c=  0 ID=  0 
0, 0
`

So, the device initializes normally via this code - 
Pin definition-
`//Touch // Address = 0x38

#define TSDA      21 

#define TSCL      22 

#define TINT      39 // Interrupt to ESP32 from Touch panel //SENSOR_CAPP = GPIO37

#define TRST      -1//14 // Reset for Touch //ok

#define Tctrl     -1//21
`
Intializer-
`TouchDriver touch = TouchDriver(Tctrl, TSCL, TSDA, TINT, TRST); // Touch initialization
`
Which triggers followin in TouchDriver.cpp; the appropriate variables are defined in TouchDriver.h-

`TouchDriver::TouchDriver(int32_t _tctrl, int32_t tscl, int32_t tsda, int32_t tint, int32_t trst){
  _tctrl= _tctrl;
  _tscl = tscl;
  _tsda = tsda;
  _tint = tint;
  _trst = trst;
}`

Then in setup{} I trigger this - 
`    touch.beginTP();      // Start TP    
`
and its code is-
`void TouchDriver::beginTP(){
  pinMode(_tctrl, OUTPUT);
  digitalWrite(_tctrl, LOW);
  pinMode(_tsda, INPUT_PULLUP);
  pinMode(_tscl, INPUT_PULLUP);
  Wire.begin(_tsda, _tscl, 400000); //sda, scl, bus speed = 100kHz -> Max 400kHz
  //SoftwareSerial mySerial(_tsda, _tscl);
  pinMode(_tint, INPUT);
  //pinMode(_tint, INPUT_PULLUP);
  //pinMode(_trst, OUTPUT);

  //Use this if TCH_RES is attached to a pin)
  //digitalWrite(_trst, LOW);
  //delay(200);
  //digitalWrite(_trst, HIGH);
  //delay(200);

  //Send Device Mode
  sendCommand(FT_DEVIDE_MODE, 0x00);

  //ID_G_MODE set interrupt
  sendCommand(FT_ID_G_MODE, 0x00);
  //  0x00 - Polling Mode
  //  0x01 - Trigger Mode

  //Touch Threshold
  sendCommand(FT_ID_G_THGROUP, 22);
  
  //Reporting Rate
  sendCommand(FT_ID_G_PERIODACTIVE, 12);
}`

Now, the setup initalizes and everything is fine. In the loop I have this -
`if (digitalRead(TINT) == 0){ //User touched the screen - interrupt
    lastActiveTime = millis();// need to keep screen up!
    getTouchPoints();
    Serial.print(pt_touched_x);
    Serial.print(", ");
    Serial.println(pt_touched_y);
//then I handle what to do with the touch points on screen

`
and at this point it breaks. I suppose the read points, eventhough incorrect, but still can be seen in the error message. It is shown as 0,0 in the error message. I'm not using I2C bus for any other device but this. I checked with speed of 100000, and 400000, but same error.

Thoughts?

Here is TouchDriver.h if you would like to check -
`#ifndef _TouchDriver_
#define _TouchDriver_
#include<Wire.h> // for I2C required

#if ARDUINO >= 100
 #include "Arduino.h"
 #include "Print.h"
#else
 #include "WProgram.h"
#endif

//Check for additional code
//https://github.com/crystalfontz/CFAF240400C0-030SC/blob/master/CFAF240400C0030T/CFAF240400C0030SC.ino

//  Touch Screen Controller Commands Used
#define FT62XX_ADDR 0x38
#define FT62XX_REG_VENDID 0xA8
#define FT62XX_REG_CHIPID 0xA3

//  Touch Screen Controller Commands Used
#define FT_DEVIDE_MODE                    (0x00)
#define FT_REG_NUM_FINGER                 (0x02)  
#define FT_TP1_REG                        (0X03)    
#define FT_TP2_REG                        (0X09)    
#define FT_ID_G_LIB_VERSION               (0xA1)    
#define FT_ID_G_MODE                      (0xA4)      
#define FT_ID_G_THGROUP                   (0x80)      
#define FT_ID_G_PERIODACTIVE              (0x88)    
#define ID_G_SPEC_GESTURE_ENABLE          (0xD0)   

class TouchDriver{
 public:

 TouchDriver(int32_t tctrl, int32_t tscl, int32_t tsda, int32_t tint, int32_t trst); //Constructor
 void beginTP(); 
 void sendCommand(uint8_t reg, uint8_t parametr);
 void touchDemo();
 void SerPrintFF(const __FlashStringHelper *fmt, ...);
 uint8_t readRegister8(uint8_t reg);
 String getTouch();

 private:
int32_t _tctrl, _tscl, _tsda, _tint, _trst;
String temp_tp;

};

#endif`

stickbreaker, 2019-05-17:
Please update to the current version, of the repo.
>[E][esp32-hal-i2c.c:1104] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init

The message if from an older version, The current version 1.0.3-rc1 would produce a different message:
```
  if(!digitalRead(sda) || !digitalRead(scl)) { // bus in busy state
        log_e("Bus Invalid State, TwoWire() Can't init sda=%d, scl=%d",digitalRead(sda),digitalRead(scl));
        return false; // bus is busy
    }
    return true;
}
```
But, the message is telling you that the pins you have used for SCL and/or SDA are not at 3.3v when you try to init `Wire.begin()`.

This is points to a hardware problem. 
* update to the latest code
* verify the voltages on your SDA, and SCL pins

Chuck.

shr1Khr, 2019-05-20:
Hi Stickbreaker, 
Sorry for the delayed response. I redid the soldering on the board, and I found one of the capacitors for the power lines was missing, and hence the jitters on the 3.3V lines were there, which I suppose was causing the issue. Its resolved now. Really appreciate your help. Thanks!

closed 2019-05-20 by stickbreaker