espressif/arduino-esp32 #2795: getting error in esp8266

rohitkhoja, 2019-05-18:
### Hardware:
Board:							NodeMcu 1.0(ESP-12E Module)
Core Installation version:			1.8.9
IDE name:						Arduino IDE
cpu Frequency:					80Mhz

Upload Speed:					115200
Computer OS:                                        Windows 10

i am getting the same error for every code 

esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:31:3c:cf
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 277008 bytes to 200097...

Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 277008 bytes (200097 compressed) at 0x00000000 in 17.7 seconds (effective 125.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

atanisoft, 2019-05-18:
This is not the right place to report esp8266 issues. This is for the esp32 only. 

Additionally, there is no error in any of the output you have provided.

rohitkhoja, 2019-05-18:
sorry for that 

but the problem is i can not see anything on serial monitor and in blink code led is also not blinking

atanisoft, 2019-05-18:
I can't help you with those issues unfortunately, you didn't provide code. I'd suggest moving this over to https://github.com/esp8266/Arduino

rohitkhoja, 2019-05-18:
thanku for help

closed 2019-05-18 by rohitkhoja