espressif/arduino-esp32 #2801: ESP32 HTTPRedirect

chuckmerja, 2019-05-20:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

chuckmerja, 2019-05-20:
Using Adafruit HUZZAH ESP32 vs Adafruit HUZZAH ESP8266.  The HTTPSRedirect library (https://github.com/electronicsguy/ESP8266/tree/master/HTTPSRedirect) on the 8266 and can successfully upload data to a google sheet.  When I try to use the same library with the ESP32, I get 4 to 6 successful uploads, although I get "sheet temporarily moved" messages and then I get "not connected to client" errors and the code won't connect until I reboot the device, then it immediately uploads again for 4 to 6 updates and then fails.  I've changed the timing on the updates, from a few seconds to 15 minutes with the same results.
So is there an ESP32 library equivalent to the 8266 HTTPSRedirect?  I hope that is the right question :o))

chuckmerja, 2019-05-20:
I need to use Adafruit HUZZAH ESP32 vs Adafruit HUZZAH ESP8266, to get more analog ports.  The HTTPSRedirect library (https://github.com/electronicsguy/ESP8266/tree/master/HTTPSRedirect) works great on the 8266 and can successfully upload data to a google sheet.  When I try to use the same library with the ESP32, I get 4 to 6 successful uploads, although I get "sheet temporarily moved" messages and then I get "not connected to client" errors and the code won't connect until I reboot the device, then it immediately uploads again for 4 to 6 updates and then fails.  I've changed the timing on the updates, from a few seconds to 15 minutes with the same results.
So is there an ESP32 library equivalent to the 8266 HTTPSRedirect?  I hope that is the right question :o))

lbernstone, 2019-05-20:
The library claims to support esp32, so you should open the issue there.

closed 2019-08-14 by stale[bot]