espressif/arduino-esp32 #2802: Partition Scheme

laercionit, 2019-05-20:
Hello .. Good night.
It would be possible to include the partition type 3MB for APP + OTA / 1MB SPIFFS
I have 99% use of APP and I can not stay in OTA and SPIFFS, I use the partition mentioned above in ESP8266.

atanisoft, 2019-05-20:
select the huge_app partition type which is close to what you are asking for (SPIFFS is just under 1MB). This can be selected from the menu when you select the ESP32 Dev Module board type.

laercionit, 2019-05-20:
Hello, thank you @atanisoft.

The problem is that this option has no OTA
What I wish would be (3MB (APP + OTA) / 1MB SPIFFS)
My project will only have to run with this partition distribution. Is it possible to include or inform me how to proceed so that I can do it myself?

atanisoft, 2019-05-20:
The ESP32 only has 4MB flash so 3MB APP with OTA is not feasible. There are ESP32 boards with 16MB flash but these are less common.

If you require OTA you have the options of: Default (1.2MB app, 1.5MB SPIFFS), Minimal (1.3MB app, 700kb SPIFFS), min_spiffs (1.9MB APP, 190KB SPIFFS).

If none of these suit your needs you can define your own partition scheme by modifying the existing ones under tools/partitions if you use Arduino IDE. If you use PlatformIO you can define a custom partition in platformio.ini in the project tree and not modify any files in the distribution.

closed 2019-08-14 by stale[bot]

QGB, 2022-05-07:
C:\Users\qgb\Documents\Arduino\libraries