espressif/arduino-esp32 #2805: Change menuconfig PSRAM Setting to provide memory via malloc()

lBenji, 2019-05-21:
I'm trying to change the psram menuconfig setting to Provide external RAM via malloc().

In the IDF Environment with Arduino extension it is working perfectlly, but I' m unable to get the config working in the normal ESP Enviroment.

I have replaced all orignal files with the new builded libraries ,bootloader and the sdk config.

But the setting is still not applied

lbernstone, 2019-05-21:
You have not explained well what you would like to achieve or what problem you are having.  Is this a specific menuconfig option you need?  arduino-esp32 is compiled with psram enabled by default.  If you choose a board type that has psram (ESP32-WROVER), psram will be enabled.  You can then allocate it with the ps_malloc (or ps_calloc) functions.

lBenji, 2019-05-22:
Yes my problem is htat I'm unable to set the specific psram setting. 

I require this setting, because my current code is using malloc instead of ps_malloc and is running on many different Mikrocontrollers.

closed 2019-08-14 by stale[bot]