espressif/arduino-esp32 #2816: Slow SD functions (open, close, remove...)

ViniP8, 2019-05-24:
Hi!!

I'm trying to read, write and create files on my SD card using ESP32. The problem is, I need to do this with multiple files and it was taking a lot of time.
So I decided to test how long it takes to create a new file, read and delete it with the following code:
```C
#include <SPI.h>
#include <SD.h>

#define SD_PIN 4
File file;

void setup() {
  Serial.begin(115200);
  SPI.begin(18, 19, 23);
  SD.begin(SD_PIN);

  long int millisA = millis();
  file = SD.open("/test.txt", FILE_WRITE);
  Serial.print("Open new file(write): ");
  Serial.println(millis() - millisA);

  file.print("Test");

  millisA = millis();
  file.close();
  Serial.print("Close file (write): ");
  Serial.println(millis() - millisA);
  
  millisA = millis();
  Serial.print("Open file(read): ");
  file = SD.open("/test.txt");
  Serial.println(millis() - millisA);
  
  millisA = millis();
  SD.remove("/test.txt");
  Serial.print("Remove file: ");
  Serial.println(millis() - millisA);
  
  millisA = millis();
  file.close();
  Serial.print("Close file (read): ");
  Serial.println(millis() - millisA);
  }

void loop() {
} 
```

The results are (in ms):
```
Open new file(write): 140
Close file (write): 75
Open file(read): 104
Remove file: 145
Close file (read): 0
```
Isn't it a very long time only to open and remove files (>100ms a small file) ?

ViniP8, 2019-05-24:
I've just reformatted the microSD and got better results, but I still think that 55ms to open a file is a lot, isn't it?

```
Open new file(write): 26
Close file (write): 68
Open file(read): 17
Open file(write): 55 //file = SD.open("/test.txt", FILE_WRITE);
Remove file: 29
Close file (read): 0
```

ViniP8, 2019-05-24:
So I tested the time on SPIFFS, and it was much better (the "create new file" isn't really needed in my project). The problem is that I'll need more storage space than 4MB, so I'll have to use the microSD card.
Isn't there any way to make this times on SD? 

**SPIFFS SPEED:**
```
Open new file(write): 234
Close file (write): 1
Open file(read): 0
Test
Open file(write): 2
Remove file: 1
Close file (read): 0
```

rel1ct, 2019-06-02:
09:16:10.737 -> SD Card Size: 7616 MB
09:16:10.737 -> Open new file(write): 4
09:16:10.783 -> Close file (write): 35
09:16:10.783 -> Open file(read): 4
09:16:10.830 -> Remove file: 18
09:16:10.830 -> Close file (read): 0

lbernstone, 2019-06-02:
[SD_MMC](https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC) uses 4 lane SDIO mode, instead of the single lane SPI in the SD library.  [Important info about SD](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card)

closed 2019-08-15 by stale[bot]